YosysHQ/yosys #4886: Yosys assert while making https://github.com/ZipCPU/wb2axip

j-f-d, 2025-02-08:
### Version

Yosys 0.33 (git sha1 2584903a060)

### On which OS did this happen?

Linux

### Reproduction Steps

I installed OSS CAD Suite, then cloned https://github.com/ZipCPU/wb2axip

$ make

### Expected Behavior

Either fail with an error, or succeed.

### Actual Behavior

```
Generating RTLIL representation for module `$paramod$2721230d014e2e2a1923a0fbb62a9fbea2305c2d\axilxbar'.
Warning: Replacing memory \r_rdata with list of registers. See axilxbar.v:1430, axilxbar.v:1425
Warning: Replacing memory \r_rresp with list of registers. See axilxbar.v:1429, axilxbar.v:1424
Warning: Replacing memory \r_bresp with list of registers. See axilxbar.v:1310, axilxbar.v:1307
Warning: Replacing memory \rgrant with list of registers. See axilxbar.v:903, axilxbar.v:898
Warning: Replacing memory \wgrant with list of registers. See axilxbar.v:755, axilxbar.v:750
Warning: Replacing memory \rrequest with list of registers. See axilxbar.v:517
Warning: Replacing memory \wrequest with list of registers. See axilxbar.v:447
Warning: Replacing memory \rrequested with list of registers. See axilxbar.v:623, axilxbar.v:601
Warning: Replacing memory \wrequested with list of registers. See axilxbar.v:590, axilxbar.v:572
Warning: Replacing memory \m_axi_rresp with list of registers. See axilxbar.v:325
Warning: Replacing memory \m_axi_rdata with list of registers. See axilxbar.v:324
Warning: Replacing memory \m_axi_bresp with list of registers. See axilxbar.v:322
Note: Assuming pure combinatorial block at axilxbar.v:774.3-780.6 in
compliance with IEC 62142(E):2005 / IEEE Std. 1364.1(E):2002. Recommending
use of @* instead of @(...) for better match of synthesis and simulation.
Note: Assuming pure combinatorial block at axilxbar.v:922.3-928.6 in
compliance with IEC 62142(E):2005 / IEEE Std. 1364.1(E):2002. Recommending
use of @* instead of @(...) for better match of synthesis and simulation.
ERROR: Assert `modules_.count(module->name) == 0' failed in kernel/rtlil.cc:922.
```

I tried to debug yosys a bit and see that this $paramod$ ast->str is used in two calls to ast.cc process_module() and the second time causes the failure in RTLIL::Design::add().

The first time process_module() is called it is from AST::process(), it is then called again from AST_INTERNAL::process_and_replace_module() and since the ast->str is the same, the design ends up with two modules with the same name.

I'm not sure what the correct behaviour would be, but two options seem to be either delete the first module as the new one would replace it, or to fetch the first module and update it with the new ast.

KrystalDelusion, 2025-02-09:
It would be helpful if you could reduce this down to a smaller example specific to the error being raised, otherwise this seems like the problem is with how wb2axip is set up; nevermind that I can't even tell where Yosys is being called from when you run `make`, it looks like it should be calling `iverilog`.

j-f-d, 2025-02-10:
I was in touch with the author (who seems like a nice bloke) of wb2axip and he told me "The crossbar proofs haven't worked in a while.  Something changed in Yosys, and ever since the crossbars have crashed Yosys.  I'm not sure what, but then again ... I'm not a Yosys debugger." I don't know when yosys changed. 

(I've also notified the wb2axip author of this issue, so he might be in touch if he has anything to contribute.)

When the make failed, I redid the command which was failing with:
```
$ cd bench/formal/axilxbar_prf1x8/src
$ yosys -ql ../model/design.log ../model/design.ys
```

That still fails, but I don't know if the 'make' is producing anything along the way which this command depends upon.

In an attempt to debug yosys, I checked out the current sources from here and built that, the behaviour was the same. This is the version that I was debugging:
```
$ cd bench/formal/axilxbar_prf1x8/src
$  kdbg ../../../../../yosys/yosys -a "-ql ../model/design.log ../model/design.ys"
```

HTH.

j-f-d, 2025-02-10:
Also, yosys which I built from source is:
```
$ yosys --version
Yosys 0.49+3 (git sha1 9d4630484, clang++ 18.1.8 -Og -fPIC)
```

whitequark, 2025-02-10:
> I was in touch with the author (who seems like a nice bloke)

The author is a well known and outspoken transphobe and in my view as one of the Yosys maintainers we should not be assisting anyone with his work.

KrystalDelusion, 2025-02-10:
> That still fails, but I don't know if the 'make' is producing anything along the way which this command depends upon.

What you're looking at is a call to Yosys from inside SBY.  And if you were using an old version of Yosys I would guess that you're also using an old version of SBY.  If you can reproduce the error outside of zipcpu code it might reveal a problem with SBY and/or Yosys, in which case you can make a new issue.  Otherwise I would still guess that it's a problem with the setup, potentially only working in the past due to undefined behaviour which has since been fixed.

j-f-d, 2025-02-11:
> > I was in touch with the author (who seems like a nice bloke)
> 
> The author is a well known and outspoken transphobe and in my view as one of the Yosys maintainers we should not be assisting anyone with his work.

Very strange to base your response to a problem report on such an unrelated topic, there is nothing transphobic that I can see about this AXI to Wishbone interface. I'm sure that we all can find lots to disagree about, but it doesn't stop me trying to help make your project better.

whitequark, 2025-02-11:
It is directly related to YosysHQ as an organization. Furthermore, if that is your attitude then I am perfectly happy for your contributions to not happen.