YosysHQ/yosys #4888: abc not recognised as a submodule when using a different abc commit

KrystalDelusion, 2025-02-09:
### Version

0.48

### On which OS did this happen?

Linux

### Reproduction Steps

1. Checkout Yosys
2. Call `git submodule update --init --remote` or `git -C abc checkout 8700bb` or any other command which changes the git commit of abc
3. Call `make check-git-abc`

### Expected Behavior

Should exit normally (status code 0)

### Actual Behavior

Exits with 0 (status code 1) and prints error message:
```
Error: 'abc' is not configured as a git submodule.
To resolve this:
1. Back up your changes: Save any modifications from the 'abc' directory to another location.
2. Remove the existing 'abc' directory: Delete the 'abc' directory and all its contents.
3. Initialize the submodule: Run 'git submodule update --init' to set up 'abc' as a submodule.
4. Reapply your changes: Move your saved changes back to the 'abc' directory, if necessary.
```

KrystalDelusion, 2025-02-09:
The current `check-git-abc` target checks if the git submodule status begins with a space (using `git -C "$(YOSYS_SRC)" submodule status abc 2>/dev/null | grep -q '^ '`, so in the following YosysHQ/abc@cac8f99ea passes because that's the current commit pointed to in latest Yosys (fc88ea3), but then changing to the latest abc commit, YosysHQ/abc@8700bb58e, causes the git submodule stats to start with a '+' and thus fails the check.
```shell
$ git submodule status abc
 cac8f99eaa220a5e3db5caeb87cef0a975c953a2 abc (v0.49)
$ git -C abc checkout 8700bb
Previous HEAD position was cac8f99ea Revert "Merge pull request #247 from QuantamHD/abc_unit_tests"
HEAD is now at 8700bb58e Merge pull request #37 from YosysHQ/povik/fix-mfs-ub
$ git submodule status abc
+8700bb58ee5cc9039d86eb8ee3cd831d3dd1b19e abc (v0.49-2-g8700bb58e)
```

widlarizer, 2025-02-17:
This is intended behavior. It indicates a mismatch between the `abc` revision that's checked out in the submodule and the abc revision that should be checked out in the submodule as determined by the current `yosys` commit

```
$ git submodule status abc
 cac8f99eaa220a5e3db5caeb87cef0a975c953a2 abc (v0.50)
$ git -C abc checkout 8700bb
Previous HEAD position was cac8f99ea Revert "Merge pull request #247 from QuantamHD/abc_unit_tests"
HEAD is now at 8700bb58e Merge pull request #37 from YosysHQ/povik/fix-mfs-ub
$ git submodule status abc
+8700bb58ee5cc9039d86eb8ee3cd831d3dd1b19e abc (v0.50-2-g8700bb58e)
$ git add abc
$ git commit -m "bump abc"
[emil/share-limit-effort 238a35554] bump abc
 1 file changed, 1 insertion(+), 1 deletion(-)
$ git submodule status abc
 8700bb58ee5cc9039d86eb8ee3cd831d3dd1b19e abc (v0.50-2-g8700bb58e)
```

KrystalDelusion, 2025-02-17:
Ah okay, that does make some amount of sense.  Mayhaps rather than falling back on the 'not a submodule' error message we add an error message for when that is the case then?

widlarizer, 2025-02-17:
Good idea. From `man git submodule status`:
```
           Each SHA-1 will possibly be prefixed with - if the submodule is not initialized, + if the
           currently checked out submodule commit does not match the SHA-1 found in the index of the
           containing repository and U if the submodule has merge conflicts.
```

PR #4908 (2025-02-21): Cover all cases of submodule status in check-git-abc
_What are the reasons/motivation for this change?_

`git submodule status` can be prefixed with `[ -+U]`, of which we expect the space, and assume anything else is a `-`.  But seeing an error message that tells you abc is not initialized when it *is* initialized but pointing at a different commit is confusing.
Also closes #4888.

_Explain how this is achieved._

Handle the `[+U]` cases and give an actually related error message instead of the generic one.

_If applicable, please suggest to reviewers how they can test the change._
closed 2025-02-21 by KrystalDelusion