YosysHQ/yosys #4893: Continuous assign drives right-hand side

rantvm, 2025-02-11:
### Version

Yosys 0.49+29 (git sha1 ce5ad2a55, clang++ 18.1.8 -fPIC -O3)

### On which OS did this happen?

Linux

### Reproduction Steps

The following Verilog exhibits the unexpected behaviour. Note, in particular, that `top.first_delay` is only present on the right-hand side of a continuous assign.

```verilog
module top
(
  input wire clock,
  input wire signal,
  output wire first_delay,
  output wire second_delay
);

  wire delay;
  dff dff1(clock, signal, delay);
  dff dff2(clock, delay, second_delay);
  
  assign second_delay = first_delay;
endmodule

module dff
(
  input wire c,
  input wire d,
  output reg q
);

  always @(posedge c)
    q <= d;

endmodule
```

Then synthesize using `yosys -p 'read_verilog top.v; synth; show top'`

### Expected Behavior

`top.first_delay` should float. Diagnostic should be issued for `top.second_delay` being doubly-driven.

### Actual Behavior

`top.second_delay` is driven onto `top.first_delay`. No diagnostic is issued.

Synthesis result illustrated in the following diagram produced by `show -format svg top`. 

[image]

KrystalDelusion, 2025-03-21:
Calling `yosys -p 'hierarchy -top top; dump' top.v` with verific shows that the connection is the right way around
```
  cell $_BUF_ $verific$i4$top.v:13$7
    connect \Y \second_delay
    connect \A \first_delay
  end
```
And also with `-f verilog` gives
```
connect \second_delay \first_delay
```
but `yosys -p 'hierarchy -top top; opt; dump` flips it around:
```
connect \first_delay \second_delay
```

The doubly-driven not having an error is a) a separate issue, and b) much less likely to be resolved given that `read_verilog` is provided with the explicit caveat that it should only be given valid verilog because it doesn't have very good error handling or reporting.

Calling `verilator --lint-only top.v`
```
%Error-BLKANDNBLK: top.v:6:15: Unsupported: Blocked and non-blocking assignments to same variable: 'second_delay'
    6 |   output wire second_delay
      |               ^~~~~~~~~~~~
                   top.v:13:23: ... Location of blocking assignment
   13 |   assign second_delay = first_delay;
      |                       ^
                   top.v:25:7: ... Location of nonblocking assignment
   25 |     q <= d;
      |       ^~
                   ... For error description see https://verilator.org/warn/BLKANDNBLK?v=5.033
%Error: Exiting due to 1 error(s)
        ... See the manual at https://verilator.org/verilator_doc.html for more assistance.
```

If you're able to reproduce this without the doubly-driven signal then there might be something here, but as-is I think this is just invalid input and can be closed as wont-fix.

KrystalDelusion, 2025-03-22:
Minimized reproducer:
```
read_rtlil << EOF
module \top
  wire \first_delay
  wire \second_delay
  wire input 1 \signal
  connect \second_delay \signal
  connect \second_delay \first_delay
end
EOF
select -assert-none top/first_delay %c %ci %D
opt_clean
select -assert-none top/first_delay %c %ci %D
```
The first `select` passes, the second fails.  Either RTLIL connections are directionless, and can be reversed freely, in which case both selects should fail; or `opt_clean` is erroneously reversing the connection and both selects should pass.  I'm not actually sure which.

closed 2025-03-22 by KrystalDelusion