YosysHQ/yosys #4897: Problems with EDIF files when reading in vivado post-synthesis project

teodor961, 2025-02-13:
### Version

Yosys 0.49+21 (git sha1 ed5d9c868, g++ 11.4.0-1ubuntu1~22.04 -fPIC -O3)

### On which OS did this happen?

Linux

### Reproduction Steps

1. Created trivial example of one module inside another module - top_adder.v and adder.v

2. Open yosys and run the following commands:
  read_verilog top_adder.v adder.v
  synth_xilinx -arch xc7 -top top_adder -flatten -edif top_adder_xc7.edif

3. Create post-synthesis project in vivado
4. Import top_adder_xc7.edif
5. Try to open synthesized netliost

### Expected Behavior

Expect to open netlist view and a working design file.

### Actual Behavior

I get an error about an undefined cell:

ERROR: [EDIF 20-83] Cannot find cell 'id00045' view 'VIEW_NETLIST' in library '' for instance 'inst_adder' [/home/teodor/work/hobby/psk-mod-ip/yosys/top_adder_xc7.edif:22393]

Upon reviewing the EDIF file I can indeed see that this particular id00045 (changes for different test designs) is indeed referenced without being defined first.

I can send the source files: top_adder.v, adder.v and the edif file on request, but this issue occurs for all designs with any kind of hierarchy. Single module designs work fine and the edif files open successfully in vivado, however even the simplest two or more module designs fail with the same error described above.

Any support/guidance is very appreciated.

KrystalDelusion, 2025-03-12:
I think this is related to #4122; the call to `flatten` adds `$scopeinfo` cells, which are not synthesizable but not automatically removed during synthesis.  And since `write_edif` has no idea about `$scopeinfo` cells, it tries to reference them with e.g. `(viewRef VIEW_NETLIST (cellRef id00001))` but they never get defined.

I don't have Vivado to test, but this should be a minimal reproducer:
```
read_verilog <<EOF
module mod (...);
input a;
output b;
assign b = a;
endmodule

module top (...);
input a;
output b;
mod inst_mod (a, b);
endmodule
EOF

hierarchy -top top
flatten
# delete t:$scopeinfo
write_edif -pvector bra mods.edif
```

The `write_edif -pvector bra <file>` replicates the `-edif <file>` from `synth_xilinx`.  If you save that script as `edif.ys`, call `yosys edif.ys`, then try to load the `mods.edif` output into Vivado I suspect it will fail.  If you then uncomment the `delete t:$scopeinfo` line it should then work in Vivado.

You should also be able to try with your own files by dropping the `-edif <file>` from `synth_xilinx`, and calling the last two lines of this script instead.

teodor961, 2025-03-14:
Thank you for the feedback. The $scopeinfo cells were the issue. I've added the delete t:$scopeinfo command into the script and there are no more undefined cells in the design.