YosysHQ/yosys #4905: synth_gatemate output netlist incompatible with Gatemate PnR

tarik-ibrahimovic, 2025-02-14:
### Version

Yosys 0.42+15 (git sha1 dd9624444, g++ 11.4.0-1ubuntu1~22.04 -fPIC -Os)

### On which OS did this happen?

Linux

### Reproduction Steps

Head over [here](https://github.com/chili-chips-ba/openCologne/tree/main/8.StressTest/1.corescore_cc) and follow the build steps. The issue is highlighted in the build steps.

### Expected Behavior

The netlist shouldn't contain anything other than Cologne primitives and synthesizable elements. Take a look at this [issue](https://github.com/chili-chips-ba/openCologne/issues/54) for a more detailed explanation.

### Actual Behavior

Instead the netlist contains an initial block employing a `$write` integrated task, unsupported by PnR, requiring manual modifications of the netlist.
```
 initial begin
    if (1'h1) begin
      $write("Preloading $paramod$76ca5ad6cddf78bc518e7753a546bede55c5cafa\\serving_ram from %s\n", $unsigned(88'h636f72655f34362e686578));
    end
    if (1'h1) begin
      $write("Preloading $paramod$ea5542b74da5d3df7ad0239dce78e365096fcc62\\serving_ram from %s\n", $unsigned(88'h636f72655f33342e686578));
    end
    if (1'h1) begin
      $write("Preloading $paramod$f9cf07300b5f1f0ca052c9602528ae8d7d785ee0\\serving_ram from %s\n",
...
```

whitequark, 2025-02-14:
This is caused by the following non-synthesizable directives being present:

https://github.com/chili-chips-ba/openCologne/blob/3576c90a609d8588a6168e7acce99136205cbd10/8.StressTest/4.veerwolves/rtl/dpram64.v#L59-L62

In other words, you get what you ask for.

The solution is to surround them by `` `ifndef SYNTHESIS ``.

tarik-ibrahimovic, 2025-02-14:
This isn't the repository in question. Please head over to the above linked `8.StressTest/1.corescore_cc`

KrystalDelusion, 2025-02-14:
> This isn't the repository in question.

You opened the issue here, thus it is being responded to here.

whitequark, 2025-02-14:
> This isn't the repository in question. Please head over to the above linked `8.StressTest/1.corescore_cc`

Ah, sure. I'm not sure exactly how you end up with a non-synthesizable construct in case of `1.corescore_cc` but the problem and the solution is the same; thus you should be able to figure out the rest yourself.

tarik-ibrahimovic, 2025-02-14:
Thanks, found it now, it's originating from Serv's repo. So no options from yosys' side to skip these kind of statements?

KrystalDelusion, 2025-02-14:
I think you could call something like `del t:$write`, but otherwise [no](https://github.com/YosysHQ/yosys/issues/4122)

tarik-ibrahimovic, 2025-02-14:
Thank you. I am now closing this issue.

closed 2025-02-14 by tarik-ibrahimovic