YosysHQ/yosys #4913: Optimization depends on ordering of registers (if flattening is enabled)

jurriaan, 2025-02-23:
### Version

Yosys 0.50+7 (git sha1 38f858374, g++ 14.2.1 -march=x86-64 -mtune=generic -O2 -fno-plt -fexceptions -fstack-clash-protection -fcf-protection -fPIC -O3)

### On which OS did this happen?

Linux

### Reproduction Steps

See https://github.com/jurriaan/shift-register-expander-experiment

Changing the ordering of the registers in https://github.com/jurriaan/shift-register-expander-experiment/blob/main/shift_register_expander.sv changes the amount of cells used quite significantly. 

Just swapping one of the register definitions gives a change of almost 70 cells (In another project it was a difference of almost 300 cells by swapping a few registers around).

I noticed that this only happens when flattening is enabled, with `-noflatten` this does not happen.

### Expected Behavior

Ordering does not influence synthesis this significantly. I understood from https://github.com/YosysHQ/yosys/issues/3713 that yosys can be a bit chaotic in this regard, but I hope this is some kind of bug that can be fixed :)

### Actual Behavior

In this simple experiment this is what happens:

```verilog
  // This is the least optimal setup (590 total, 125 just for this module with -noflatten):
  logic shift_enable_flag;  
  logic shift_clk_internal;
  logic [NUM_BITS-1:0] shift_reg; 
  logic latching;
  logic [$clog2(NUM_BITS)-1:0] bit_counter;
```

```
=== top ===

   Number of wires:                516
   Number of wire bits:           1045
   Number of public wires:         516
   Number of public wire bits:    1045
   Number of ports:                  5
   Number of port bits:              5
   Number of memories:               0
   Number of memory bits:            0
   Number of processes:              0
   Number of cells:                590
     $scopeinfo                      5
     ALU                            31
     DFFE                           23
     DFFR                           29
     DFFRE                          38
     DFFSE                          21
     GND                             1
     IBUF                            2
     LUT1                           48
     LUT2                           38
     LUT3                           44
     LUT4                          162
     MUX2_LUT5                      91
     MUX2_LUT6                      33
     MUX2_LUT7                      15
     MUX2_LUT8                       5
     OBUF                            3
     VCC                             1
```

v.s.

```verilog
  // This results in way better synthesis (519 total, 125 just for this module with -noflatten):
  logic shift_enable_flag;  
  logic latching;
  logic [NUM_BITS-1:0] shift_reg;  
  logic shift_clk_internal;
  logic [$clog2(NUM_BITS)-1:0] bit_counter;
```

```
=== top ===

   Number of wires:                441
   Number of wire bits:            959
   Number of public wires:         441
   Number of public wire bits:     959
   Number of ports:                  5
   Number of port bits:              5
   Number of memories:               0
   Number of memory bits:            0
   Number of processes:              0
   Number of cells:                519
     $scopeinfo                      5
     ALU                            31
     DFFE                           23
     DFFR                           29
     DFFRE                          38
     DFFSE                          21
     GND                             1
     IBUF                            2
     LUT1                           39
     LUT2                           32
     LUT3                           42
     LUT4                          145
     MUX2_LUT5                      72
     MUX2_LUT6                      24
     MUX2_LUT7                       9
     MUX2_LUT8                       2
     OBUF                            3
     VCC                             1
```

KrystalDelusion, 2025-02-23:
I think that while this isn't strictly speaking a duplicate of #3713, they are both symptoms of the same underlying issue; ABC is highly chaotic and even seemingly irrelevant changes in input can have a vastly different output result.  And similarly to that issue, the results can be changed (and in some cases improved) by calling `rename -scramble-name;` before synth.  Locally (with a slightly older version of Yosys, 0.48+51) I get 572 and 498 cells respectively for your "least optimal" and "way better" orderings respectively, but using `-scramble-name` with the default seed I get 405 and 450 respectively  Without looking into it too deeply, I believe `flatten` renames objects to include their source line number, which may explain why `-noflatten` removes the order dependency.

I'm 50:50 on whether there is enough of a difference here to justify a separate issue, but I'll leave it for now unless someone else decides to close this as a duplicate.

jurriaan, 2025-02-24:
@KrystalDelusion Thanks for looking into this! 

I was using this shift expander module in a bit bigger setup, and there the difference was a bit higher (300 cells), it's interesting how it depends on other connected modules. 
But I also fully understand that this seems like a minor issue (calling one of the solutions 'way better' is a bit much indeed 😅 ), but on FPGAs with 1k LUTs it can mean that your design does not fit. 

I will include the scramble-names step locally to improve things a tiny bit for now.

KrystalDelusion, 2025-02-24:
I didn't mean to call out saying it was "way better," I was simply trying to distinguish between the two; I am well aware that a few hundred cells can be quite significant.

jurriaan, 2025-02-24:
I also just noticed that the same thing can happen in an `always_ff` block. 

I was just implementing a reset condition, and noticed that for some ordering it synthesizes with 100 cells more than others, even with `-scramble-name` enabled.

wjrforcyber, 2025-03-28:
Hi there!
I saw this issue tagged with ABC so I take a look.
First of all, have you checked the equivalence?
Secondly, if the equivalence is checked, then there's a tip about ABC:  If you change the order of PI/PO, I could assure you the result will change, since the optimisation on combinational logic is topological, after collecting it will have different order of nodes(which means different order of local process), even the `cec`(combinational equivalence check) passes, the result is quite different, sometimes the gap is quite big, this is an ancient problem, it's just the nature result of the "topological traversal and optimisation". If you could make sure the order of the PI/PO(CI/CO) of the combinational part is the same after the change, then after the same optimisation the result will definitely be the same.
So, I saw the sequential part is the same in both report, I am guessing that the order of PI/PO(CI/CO) of the combinational part is changed after you swap the reg? Could you please verify it?