YosysHQ/yosys #4915: yosys synth_ice40 infers incorrect number of BRAMS for some memory sizes

johnwinans, 2025-02-25:
### Version

Yosys 0.30+38 (git sha1 2310a0ea9, clang 10.0.0-4ubuntu1 -fPIC -Os)

### On which OS did this happen?

Linux

### Reproduction Steps

Compile this with: `yosys -p "synth_ice40 -top top -json top.json" top.v`

```
module top (
    input                               clk,
    input wire [16:0]                   addr1,
    input wire [16:0]                   addr2,
    input wire [7:0]                    data1,
    output reg [7:0]                    data2
    );

    // Fail : NUM_BRAMS(Actual BRAMS) = 23(24), 27(28), 31(32)
    //      : All other values of NUM_BRAMS between 1 => 32 pass/match.
    localparam NUM_BRAMS = 27;

    reg [7:0] mem1 [0:(512*NUM_BRAMS)-1];

    always @(posedge clk) begin
        mem1[addr1] <= data1;
        data2 <= mem1[addr2];
    end

endmodule
```

Try setting NUM_BRAMS to 23, 27, or 31 and see an extra mystery BRAM is consumed.
Others I am collaborating with have reported identical failures on versions:

Yosys 0.49+17 (git sha1 f2d96cf61, g++ 12.2.0-14 -fPIC -O3)  [WSL2 Debian Bookworm]
Yosys 0.23 (git sha1 7ce5011c24b)  [on Debian GNU/Linux 12 (bookworm) Kernel: Linux 6.6.74+rpt-rpi-v8]

This problem does NOT appear to be an issue on yosys version:

Yosys 0.9 (git sha1 1979e0b)  [raspberry pi release: Debian GNU/Linux 11 (bullseye)]

### Expected Behavior

Expecting the number of BRAMs to be correlated to the inferred memory size.

### Actual Behavior

One extra BRAM appears to get allocated for some certain specific sizes (see example code.)

KrystalDelusion, 2025-02-25:
Without getting too deep into this, it looks like `memory_libmap` is choosing to reduce the cost of control logic by adding an extra BRAM.  I would've expected `debug memory_libmap` to show the choice between 23/24 BRAMs with/without the additional control when your NUM_BRAMS = 23, but it seems to just skip it...  But basically memory geometry is more complex than just "what is the minimum number of BRAMs are needed for this block of memory".

If you run through NUM_BRAMS for 15-24 these are the resource counts:

```
NUM_BRAMS        15   16   17   18   19   20   21   22   23   24
SB_DFF           34   32   36   35   36   34   36   35   34   34
SB_LUT4         136   40  180  103  188   64  210  123   79   79
SB_RAM40_4K      15   16   17   18   19   20   21   22   24   24
libmap cost    1037 1052 1174 1202 1311 1312 1448 1467 1573 1573
```

You can kind of see from 15/19 compared to 16/20 that the libmap cost for the extra control logic (implemented largely in LUTs) is growing faster than the cost for adding an extra BRAM.  So by the time it reaches 23 the extra BRAM is 'cheaper' than the extra control that would be needed.

I'm not sure if there is anyway to do it from within Yosys, but if you change the `cost` line in the `.../share/yosys/ice40/brams.txt` to e.g. 128 instead of 64 then NUM_BRAMS=23 does use 23 BRAMs.

KrystalDelusion, 2025-02-25:
So I don't think this is a bug or an error, although it might be nice to have a way to adjust memory costs in `memory_libmap` without having to do something like

```
read_verilog <input_files>
synth -top top -run :map_ram
memory_libmap -lib custom_brams.txt
synth_ice40 -run map_ram:
```

But that is an option if you need it (using 36/228/23 DFF/LUT/BRAM)

closed 2025-02-25 by KrystalDelusion