YosysHQ/yosys #4934: Lage memory RTLIL creation core dump

alaindargelas, 2025-03-12:
### Version

latest

### On which OS did this happen?

Linux

### Reproduction Steps

tar xvzf ariane_mem.tar.gz
cd ariane_mem
yosys -s yosys.ys

[ariane_mem.tar.gz](https://github.com/user-attachments/files/19215063/ariane_mem.tar.gz)

....
4.1. Executing AST frontend in derive mode using pre-parsed AST for module `\SyncSpRamBeNx64'.
Parameter \ADDR_WIDTH = 25
Parameter \DATA_DEPTH = 33554432
Parameter \OUT_REGS = 0
Parameter \SIM_INIT = 1
Generating RTLIL representation for module `$paramod$82e77b9edf67d2b61c69d65d8ee3287fbf448e1e\SyncSpRamBeNx64'.
Killed

### Expected Behavior

Create a correct netlist

### Actual Behavior

Core dump

marzoul, 2025-03-14:
@alaindargelas I tested your design with the slang plugin for yosys as described in #4601
It works but your depth (2^25 = 33 million) is still a bit too high Yosys it seems...

Edit your top-level with:
```
SyncSpRamBeNx64 #(.ADDR_WIDTH(ADDR_WIDTH),.DATA_DEPTH(1 << ADDR_WIDTH),.OUT_REGS(0),.SIM_INIT(1)) mem  (.*);
```
Your Yosys script:
```
read_slang dut.sv SyncSpRamBeNx64.sv --top dut
```
And the command-line:
```
yosys -m slang -s yosys.ys
```
My observations on my laptop, with different values for parameter ADDR_WIDTH:
```
ADDR_WIDTH=14  ->  0.16 sec, 34 MB RAM
ADDR_WIDTH=15  ->  0.64 sec, 36 MB RAM
ADDR_WIDTH=16  ->  2.70 sec, 40 MB RAM
```
So there is no longer a crash due to excessive amount of RAM used.
But each additional bit of address makes run time 4x and RAM usage +4MB
For 25 bits you could expect `2.7 * 4^(25-16) = 707k sec = 8 days`, still an issue but a different one and maybe this one has a solution ;-)

Note that the slang plugin is by default more strict with the language, but it will give you much better error/warning/diagnose messages.

povik, 2025-03-14:
> But each additional bit of address makes run time 4x and RAM usage +4MB

Interesting, so it's quadratic. I've confirmed this is due to the basic memory access; initialization is disabled via the comment directives. This is worth opening a ticket for.

QuantamHD, 2025-03-14:
A tool I've used in the past to look into Yosys memory issues is tcmalloc based heap profiling. Might help narrow down these issues faster. 

https://postbits.de/heap-profiling-with-tcmalloc.html

alaindargelas, 2025-03-17:
Hi all, thanks, I can close this issue. There is indeed an issue with the parameter size of this testcase.
I added a size limit check to our Yosys fork, which errors out for large memories:
https://github.com/Silimate/yosys/pull/71/files 
Verific issues a similar message when verific -cfg veri_extract_multiport_rams 0, but in this case, we have it set to 1, and Yosys receives that large memory size from Verific. The new error message prevents the crash and reports the location, up to the user to limit his/her memory element size.

closed 2025-03-17 by alaindargelas