YosysHQ/yosys #4936: $finish raises an unexpected error when testing variable within "initial begin end"

krynentechnology, 2025-03-14:
### Version

Yosys 0.44+9 (git sha1 4b9f45273, x86_64-w64-mingw32-g++ 13.2.1 -O3)

### On which OS did this happen?

Windows

### Reproduction Steps

yosys.exe -p "synth_ecp5 -abc9 -json test.json" test.v

test.v:

module test #( parameter [0:0] TEST = 0 );
reg test_bit = 0;
initial begin
    if ( TEST ) begin
        $finish;
    end
    test_bit = 0;
    if ( test_bit ) begin
        $finish;
    end
end
endmodule

### Expected Behavior

No error. Initial value of test_bit is tested and should not raise an unexpected error when test_bit = 0. When TEST = 1 it raises an expected error.

### Actual Behavior

2.3.1. Executing AST frontend in derive mode using pre-parsed AST for module `\test'.
Generating RTLIL representation for module `\test'.
test.v:9: ERROR: System task `$finish' executed.

KrystalDelusion, 2025-03-17:
I think what's happening here is that the AST frontend doesn't check the value of `test_bit` because it isn't a constant (even though within the initial block it *is* always zero).  We can see this in the following too:
```verilog
module test();
  reg test_bit = 0;
  initial begin
    test_bit = 0;
    if ( test_bit ) begin
      $display(test_bit);
    end
  end
endmodule
```
```
test.v:6: Info: Skipping system task `$display' with non-constant argument at position 1.
```

And using `read_verilog -no_dump_ptr -dump_ast1 test.v` we see this in the AST dump:

```
AST_CASE <test.v:5.5-7.8>
  AST_REDUCE_BOOL <test.v:0.0-0.0>
    AST_IDENTIFIER <test.v:5.10-5.18> str='\test_bit'
  AST_COND <test.v:5.10-5.18>
    AST_CONSTANT <test.v:0.0-0.0> bits='1'(1) range=[0:0] int=1
    AST_BLOCK <test.v:5.20-7.8>
      AST_BLOCK <test.v:5.21-7.8>
        AST_TCALL <test.v:6.7-6.26> str='$display'
          AST_IDENTIFIER <test.v:6.16-6.24> str='\test_bit'
```

So even though the AST says it should be checking the value of `test_bit`, it only actually uses the value to simplify the AST_CASE if the thing being checked is a constant (and only has 0 or 1 bits from what I can tell), and because it _doesn't_ skip the true case, it tries to simplify it, finding and executing the AST_TCALL.

krynentechnology, 2025-03-18:
I encountered this issue by a Yosys synthesis of [github.com/krynentechnology/sr2cb/lib/randomizer.v](https://github.com/krynentechnology/sr2cb/blob/main/lib/randomizer.v). I had to commented out line 190 - $finish - which checks a variable (register) initialization. $finish is a simulation control system task and ignored by Vivado, Quartus and Diamond synthesis (I test my published HDL modules on different simulation and synthesis platforms). In my opinion, the Yosys synthesis AST frontend should ignore $finish when it test a constant or a variable (register) and the test is false.

KrystalDelusion, 2025-03-29:
Loading `simlib.v` without `SIMLIB_NOCHECKS` defined also fails with `ERROR: System task `$stop' outside initial block is unsupported.`

```systemverilog
module \$assert (A, EN);

input A, EN;

`ifndef SIMLIB_NOCHECKS
always @* begin
	if (A !== 1'b1 && EN === 1'b1) begin
		$display("Assertion %m failed!");
		$stop;
	end
end
`endif

endmodule
```