YosysHQ/yosys #4947: Running the splitcells command causes a crash

FSY369, 2025-03-19:
### Version

 Yosys 0.48+47 (git sha1 cbb95cb51, clang++ 18.1.8 -fPIC -O3)

### On which OS did this happen?

Linux

### Reproduction Steps

Run the following command on rtl.v
```
 read_verilog rtl.v  
           prep
           splitcells
```

### Expected Behavior

Run successfully or report an error

### Actual Behavior

2.6. Executing CHECK pass (checking for obvious problems).
Checking module module12...
Checking module module124...
Checking module module20...
Checking module module86...
Checking module module92...
Checking module top...
Floating point exception (core dumped)

povik, 2025-03-19:
Likely duplicate of #4844 fixed in v0.50. Please reopen if you can reproduce with a newer version.

FSY369, 2025-03-20:
Ok, I ran it once over on this version and it still crashes!
`Yosys 0.50+7 (git sha1 38f858374, clang++ 14.0.0-1ubuntu1.1 -fPIC -O3)`

KrystalDelusion, 2025-03-20:
Minimized to:
```
read_rtlil << EOF
# Generated by Yosys 0.50 (git sha1 b5170e139, ccache clang++ 18.1.8 -fPIC -O3)
autoidx 1714
attribute \src "rtl.v:1.1-294.10"
attribute \top 1
module \top
  parameter \param216 1'0
  attribute $bugpoint 1
  wire output 1 $auto$bugpoint.cc:258:simplify_something$1699
  attribute \src "rtl.v:195.24-198.53"
  cell $shl $shl$rtl.v:195$98
    parameter \A_SIGNED 1
    parameter \A_WIDTH 1
    parameter \B_SIGNED 0
    parameter \B_WIDTH 49
    parameter \Y_WIDTH 1
    connect \A 1'x
    connect \B 49'x
    connect \Y $auto$bugpoint.cc:258:simplify_something$1699
  end
end
EOF
check
```
by running your rtl.v through this script:
```
read_verilog rtl.v
prep
bugpoint -command check -clean
write_rtlil min.il
```

Note that I only do `hierarchy` and `proc` because it crashes on the `check` inside `prep`; it never even makes it to `splitcells`.  Next time you upload a fuzzer generated issue without minimizing it or running it through `bugpoint` I will automatically close it.  If you need help with running `bugpoint`, post a question to the slack, or open a question in the github discussions.  Don't make an issue.

FSY369, 2025-03-20:
OK very sorry, I may have to learn how to minimize code, I am very sorry to bring you trouble, sincerely ask you bugpoint is yosys comes with it, how can I operate to achieve the code minimization, directly run the script you gave me whether it is OK, do I need to do extra operation?

KrystalDelusion, 2025-03-20:
There is a tutorial for `bugpoint` in development, a [preview of it is available](https://yosyshq.readthedocs.io/projects/yosys/en/docs-preview-bugpoint/using_yosys/bugpoint.html), but as of right now it is still in bullet point format so not the most readable.  The command itself is built in to Yosys, and has [help text available](https://yosyshq.readthedocs.io/projects/yosys/en/latest/cmd/bugpoint.html).

FSY369, 2025-03-20:
Okay. Thank you very much.

georgerennie, 2025-03-20:
Interestingly I have a `g++` build which didn't reproduce this.

The issue is in [celledges.cc:256]( https://github.com/YosysHQ/yosys/blame/f92ad7d900af8d9095cf91abd5d0dfce8ea42a3d/kernel/celledges.cc#L256).

```c++
for (int k = 0; k < b_width; k++) {
	// left shifts
	if (cell->type.in(ID($shl), ID($sshl))) {
		if (a_width == 1 && is_signed) {
			int skip = 1 << (k + 1);
			int base = skip -1;
			if (i % skip != base && i - a_width + 2 < 1 << b_width_capped)
				db->add_edge(cell, ID::B, k, ID::Y, i, -1);	
```

It looks like what is happening is that `k + 1` becomes larger than the width of `int` so `skip` overflows to 0. This means that `i % skip` is remainder with a `0` divisor which is UB and causes the exception.

georgerennie, 2025-03-20:
This is now definitely a duplicate of #4844 so I will close this and reopen that given we were already tracking UB in that patch of code there.

closed 2025-03-20 by georgerennie

KrystalDelusion, 2025-03-20:
Thanks George!