YosysHQ/yosys #4957: Ternary Expressions Cause Assertion Failure

FSY369, 2025-03-24:
### Version

Yosys 0.50+7 (git sha1 38f858374, clang++ 14.0.0-1ubuntu1.1 -fPIC -O3)

### On which OS did this happen?

Linux

### Reproduction Steps

I encountered an assertion failure in Yosys while processing a Verilog design. I used bugpoint to reduce the code to a minimal test case.

This failure can be reproduced with the following script:
```
read_verilog 3510.v
prep
splitcells
```
Minimisation Code:
```
# Generated by Yosys 0.50+7 (git sha1 38f858374, clang++ 14.0.0-1ubuntu1.1 -fPIC -O3)
autoidx 3064
attribute \src "rtl.v:874.1-1178.10"
module \module122
  attribute $bugpoint 1
  wire output 1 $auto$bugpoint.cc:258:simplify_something$3056
  attribute \src "rtl.v:994.42-996.47"
  cell $sshl $sshl$rtl.v:994$755
    parameter \A_SIGNED 1
    parameter \A_WIDTH 1
    parameter \B_SIGNED 0
    parameter \B_WIDTH 35
    parameter \Y_WIDTH 1
    connect \A 1'x
    connect \B 35'x
    connect \Y $auto$bugpoint.cc:258:simplify_something$3056
  end
end
```

### Expected Behavior

The command should not cause an assert failure.

### Actual Behavior

3. Executing SPLITCELLS pass (splitting up multi-bit cells).
Splitting $dff cell top/$procdff$38 into 2 slices:
  slice 1: $procdff$38[1:0] => \reg127 [1:0]
  slice 2: $procdff$38[2] => \reg127 [2]
Splitting $xnor cell top/$xnor$rtl.v:25$28 into 2 slices:
  slice 1: $xnor$rtl.v:25$28[1:0] => $0\reg127[3:0] [1:0]
ERROR: Assert `length >= 0' failed in kernel/rtlil.cc:4880.

georgerennie, 2025-03-24:
That minimized code doesn't look like what you actually are running, given splitcells is showing $xnor and $dff cells but it just contains $sshl, so I cannot reproduce this

georgerennie, 2025-03-24:
But even so this is obviously a duplicate of #4909 (you even both refer to it as `3510.v`, have you used the same seed?)

closed 2025-03-24 by georgerennie

KrystalDelusion, 2025-03-24:
@FSY369 please do not report any new issues unless you can identify:

1. where in the Yosys source code they are being raised and why; and
2. there is not an existing issue for it.

This is your last warning.

FSY369, 2025-03-25:
Okay, I'm very sorry, I'll take note of the request you gave, I noticed that it might be the same problem as #4909, sorry again for the intrusion!