YosysHQ/yosys #4958: Gowin BRAMS mapping using more DPB's than expected in frame buffer

loco-choco, 2025-03-24:
### Version

Yosys 0.51 (git sha1 c4b5190229616f7ebf8197f43990b4429de3e420, g++ 14.2.1 -fPIC -O3)

### On which OS did this happen?

Linux

### Reproduction Steps

1. Get the following Verilog file, which creates a frame buffer.
```verilog
module dpb_issue_example #(
    parameter RESOLUTION_X = 120,
    parameter RESOLUTION_Y = 68,
    localparam RES_X_WIDTH = $clog2(RESOLUTION_X),
    localparam RES_Y_WIDTH = $clog2(RESOLUTION_Y),
    parameter R_PIXEL_WIDTH = 5,
    parameter G_PIXEL_WIDTH = 6,
    parameter B_PIXEL_WIDTH = 5,
    localparam RGB_PIXEL_WIDTH = R_PIXEL_WIDTH + G_PIXEL_WIDTH + B_PIXEL_WIDTH,
) (
    input clk_in,
    input nread_write,
    input [RES_X_WIDTH-1:0] pixel_position_x,
    input [RES_Y_WIDTH-1:0] pixel_position_y,
    input [R_PIXEL_WIDTH-1:0] in_pixel_color_r,
    input [G_PIXEL_WIDTH-1:0] in_pixel_color_g,
    input [B_PIXEL_WIDTH-1:0] in_pixel_color_b,
    output [R_PIXEL_WIDTH-1:0] out_pixel_color_r,
    output [G_PIXEL_WIDTH-1:0] out_pixel_color_g,
    output [B_PIXEL_WIDTH-1:0] out_pixel_color_b
);
  reg[RGB_PIXEL_WIDTH-1:0] current_pixel_value;
  assign out_pixel_color_r = current_pixel_value[R_PIXEL_WIDTH-1:0];
  assign out_pixel_color_g = current_pixel_value[G_PIXEL_WIDTH-1+R_PIXEL_WIDTH:R_PIXEL_WIDTH];
  assign out_pixel_color_b = current_pixel_value[B_PIXEL_WIDTH-1+R_PIXEL_WIDTH+G_PIXEL_WIDTH
                                                :R_PIXEL_WIDTH+G_PIXEL_WIDTH];
  
  reg [RGB_PIXEL_WIDTH-1:0] frame_buffer[RESOLUTION_X][RESOLUTION_Y];

  always @(posedge (clk_in)) begin
    // Frame Buffer Acess Logic
    current_pixel_value <= 0;
    if(nread_write) begin
      frame_buffer[pixel_position_x][pixel_position_y] <= {in_pixel_color_b,
                                                                 in_pixel_color_g,
                                                                 in_pixel_color_r};
    end else begin
      // Output current pixel
      current_pixel_value <= frame_buffer[pixel_position_x][pixel_position_y];
    end
  end
  // ------------------------------------------------------------------------------------------
endmodule
```
2. Run the command `yosys -p "read_verilog -sv dpb_issue_example.v ; synth_gowin -top dpb_issue_example"`, and see that it uses 8 SP's.
3.  Run the command `yosys -p "read_verilog -sv dpb_issue_example.v ; synth_gowin -top dpb_issue_example" | grep Warning`, and see the multiple warnings related a out of bounds selection in `yosys/gowin/brams_map.v`.

### Expected Behavior

With a BRAMS (SP) being able to store 16k words of 32bits in the gowin primitives (https://cdn.gowinsemi.com.cn/UG285E.pdf), I expected that a matrix of 120x68=8160 pixels, with each pixel using 16bits, would only need a Single Port BRAMS.

### Actual Behavior

The mapper ends up using more BRAMS than expected:
```bash

2.30. Printing statistics.

=== dpb_issue_example ===

   Number of wires:                 67
   Number of wire bits:            407
   Number of public wires:          67
   Number of public wire bits:     407
   Number of ports:                 10
   Number of port bits:             48
   Number of memories:               0
   Number of memory bits:            0
   Number of processes:              0
   Number of cells:                 86
     ALU                            22
     GND                             1
     IBUF                           32
     LUT2                            6
     OBUF                           16
     SP                              8
     VCC                             1

2.31. Executing CHECK pass (checking for obvious problems).
Checking module dpb_issue_example...
Found and reported 0 problems.

Warnings: 4 unique messages, 4 total
End of script. Logfile hash: 0d7176a733, CPU: user 0.47s system 0.01s, MEM: 58.82 MB peak
Yosys 0.51 (git sha1 c4b5190229616f7ebf8197f43990b4429de3e420, g++ 14.2.1 -fPIC -O3)
Time spent: 35% 8x techmap (0 sec), 16% 20x read_verilog (0 sec), ...
```
And outputs several warnings related to out of range bit mapping.
```bash
> yosys -p "read_verilog -sv dpb_issue_example.v ; synth_gowin -top dpb_issue_example" | grep Warning
/nix/store/i167d4cjdxv18f9l8x41xhd0h3yhfaf3-yosys-0.51/bin/../share/yosys/gowin/brams_map.v:112: Warning: Range [7:0] select out of bounds on signal `\PORT_A_WR_DATA': Setting 6 MSB bits to undef.
/nix/store/i167d4cjdxv18f9l8x41xhd0h3yhfaf3-yosys-0.51/bin/../share/yosys/gowin/brams_map.v:112: Warning: Range select [16:9] out of bounds on signal `\PORT_A_WR_DATA': Setting all 8 result bits to undef.
/nix/store/i167d4cjdxv18f9l8x41xhd0h3yhfaf3-yosys-0.51/bin/../share/yosys/gowin/brams_map.v:112: Warning: Range select [25:18] out of bounds on signal `\PORT_A_WR_DATA': Setting all 8 result bits to undef.
/nix/store/i167d4cjdxv18f9l8x41xhd0h3yhfaf3-yosys-0.51/bin/../share/yosys/gowin/brams_map.v:112: Warning: Range select [34:27] out of bounds on signal `\PORT_A_WR_DATA': Setting all 8 result bits to undef.
ABC: Warning: The network is combinational.
Warnings: 4 unique messages, 4 total
```

Here is the full log: [log.txt](https://github.com/user-attachments/files/19413830/log.txt)

KrystalDelusion, 2025-03-24:
That's 16k *bits* not 16k *words*.  Using the max width of 32 bits reduces the address depth to 9 bits, i.e. 512 words of 32 bit.  8160 pixels by 16bits per pixel is in fact just shy of the maximum memory available in 8x16k block rams.

The out of range bit mapping is because `memory_libmap` has set `Parameter \PORT_A_WIDTH = 2`, meaning that each memory address corresponds to 2 bits (such that 2 bits from 8 memory blocks gives a full 16 bit pixel), but the way the code is structured it is trying to read `x8_wr_data(data) {data[34:27], data[25:18], data[16:9], data[7:0]}` of an input signal of width 2.  While the warnings there aren't ideal, they aren't unexpected and reasonable in context.

So yeah, there's no actual issue here, just a misinterpretation of the size of the BRAM.

closed 2025-03-24 by KrystalDelusion

loco-choco, 2025-03-25:
Ah, sorry, that's fair. Thanks for the explanation!

KrystalDelusion, 2025-03-25:
No problem :)