YosysHQ/yosys #4968: Can't build from 0.51 tarball

spth, 2025-03-28:
### Version

0.51

### On which OS did this happen?

Linux

### Reproduction Steps

Same problem as https://github.com/YosysHQ/yosys/issues/4470, which was fixed in 0.47, but apparently is back in 0.51. Could you put some test in place for future releases, that check, that release source tarballs are actually buildable?

### Expected Behavior

yosys builds (like it does for me when using the 0.40, 0.42 and 0.47 tarballs).

### Actual Behavior

```
philipp@alecto:~/yosys-0.51$ make
[Makefile.conf] CONFIG := gcc
Initialize the submodule: Run 'git submodule update --init' to set up 'abc' as a submodule.
make: *** [Makefile:780: check-git-abc] Fehler 1
```

```
philipp@alecto:~/yosys-0.51$ LANG=C git submodule update --init --recursive
fatal: not a git repository (or any of the parent directories): .git
```

KrystalDelusion, 2025-03-28:
I just tried and it seems to be working fine?
```
dawn@holly:~/Downloads$ tar -xzf yosys.tar.gz -C yosys-0.51
dawn@holly:~/Downloads$ cd yosys-0.51
dawn@holly:~/Downloads/yosys-0.51$ make check-git-abc
'abc' comes from a tarball. Continuing.
```

It's the `check-git-abc` make target that gives the `"Initialize the submodule: Run 'git submodule update --init' to set up 'abc' as a submodule."` error, which should only happen if:
1. the `abc` directory exists,
2. `git submodule status` fails, and
3. there is no `.gitcommit` file in the `abc` directory.

If there *is* a `abc/.gitcommit`, and it contains `$Format:%H$`, then ABC comes from git, but `git submodule status` has failed so it can't be a submodule and it displays `"Error: 'abc' is not configured as a git submodule."`.

If `abc/.gitcommit` contains anything else, say `f2d68d590fa6f8fc32295a2edd79afc0d14a1414` like it does in the 0.51 release .tar.gz, then it gives the message `"'abc' comes from a tarball. Continuing."`, which is what it displays in the test I just did.

Are hidden files somehow being ignored when you extract the archive?

KrystalDelusion, 2025-03-28:
Oh, unless you downloaded the "Source code (tar.gz)" instead of the yosys.tar.gz file, which isn't a release tarball, it's just the source code.

spth, 2025-03-28:
So I guess I just downloaded the wrong tarball? Sorry for the noise, then.

closed 2025-03-28 by spth

KrystalDelusion, 2025-03-28:
Not at all, the fact that `check-git-abc` assumes that `yosys` is a git repo isn't exactly ideal.  We should be prepared for people to download the source code sans git and be able to give an appropriate message; I've just opened #4969 for that exact reason.