YosysHQ/yosys #4972: An infinite loop occurred in the log during the `synth_greenpak4` process.

sdjasj, 2025-03-31:
### Version

Yosys 0.50+49 (git sha1 05c81b3f1, clang++ 14.0.0-1ubuntu1.1 -Og -fPIC -O1 -fno-omit-frame-pointer -fno-optimize-sibling-calls -fsanitize=address,undefined)

### On which OS did this happen?

Linux

### Reproduction Steps

The following is the problematic Verilog code file bug.v, which I have minimized as much as possible out of responsibility:
```verilog
module a(input b, c, output reg d);
  reg [31:0] e;

  always @(*) begin
    if (b >> e) begin
        d = e;
    end

    if (c - c) begin
      e = 0;
    end

    e = !e;
  end
endmodule
```
Running the following command causes Yosys to enter an infinite logging loop, possibly due to an internal infinite loop during synthesis:
```
yosys -p "read_verilog bug.v; synth_greenpak4"
```
I discovered this bug while testing Yosys using a fuzzing tool I am developing.

### Expected Behavior

The `synth_greenpak4` synthesis process should not result in an infinite loop.

### Actual Behavior

The following log was produced:
```
2.24.2. Continuing TECHMAP pass.
Using template $paramod\$lut\WIDTH=32'00000000000000000000000000000001\LUT=2'01 for cells of type $lut.
Using template $paramod\$lut\WIDTH=32'00000000000000000000000000000010\LUT=4'0111 for cells of type $lut.
No more expansions possible.
<suppressed ~41 debug messages>

2.25. Executing GREENPAK4_DFFINV pass (merge input/output inverters into FF/latch cells).
Merged input inverter into cell a.$auto$ff.cc:266:slice$122: GP_DLATCH -> GP_DLATCHI
Merged input inverter into cell a.$auto$ff.cc:266:slice$122: GP_DLATCHI -> GP_DLATCH
Merged input inverter into cell a.$auto$ff.cc:266:slice$122: GP_DLATCH -> GP_DLATCHI
Merged input inverter into cell a.$auto$ff.cc:266:slice$122: GP_DLATCHI -> GP_DLATCH
Merged input inverter into cell a.$auto$ff.cc:266:slice$122: GP_DLATCH -> GP_DLATCHI
Merged input inverter into cell a.$auto$ff.cc:266:slice$122: GP_DLATCHI -> GP_DLATCH
Merged input inverter into cell a.$auto$ff.cc:266:slice$122: GP_DLATCH -> GP_DLATCHI
Merged input inverter into cell a.$auto$ff.cc:266:slice$122: GP_DLATCHI -> GP_DLATCH
```
and it keeps repeating without termination.

KrystalDelusion, 2025-03-31:
> which I have minimized as much as possible

Much appreciated.

FWIW this can be minimized further with this script:
```
read_verilog bug.v
synth_greenpak4 -run :map_cells
shregmap -tech greenpak4
dfflibmap -liberty +/greenpak4/gp_dff.lib
dffinit -ff GP_DFF Q INIT
dffinit -ff GP_DFFR Q INIT
dffinit -ff GP_DFFS Q INIT
dffinit -ff GP_DFFSR Q INIT
iopadmap -bits -inpad GP_IBUF OUT:IN -outpad GP_OBUF IN:OUT -inoutpad GP_OBUF OUT:IN -toutpad GP_OBUFT OE:IN:OUT -tinoutpad GP_IOBUF OE:OUT:IN:IO
attrmvcp -attr src -attr LOC t:GP_OBUF t:GP_OBUFT t:GP_IOBUF n:*
attrmvcp -attr src -attr LOC -driven t:GP_IBUF n:*
techmap -map +/greenpak4/cells_map.v
bugpoint -runner "timeout 1" -command greenpak4_dffinv
select *
write_rtlil -selected min.il
```

Which I was able to put together because the failure was in `GREENPAK4_DFFINV`, and `greenpak4_dffinv` gets called in the `map_cells` step.  So we call `synth_greenpak4 -run :map_cells`, and then all of the commands *before* `greenpak4_dffinv`.  Then `bugpoint -runner "timeout 1"` tries to cut parts of the design out and testing if the command is able to successfully finish in less than 1 second.  The `select *` there is because otherwise the output contains all the greenpak4 cell definitions, and `select *` selects all non-blackbox objects.  Then we write out to rtlil (though verilog would also work fine) which gives:
```
# Generated by Yosys 0.50 (git sha1 b5170e139, ccache clang++ 18.1.8 -fPIC -O3)

autoidx 189

attribute \src "bug.v:1.1-15.10"
attribute \top 1
module \a

  wire \e

  attribute \module_not_derived 1
  attribute \src "/usr/local/bin/../share/yosys/greenpak4/cells_map.v:125.11-125.50"
  cell \GP_INV $abc$167$auto$blifparse.cc:535:parse_blif$170
    connect \IN \e
    connect \OUT \e
  end

  attribute \module_not_derived 1
  attribute \src "bug.v:4.3-14.6|/usr/local/bin/../share/yosys/greenpak4/cells_latch.v:2.12-6.4"
  cell \GP_DLATCH $auto$ff.cc:266:slice$122
    connect \D \e
    connect \Q 1'x
    connect \nCLK 1'x
  end

  connect { } { }
  connect { } { }
end
```
And then we can further cut out the unnecessary attributes and such down to
```
module \a
  wire \e

  cell \GP_INV \a1
    connect \IN \e
    connect \OUT \e
  end

  cell \GP_DLATCH \a2
    connect \D \e
    connect \Q 1'x
    connect \nCLK 1'x
  end
end
```

At a glance this is because of the logic loop in the design, where `greenpak4_dffinv` runs this code:
```
while (inv_out2in.count(d_bit))
{
	sig_use_cnt[d_bit]--;
	invert_gp_dff(cell, true);
	d_bit = inv_out2in.at(d_bit);
	cell->setPort(ID::D, d_bit);
	sig_use_cnt[d_bit]++;
}
```
And the `d_bit` there never gets resolved because of the logic loop.

sdjasj, 2025-03-31:
> FWIW this can be minimized further with this script:

Thank you for your prompt response! I also learned a lot from the minimized script :)

KrystalDelusion, 2025-05-01:
Another logic loop for #5058