YosysHQ/yosys #4977: yosys-abc: src/map/scl/sclLibUtil.c:77: void abc::Abc_SclHashCells(SC_Lib *): Assertion `*pPlace == -1' failed.

oharboe, 2025-03-31:
### Version

Yosys 0.51 (git sha1 c4b519022, clang++ 18.1.3 -fPIC -O3)

### On which OS did this happen?

Linux

### Reproduction Steps

untar and run https://drive.google.com/file/d/1teJ8WAzw8EU9B1FdBe42cxbNczJQhb3w/view?usp=sharing

```
$ ./run-me-ethmac-asap7-base.sh
[deleted]
BC: Library "asap7_merged" from "/home/oyvind/OpenROAD-flow-scripts/flow/designs/asap7/ethmac_lvt/foo/./designs/asap7/ethmac_lvt/objects/asap7/ethmac/base/lib/merged.lib" has 212 cells (46 skipped: 46 seq; 0 tri-state; 0 no func; 146 dont_use).  Time =     0.50 sec
ABC: Memory =  164.45 MB. Time =     0.50yosys-abc: src/map/scl/sclLibUtil.c:77: void abc::Abc_SclHashCells(SC_Lib *): Assertion `*pPlace == -1' failed.
ABC: Aborted (core dumped)
```

Originally encountered in https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/3020

### Expected Behavior

An actionable error message for what's wrong or that it works.

### Actual Behavior

assert()

oharboe, 2025-03-31:
@povik This issue isn't great and I'm unsure how to whittle it down as I have no clue what the error message is telling me or what is wrong, but I thought I should file a github issue anyway to at least share with the group reproducible case.

QuantamHD, 2025-03-31:
212 cells (46 skipped: 46 seq; 0 tri-state; 0 no func; 146 dont_use) = 196 cells that are not available for ABC. That seems odd. ABC often needs certain cells like inv, and, buf. I'd look into why so many cells are marked dont_use

oharboe, 2025-03-31:
Fixed my design https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/3028.

So this issue is "demoted" from a bug to a feature request: error messages can always be more helpful.

An assert() is normally associated with something that should be fixed in the code, even if the input is broken.

povik, 2025-03-31:
@oharboe this comes from abc, and is probably not the only user-exposed assert.

This is the code for anyone wondering:

[image]

I'm not sure why the print hasn't made its way to you -- either `pCell->name` was null or Yosys is to blame as it captures and then reproduces abc output.

oharboe, 2025-03-31:
@povik Yes, I've encountered what are errors in input in abc manifesting itself as assert()s in abc, which surprised me the first time. Also, yes, I've found that I have to massage ORFS before it will print out anything regarding abc errors. In my case, I set YOSYS_FLAGS= in ORFS before I get even the output above as `YOSYS_FLAGS=-v 3` by default which hides all abc errors(effectively).

povik, 2025-03-31:
There's still the "There are two standard cells with the same name" print which is in the code but hasn't shown up for you.

wjrforcyber, 2025-04-02:
@oharboe If you could minimise the Reproduction Steps it would be helpful, since I saw dependencies `yaml` that I have to install in a python virtual env to run your script. And the latest tcl will show invalid command on `lmap`, it's better if you could specify the versions. This error is only abc and library related so it would be really helpful if you could minimise it. See [Minimal reproducible example](https://en.wikipedia.org/wiki/Minimal_reproducible_example#:~:text=In%20computing%2C%20a%20minimal%20reproducible,to%20be%20demonstrated%20and%20reproduced).
Anyway, maybe you could just provide the merged library `Library "asap7_merged"` ? Although the error has been fixed, I'll try to see if it's the ABC's fault that the error message hasn't been printed.

oharboe, 2025-04-02:
Here is a smaller reproduction example that has the python dependencies and some other files removed. [assert2.tar.gz](https://github.com/user-attachments/files/19561970/assert2.tar.gz)

> [@oharboe](https://github.com/oharboe) If you could minimise the Reproduction Steps it would be helpful, since I saw dependencies `yaml` that I have to install in a python virtual env to run your script.

Python dependencies removed.

> And the latest tcl will show invalid command on `lmap`, it's better if you could specify the versions.

I'm running the .tcl code from within Yosys so I'm unsure how I can specify the version of tcl since Yosys is the tcl host.

> Anyway, maybe you could just provide the merged library `Library "asap7_merged"` ?

I don't have a file called asap7_merged.

> Although the error has been fixed, I'll try to see if it's the ABC's fault that the error message hasn't been printed.

👍

hexratcc, 2026-09-21:
The abc error should now be propagated correctly as of https://github.com/berkeley-abc/abc/pull/568

closed 2026-09-21 by hexratcc