YosysHQ/yosys #4978: muxcover pass stalls if $_MUX_ output is disconnected

KrystalDelusion, 2025-04-01:
### Version

Yosys 0.50+14 (git sha1 eb5abf9c1, ccache clang++ 18.1.8 -fPIC -O3)

### On which OS did this happen?

Linux

### Reproduction Steps

Not sure how likely this is to come up in the wild, but while investigating #4590 (and its possible connection to #4722) I discovered that if muxcover gets called with a $_MUX_ cell where the output (`Y`), and at least two inputs (`A`, `B`, `S`) are disconnected (set to `1'x`) then `muxcover` gets stuck in an infinite loop.
e.g. stall_muxcover.il
```
module \top

  wire input 1 \a
  wire input 2 \b
  wire input 3 \s
  wire output 4 \y

  cell $_MUX_ \badmux
    connect \A 1'x
    connect \B 1'x
    connect \S \s
    connect \Y 1'x
  end
end
```
and calling
`yosys -p muxcover stall_muxcover.il`

### Expected Behavior

No infinite loop, maybe a log error, at the very least an assertion error.

### Actual Behavior

```
-- Parsing `temp_tests/stall_muxcover.il' using frontend `rtlil' --

1. Executing RTLIL frontend.
Input filename: temp_tests/stall_muxcover.il

-- Running command `muxcover' --

2. Executing MUXCOVER pass (mapping to wider MUXes).
Covering MUX trees in module top..
  Treeifying 1 MUXes:
^C
```

KrystalDelusion, 2025-04-01:
Looks like this is the culprit https://github.com/YosysHQ/yosys/blob/c08f72b806f9c3d068bb6b365d351f0bb9e8a8fa/passes/techmap/muxcover.cc#L139-L147

sdjasj, 2025-04-01:
Yes, I encountered the same issue during fuzz testing. Specifically, an infinite loop occurred during `2.36. Executing MUXCOVER pass (mapping to wider MUXes)`, causing Yosys to hang.
Below is the verilog code I minimized using creduce. Although it still looks somewhat lengthy, I've already reduced it as much as possible:
```verilog
module a (
    input b,
    input in_gekt,
    input in_ciai,
    output reg out_nkxft
);

reg reg_kbihd;
reg d;
wire wire_rrl, wire_molxc, wire_cxu;
wire in_lcs;
wire in_cvi;
reg e, f, g, reg_oqxwyit;

always @(posedge b or negedge c) begin
    if (!c) begin
        out_nkxft = 'h0;
    end else begin
        out_nkxft = wire_rrl > reg_kbihd;
        reg_kbihd = (wire_rrl & in_lcs)
                        ? (|wire_molxc ? in_gekt
                                       : (&in_ciai ? 0 : reg_oqxwyit / wire_cxu))
                        : 0;
    end
end

always @(in_gekt or in_cvi) begin
    case (0)
        32'd34: ;
        default: begin
            d = wire_molxc;
        end
    endcase
end

always begin
    if (e) begin
        reg_kbihd = 0;
    end else begin
        d = g;
        reg_kbihd = f;
    end
end

endmodule
```