YosysHQ/yosys #4985: A `core dumped` occurred during the `synth_microchip` process when executing the `MICROCHIP_DSP` pass.

sdjasj, 2025-04-03:
### Version

Yosys 0.51+107 (git sha1 f03b44959, clang++ 14.0.0-1ubuntu1.1 -fPIC -O3)

### On which OS did this happen?

Linux

### Reproduction Steps

The following is the problematic Verilog code file `min.v`, which I have minimized as much as possible out of responsibility:
```verilog
module a (
    input  [2:0] b,
    input        e,
    output       c
);
    wire [31:0]  d;
    reg  [31:0]  h;
    reg          f;
    reg          g;
    reg          i;
    
    always @(posedge e) begin
        c <= b;
        h <= b;
    end
    
    always @(*) begin
        e = f;
        case (f)
            'd1: begin
                f = d * h ? 0 : g;
                h = f + i;
            end
        endcase
    end
endmodule
```
Although it looks a bit long, it has already been minimized — removing any single line will prevent the core dump from occurring :)
The following is the minimized  `min.il` file using Yosys's bugpoint.
```
# Generated by Yosys 0.51+107 (git sha1 f03b44959, clang++ 14.0.0-1ubuntu1.1 -fPIC -O3)
autoidx 38
attribute \cells_not_processed 1
attribute \src "min.v:1.1-26.10"
module \a
  attribute \src "min.v:12.5-15.8"
  wire width 32 $0\h[31:0]
  attribute \src "min.v:17.5-25.8"
  wire width 32 $1\h[31:0]
  attribute \src "min.v:17.5-25.8"
  wire width 32 $2\h[31:0]
  attribute \src "min.v:22.21-22.26"
  wire width 32 $add$min.v:22$7_Y
  attribute $bugpoint 1
  wire width 32 $auto$bugpoint.cc:258:simplify_something$11
  attribute $bugpoint 1
  wire $auto$bugpoint.cc:258:simplify_something$13
  attribute $bugpoint 1
  wire width 32 $auto$bugpoint.cc:258:simplify_something$14
  attribute $bugpoint 1
  wire width 32 output 1 $auto$bugpoint.cc:258:simplify_something$16
  attribute $bugpoint 1
  wire $auto$bugpoint.cc:258:simplify_something$17
  attribute $bugpoint 1
  wire $auto$bugpoint.cc:258:simplify_something$18
  attribute $bugpoint 1
  wire width 32 $auto$bugpoint.cc:258:simplify_something$8
  attribute $bugpoint 1
  wire width 32 $auto$bugpoint.cc:258:simplify_something$9
  wire $delete_wire$21
  wire $delete_wire$22
  wire $delete_wire$23
  wire width 32 $delete_wire$24
  attribute \src "min.v:7.18-7.19"
  wire width 32 \h
  attribute \src "min.v:22.21-22.26"
  cell $add $add$min.v:22$7
    parameter \A_SIGNED 0
    parameter \A_WIDTH 1
    parameter \B_SIGNED 0
    parameter \B_WIDTH 1
    parameter \Y_WIDTH 32
    connect \A $auto$bugpoint.cc:258:simplify_something$17
    connect \B $auto$bugpoint.cc:258:simplify_something$18
    connect \Y $add$min.v:22$7_Y
  end
  attribute \src "min.v:21.21-21.26"
  cell $mul $mul$min.v:21$3
    parameter \A_SIGNED 0
    parameter \A_WIDTH 32
    parameter \B_SIGNED 0
    parameter \B_WIDTH 32
    parameter \Y_WIDTH 32
    connect \A $auto$bugpoint.cc:258:simplify_something$14
    connect \B \h
    connect \Y $auto$bugpoint.cc:258:simplify_something$16
  end
  attribute \src "min.v:12.5-15.8"
  process $proc$min.v:12$1
    sync posedge $delete_wire$22
      update \h $0\h[31:0]
  end
  attribute \src "min.v:17.5-25.8"
  process $proc$min.v:17$2
    assign $1\h[31:0] $2\h[31:0]
    attribute \src "min.v:19.9-24.16"
    switch { 31'0000000000000000000000000000000 $delete_wire$21 }
      case 1
        assign $2\h[31:0] $add$min.v:22$7_Y
      case 
        assign $2\h[31:0] \h
    end
    sync always
      update \h $1\h[31:0]
  end
end
```
The bug can be reproduced with the following commands:
```
yosys -p "read_verilog min.v; synth_microchip"
```
or
```
yosys -p "read_rtlil min.il; synth_microchip"
```

### Expected Behavior

A core dump should not occur during the Yosys synthesis process.

### Actual Behavior

Yosys produced the following log during execution:
```
2.18. Executing OPT_EXPR pass (perform const folding).
Optimizing module a.
<suppressed ~1 debug messages>

2.19. Executing WREDUCE pass (reducing word size of cells).
Removed top 11 bits (of 15) from port B of cell a.$techmap$mul$min.v:21$3.$add$/usr/local/bin/../share/yosys/mul2dsp.v:173$125 ($add).

2.20. Executing MICROCHIP_DSP pass (pack resources into DSPs).
Analysing a.$mul$min.v:21$3.genblk1.genblk1.genblk1.genblk1.genblk1.genblk1.genblk1.sliceA[0].mul for Microchip MACC_PA packing.
Segmentation fault (core dumped)
```

After instrumenting with Asan and UBsan, the following error was reported:
```
2.20. Executing MICROCHIP_DSP pass (pack resources into DSPs).
Analysing a.$mul$bug.v:5$4.genblk1.genblk1.genblk1.genblk1.genblk1.genblk1.genblk1.sliceA[2].mul for Microchip MACC_PA packing.
  postadder $techmap$mul$bug.v:5$4.$add$/usr/local/bin/../share/yosys/mul2dsp.v:159$146 ($add)
passes/pmgen/microchip_dsp.cc:126:23: runtime error: member access within null pointer of type 'Yosys::RTLIL::AttrObject'
SUMMARY: UndefinedBehaviorSanitizer: undefined-behavior passes/pmgen/microchip_dsp.cc:126:23 in 
passes/pmgen/microchip_dsp.cc:126:34: runtime error: member call on null pointer of type 'Yosys::hashlib::dict<Yosys::RTLIL::IdString, Yosys::RTLIL::Const>'
SUMMARY: UndefinedBehaviorSanitizer: undefined-behavior passes/pmgen/microchip_dsp.cc:126:34 in 
AddressSanitizer:DEADLYSIGNAL
=================================================================
==8876==ERROR: AddressSanitizer: SEGV on unknown address 0x000000000000 (pc 0x55c09e5008bb bp 0x7ffd6d297af0 sp 0x7ffd6d297a20 T0)
==8876==The signal is caused by a READ memory access.
==8876==Hint: address points to the zero page.
...
SUMMARY: AddressSanitizer: SEGV (/usr/local/bin/yosys+0x1dca8bb) (BuildId: bad06cfe5b0c7051344047df9ec204002aba65ee) in std::vector<int, std::allocator<int> >::empty() const
==8876==ABORTING
```

The error occurred at `passes/pmgen/microchip_dsp.cc:126:34`. The corresponding code snippet is:
```cpp
for (auto c : Q.chunks()) {
	auto it = c.wire->attributes.find(ID::init);
	if (it == c.wire->attributes.end())
		continue;
	for (int i = c.offset; i < c.offset + c.width; i++) {
		log_assert(it->second[i] == State::S0 || it->second[i] == State::Sx);
		it->second.bits()[i] = State::Sx;
	}
}
```

The line that caused the crash is:
```cpp
auto it = c.wire->attributes.find(ID::init);
```

sdjasj, 2025-04-04:
Following the suggestions and guidance from **KrystalDelusion**, I performed further minimization using the steps below. First, I ran the following commands to isolate the last step before the **core dump was triggered**:

```yosys
read_verilog min.v
synth_microchip -run :map_dsp
memory_dff
techmap -map +/mul2dsp.v -map +/microchip/polarfire_dsp_map.v -D DSP_A_MAXWIDTH=18 -D DSP_B_MAXWIDTH=18 -D DSP_A_MAXWIDTH_PARTIAL=18 -D DSP_A_MINWIDTH=2 -D DSP_B_MINWIDTH=2 -D DSP_Y_MINWIDTH=9 -D DSP_SIGNEDONLY=1 -D DSP_NAME=$__MUL18X18
select a:mul2dsp
setattr -unset mul2dsp
opt_expr -fine
wreduce
select -clear
```

Then I used `bugpoint` to automatically minimize the test case:

```yosys
bugpoint -command "microchip_dsp -family polarfire"
```

Finally, I extracted the intermediate code generated by `bugpoint` with:

```yosys
select *
write_rtlil -selected min.il
```

After that, I manually minimized and renamed variables to produce the following **minimal RTLIL example** that still triggers the **core dump**:

```il
# Generated by Yosys 0.51+107 (git sha1 f03b44959, clang++ 14.0.0-1ubuntu1.1 -fPIC -O3)

autoidx 156

attribute \top 1
module \a

  wire width 48 output 1 \out
  wire width 2 \h

  cell $dff \dff
    parameter \CLK_POLARITY 1
    parameter \WIDTH 3
    connect \CLK 1'x
    connect \D 3'x
    connect \Q { 1'0 \h }
  end

  cell \MACC_PA \mac
    connect \A 18'x
    connect \B { 16'0 \h }
    connect \P \out
  end
end
```

To reproduce the crash, run the following:

```yosys
read_rtlil min.il
microchip_dsp -family polarfire
```
Here is the English translation of your explanation:

---

I suspect that the cause of the `nullptr` dereference is the line `connect \Q { 1'0 \h }`.  
Here, `Q` is `{1'0, \h}`, which means that `Q[2]` is a constant `0`, while `Q[1:0]` is the wire `\h` (2 bits wide).  
When iterating over `Q.chunks()`, the code seems to assume that every `c` contains a wire and directly calls `c.wire->attributes.find(...)`, which leads to a null pointer dereference.

sdjasj, 2025-04-05:
According to the suggestion in [https://github.com/YosysHQ/yosys/issues/4987#issuecomment-2779828853](https://github.com/YosysHQ/yosys/issues/4987#issuecomment-2779828853), replacing `opt_expr -fine` with `opt -fast -fine` still reproduces the issue. The updated script is as follows:

```verilog
read_verilog min.v
synth_microchip -run :map_dsp
memory_dff
techmap -map +/mul2dsp.v -map +/microchip/polarfire_dsp_map.v -D DSP_A_MAXWIDTH=18 -D DSP_B_MAXWIDTH=18 -D DSP_A_MAXWIDTH_PARTIAL=18 -D DSP_A_MINWIDTH=2 -D DSP_B_MINWIDTH=2 -D DSP_Y_MINWIDTH=9 -D DSP_SIGNEDONLY=1 -D DSP_NAME=$__MUL18X18
select a:mul2dsp
setattr -unset mul2dsp
opt -fast -fine // originally opt_expr -fine
wreduce
select -clear
microchip_dsp -family polarfire
```

KrystalDelusion, 2025-05-01:
#5058 on this one too