YosysHQ/yosys #4991: `ERROR: Assert `sig_macc.count(n->y) == 0' failed` occurs when multiple `$add` cells drive the same signal.

sdjasj, 2025-04-05:
### Version

Yosys 0.51+107 (git sha1 f03b44959, clang++ 14.0.0-1ubuntu1.1 -fPIC -O3)

### On which OS did this happen?

Linux

### Reproduction Steps

The following is the minimized source file `bug.v`:

```verilog
module a(input g, c, d, output reg b);
    always @(*) begin
        b = g + d;
    end

    always @(*) begin
        b = c + d;
    end
endmodule
```

When synthesizing this design using the following script, an assertion failure occurs:

```
ERROR: Assert `sig_macc.count(n->y) == 0' failed in passes/techmap/alumacc.cc:184.
```

Synthesis script:
```
read_verilog bug.v
synth_gowin -run :coarse
proc
opt_expr
opt_clean
check
opt -nodffe -nosdff
fsm
opt
wreduce
peepopt
opt_clean
alumacc # Error occurs here
```

The following is the minimized intermediate RTLIL code generated by `bugpoint`:

```verilog
module \a

  wire output 1 \o

  cell $add \c0
    parameter \A_SIGNED 0
    parameter \A_WIDTH 1
    parameter \B_SIGNED 0
    parameter \B_WIDTH 1
    parameter \Y_WIDTH 1
    connect \A 1'x
    connect \B 1'x
    connect \Y \o
  end

  cell $add \c1
    parameter \A_SIGNED 0
    parameter \A_WIDTH 1
    parameter \B_SIGNED 0
    parameter \B_WIDTH 1
    parameter \Y_WIDTH 1
    connect \A 1'x
    connect \B 1'x
    connect \Y \o
  end
end
```

I suspect the issue is caused by signal `b` being driven by two `always` blocks simultaneously. However, it is strange that the following variant does **not** cause an assertion failure, even though `b` is still driven by two `always` blocks:

```verilog
module a(input g, c, d, output reg b);
    always @(*) begin
        b = g + d;
    end

    always @(*) begin
        b = c / d; // changed `+` to `/`
    end
endmodule
```

And the corresponding RTLIL code is:

```verilog
module \a

  wire \w0
  wire \w1

  wire output 4 \o
  wire input 2 \i0
  wire input 3 \i1
  wire input 1 \i2

  cell $alu \c0
    parameter \A_SIGNED 0
    parameter \A_WIDTH 1
    parameter \B_SIGNED 0
    parameter \B_WIDTH 1
    parameter \Y_WIDTH 1
    connect \A \i1
    connect \B \i2
    connect \BI 1'0
    connect \CI 1'0
    connect \CO \w1
    connect \X \w0
    connect \Y \o
  end

  cell $div \c1
    parameter \A_SIGNED 0
    parameter \A_WIDTH 1
    parameter \B_SIGNED 0
    parameter \B_WIDTH 1
    parameter \Y_WIDTH 1
    connect \A \i0
    connect \B \i1
    connect \Y \o
  end
end
```

A possible explanation is that Yosys attempts to extract and merge the logic for the addition operation (into `$macc`/`$alu`), but does not process the division operation in the same way.

### Expected Behavior

The behavior of the two code examples should be consistent — they should either both trigger the assertion or both synthesize successfully. Alternatively, the assertion should not be triggered.

### Actual Behavior

The first piece of code triggers the assertion, while the second one does not.

KrystalDelusion, 2025-04-07:
What does Verilator do with this?  `read_verilog` has a tendency to handle invalid designs poorly

sdjasj, 2025-04-07:
> What does Verilator do with this? `read_verilog` has a tendency to handle invalid designs poorly

When using `verilator -cc test.v`, it detects that the signal is multiply driven and gives the warning `%Warning-MULTIDRIVEN: test.v:1:36: Bits [0:0] of signal 'b' have multiple combinational drivers`, but it still generates the output files.

KrystalDelusion, 2025-05-01:
Potentially closed by #5058