YosysHQ/yosys #5002: Possible memory leak at CHECK pass

RicardoLera, 2025-04-08:
### Version

Yosys 0.44 (git sha1 80ba43d26, g++ 14.2.1 -march=x86-64 -mtune=generic -O2 -fno-plt -fexceptions -fstack-clash-protection -fcf-protection -fno-omit-frame-pointer -mno-omit-leaf-frame-pointer -ffile-prefix-map=/build/yosys/src=/usr/src/debug/yosys -fPIC -O3)

### On which OS did this happen?

Linux

### Reproduction Steps

ghdl version:
```
GHDL 6.0.0-dev (b6.0.0.r51.g26ee28a24) [Dunoon edition]
 Compiled with GNAT Version: 14.2.1 20250207
 GCC 12.4.0 code generator
```
ghdl-yosis-plugin version:
`ghdl-yosys-plugin-git r231.8c29f2c-1` (AUR)

Bash script used to invoke yosys:
```
(...)

if [ "$1" = "synth" ] ; then
  
  comp_files=(big list of files including hecate_oa.vhd)

  yosys -m ghdl -p \
    "ghdl --std=08 -fsynopsys --latches $comp_files -e hecate_oa; synth_xilinx -top hecate_oa -family $family -flatten; json -o yosys_out/$family.json" \
  &> yosys_out/xc7.txt

else

(...)
```
Command used to run the script:
`systemd-run --user -G -p MemoryHigh=4G MemoryMax=10G ./run_sim.sh synth`

Note: I can't provide all the project files here, but there isn't anything wrong with them besides it just being a big project. The bug should be reproducible with any sufficiently complex vhdl project.

### Expected Behavior

Synthesis of the architecture. If failure to do so due to a problem in the code, then an error is expected.

### Actual Behavior

Upon reaching stage `2.10. Executing CHECK pass (checking for obvious problems)` the physical memory consumption spikes tremendously until the system runs out of memory and the OOM killer terminates the process.

[image]

The image shows the output of `top` moments before the system ran out of memory. Throughout the process, yosys was kept below 4G of memory usage by `systemd-run`, as expected. Strangely, as the system started running out of memory, `top` did not associate the usage with any process including yosys. Nevertheless, when memory consumption reached the available memory, the OOM killer terminated the process started by `systemd-run`, which recovered the system. This was attempted multiple times with no other programs running, to the same result.

My guess is some sort of memory leak issue when handling large modules during the CHECK pass.

whitequark, 2025-04-08:
This seems rather difficult to reproduce from the steps shown. Since this would be an issue with the Yosys core, please provide a reproducer with a complete input that relies only on Yosys, or else perform memory profiling on your own if you are unable to.

povik, 2025-04-08:
@RicardoLera to narrow down the cause you can try patching this https://github.com/YosysHQ/yosys/blob/406ee4c8d3a9c594ddcb11069f84158b1713a5de/passes/cmds/check.cc#L217 to `if (false) {` and see if the issue persists.

KrystalDelusion, 2025-04-30:
@RicardoLera I'm going to close this issue if you're unable to provide any additional information that could help reproduce this.

RicardoLera, 2025-05-02:
Apologies for the delay. I've been investigating the issue and asked some people in the field about it. Supposedly (and also bafflingly), the leaks are due to Xilinx's own proprietary libs, and would still be present if this project was synthesized in Vivado instead if yosys. I haven't had the opportunity to test it for myself yet, hence the delay. If this turns out to be true, I will report here for future reference.