YosysHQ/yosys #5010: memory pass is not respecting (* keep *)

KelvinChung2000, 2025-04-09:
### Version

Yosys 0.51+85 (git sha1 1113c8c95, g++ 11.4.0-1ubuntu1~22.04 -fPIC -O3)

### On which OS did this happen?

Linux

### Reproduction Steps

With the following module:

```verilog
module comb_mem_d1 #(
    parameter WIDTH = 32,
    parameter SIZE = 16,
    parameter IDX_SIZE = 4
) (
   input wire                logic [IDX_SIZE-1:0] addr0,
   input wire                logic [ WIDTH-1:0] write_data,
   input wire                logic write_en,
   input wire                logic clk,
   input wire                logic reset,
   (* keep *)output                    logic [ WIDTH-1:0] read_data,
);

  (* keep *)logic [WIDTH-1:0] mem[SIZE-1:0];

  /* verilator lint_off WIDTH */
  assign read_data = mem[addr0];

  always_ff @(posedge clk) begin
    if (!reset && write_en)
      mem[addr0] <= write_data;
  end

  // Check for out of bounds access
  `ifdef VERILATOR
    always_comb begin
      if (addr0 >= SIZE)
        $error(
          "comb_mem_d1: Out of bounds access\n",
          "addr0: %0d\n", addr0,
          "SIZE: %0d", SIZE
        );
    end
  `endif
endmodule

module test();
  logic [3:0] addr0;
  logic [31:0] write_data;
  logic write_en;
  logic clk;
  logic reset;
  logic [31:0] read_data1;
  logic [31:0] read_data2;
  logic done;

  assign write_en = 1;

  (* keep=1 *)
  comb_mem_d1 #(
    .WIDTH(32),module comb_mem_d1 #(
    parameter WIDTH = 32,
    parameter SIZE = 16,
    parameter IDX_SIZE = 4
) (
   input wire                logic [IDX_SIZE-1:0] addr0,
   input wire                logic [ WIDTH-1:0] write_data,
   input wire                logic write_en,
   input wire                logic clk,
   input wire                logic reset,
   (* keep *)output                    logic [ WIDTH-1:0] read_data,
);

  (* keep *)logic [WIDTH-1:0] mem[SIZE-1:0];

  /* verilator lint_off WIDTH */
  assign read_data = mem[addr0];

  always_ff @(posedge clk) begin
    if (!reset && write_en)
      mem[addr0] <= write_data;
  end

endmodule

module test();
  logic [3:0] addr0;
  logic [31:0] write_data;
  logic write_en;
  logic clk;
  logic reset;
  logic [31:0] read_data1;
  logic [31:0] read_data2;
  logic done;

  assign write_en = 1;

  (* keep=1 *)
  comb_mem_d1 #(
    .WIDTH(32),
    .SIZE(16),
    .IDX_SIZE(4)
  ) data1 (
    .addr0(addr0),
    .write_data(),
    .write_en(),
    .clk(clk),
    .reset(reset),
    .read_data(read_data1),
  );

  (* keep=1 *)
  comb_mem_d1 #(
      .WIDTH(32),
      .SIZE(16),
      .IDX_SIZE(4)
  ) data2 (
      .addr0(addr0),
      .write_data(),
      .write_en(),
      .clk(clk),
      .reset(reset),
      .read_data(read_data2),
  );

  (* keep=1 *)
  comb_mem_d1 #(
      .WIDTH(32),
      .SIZE(16),
      .IDX_SIZE(4)
  ) data3 (
      .addr0(addr0),
      .write_data(write_data),
      .write_en(write_en),
      .clk(clk),
      .reset(reset),
      .read_data()
  );

  assign write_data = read_data1 + read_data2;

endmodule

    .SIZE(16),
    .IDX_SIZE(4)
  ) data1 (
    .addr0(addr0),
    .write_data(),
    .write_en(),
    .clk(clk),
    .reset(reset),
    .read_data(read_data1),
  );

  (* keep=1 *)
  comb_mem_d1 #(
      .WIDTH(32),
      .SIZE(16),
      .IDX_SIZE(4)
  ) data2 (
      .addr0(addr0),
      .write_data(),
      .write_en(),
      .clk(clk),
      .reset(reset),
      .read_data(read_data2),
  );

  (* keep=1 *)
  comb_mem_d1 #(
      .WIDTH(32),
      .SIZE(16),
      .IDX_SIZE(4)
  ) data3 (
      .addr0(addr0),
      .write_data(write_data),
      .write_en(write_en),
      .clk(clk),
      .reset(reset),
      .read_data()
  );

  assign write_data = read_data1 + read_data2;

endmodule
```

And running the following command:
```
read_verilog -sv example.sv
hierarchy -auto-top; flatten; proc; opt -full
memory -nomap
stat
```

### Expected Behavior

After calling the stat command, I should still see 3 `$memwr_v2` cell.

### Actual Behavior

I am only left with 1 `$memwr_v2` cell.

```
=== test ===

   Number of wires:                 26
   Number of wire bits:            317
   Number of public wires:          26
   Number of public wire bits:     317
   Number of ports:                  0
   Number of port bits:              0
   Number of memories:               0
   Number of memory bits:            0
   Number of processes:              0
   Number of cells:                  5
     $add                            1
     $mem_v2                         1
     $scopeinfo                      3
```

KrystalDelusion, 2025-04-09:
This is specifically the `opt_mem` pass:
```
hierarchy -auto-top; flatten; proc; opt -full
select -assert-count 3 m:* # passes
opt_mem
select -assert-count 3 m:* # fails
```
And if you `stat` before/after it you get
```
=== test ===

   Number of wires:                 26
   Number of wire bits:            317
   Number of public wires:          26
   Number of public wire bits:     317
   Number of ports:                  0
   Number of port bits:              0
   Number of memories:               3
   Number of memory bits:         1536
   Number of processes:              0
   Number of cells:                 10
     $add                            1
     $memrd                          3
     $memwr_v2                       3
     $scopeinfo                      3
```
and
```
=== test ===

   Number of wires:                 26
   Number of wire bits:            317
   Number of public wires:          26
   Number of public wire bits:     317
   Number of ports:                  0
   Number of port bits:              0
   Number of memories:               1
   Number of memory bits:          512
   Number of processes:              0
   Number of cells:                  6
     $add                            1
     $memrd                          1
     $memwr_v2                       1
     $scopeinfo                      3
```

KelvinChung2000, 2025-07-10:
After a bit more testing, the problem seems to be coming from flatten. The `flatten` pass flattens the design, but does not keep the `keep` attribute, which leads to subsequent `opt -full` removing the dffs using opt_dff, then removing the memor.y

```
read_verilog << EOF
module Mem #(
    parameter WIDTH = 32,
    parameter SIZE = 16,
    parameter IDX_SIZE = 16
) (
    input wire [WIDTH-1:0] addr0,
    input wire [WIDTH-1:0] write_data,
    input wire write_en,

    input wire clk,

    output reg [WIDTH-1:0] read_data,

);
  reg [WIDTH-1:0] mem[SIZE-1:0];

  always @(posedge clk) begin
    if (write_en)
      mem[addr0[IDX_SIZE-1:0]] <= write_data;

    read_data <= mem[addr0[IDX_SIZE-1:0]];
  end

endmodule

module test_keep_at_instance ();

(* keep *)
Mem mem (
    .clk(),
    .write_en(),
    .write_data(),
    .read_data(),
    .addr0()
);

endmodule

EOF

hierarchy -auto-top;
select -assert-count 1 a:keep
flatten
select -assert-count 1 a:cell_keep
select -list a:keep
proc; opt -full;
select -assert-count 1 m:* #fail

```

The old example still works after the `opt-full` is likely because I have added a `keep` at the output read_data, which prevents `opt` from allowing it to be removed/or cleaned. In this case, I have removed it from the output, and we can see that no memory is created. 

As a result, if I changed the attribute from `keep` to `keep_hierarchy`, then the `opt_mem` will not remove the memory. So I guess the real question is what `flatten` should do with the attributes?

KelvinChung2000, 2025-07-10:
I think propagating all the attribute make sense, but it is still "surprising", since an attribute on an instance might not imply that all the cells and wires within that instance should also have the attribute. 

I think what could be done is to make the `keep` attribute, when used with instantiation, imply `keep_hierarchy`?

KelvinChung2000, 2025-07-10:
Anyway, the opt_mem itself will still need a fix to make it respect the keep attribute. 

```
read_verilog << EOF
module Mem #(
    parameter WIDTH = 8,
    parameter SIZE = 16,
    parameter IDX_SIZE = 16
) (
    input wire [WIDTH-1:0] addr0,
    input wire [WIDTH-1:0] write_data,
    input wire write_en,

    input wire clk,

    output reg [WIDTH-1:0] read_data,

);
  reg [WIDTH-1:0] mem[SIZE-1:0];

  always @(posedge clk) begin
    if (write_en)
      mem[addr0[IDX_SIZE-1:0]] <= write_data;

    read_data <= mem[addr0[IDX_SIZE-1:0]];
  end

endmodule

module test_keep_at_instance ();

Mem mem (
    .clk(),
    .write_en(),
    .write_data(),
    .read_data(),
    .addr0()
);

endmodule

EOF

hierarchy -auto-top;
flatten
proc;

select -assert-any t:$mem*
setattr -set keep 1 t:$mem*
select -list a:keep

opt -full;

select -assert-any t:$mem* #pass
opt_mem
select -assert-any t:$mem* #fail

```