YosysHQ/yosys #5013: Yosys not reporting non-existing files

MrJake222, 2025-04-10:
### Version

Yosys 0.51+125 (git sha1 cc8fd3efc, g++ 12.2.0-14 -fPIC -O3)

### On which OS did this happen?

Linux

### Reproduction Steps

1. Run yosys on non-existent files: `yosys nonexistent.v`

### Expected Behavior

Yosys reports the files as missing / non-existent.

### Actual Behavior

Yosys doesn't report missing files. It happily concludes the parsing with **Successfully finished Verilog frontend.** despite the file not being there.

```
$ ls nonexistent.v
ls: cannot access 'nonexistent.v': No such file or directory
$ yosys nonexistent.v

 /----------------------------------------------------------------------------\
 |  yosys -- Yosys Open SYnthesis Suite                                       |
 |  Copyright (C) 2012 - 2025  Claire Xenia Wolf <claire@yosyshq.com>         |
 |  Distributed under an ISC-like license, type "license" to see terms        |
 \----------------------------------------------------------------------------/
 Yosys 0.51+125 (git sha1 cc8fd3efc, g++ 12.2.0-14 -fPIC -O3)

-- Parsing `nonexistent.v' using frontend ` -vlog2k' --

1. Executing Verilog-2005 frontend: nonexistent.v
Parsing Verilog input from `nonexistent.v' to AST representation.
Successfully finished Verilog frontend.

yosys> 
exit

End of script. Logfile hash: 277a437cbe, CPU: user 0.00s system 0.01s, MEM: 15.56 MB peak
Yosys 0.51+125 (git sha1 cc8fd3efc, g++ 12.2.0-14 -fPIC -O3)
Time spent: 93% 2x read_verilog (0 sec), 6% 1x read (0 sec)
```

KrystalDelusion, 2025-04-22:
It looks like this bug snuck in with #4834
```
 Yosys 0.50 (git sha1 b5170e139, ccache clang++ 18.1.8 -fPIC -O3)

yosys> read_verilog bad.v
ERROR: Can't open input file `bad.v' for reading: No such file or directory
```

KrystalDelusion, 2025-04-23:
@widlarizer Can you take a look at this?  The other users of `Yosys::uncompressed()` look like:
```c
std::istream* f = uncompressed(liberty_file.c_str());
yosys_input_files.insert(liberty_file);
if (f->fail())
	log_cmd_error("Can't open liberty file `%s': %s\n", liberty_file.c_str(), strerror(errno));
```
while `register.cc` is:
```c
f = uncompressed(filename, bin_input ? std::ifstream::binary : std::ifstream::in);
if (f == NULL)
	log_cmd_error("Can't open input file `%s' for reading: %s\n", filename.c_str(), strerror(errno));
```
And I'm not sure if the correct solution is to return `nullptr` and change the lib users to test `f == NULL`, or change `register.cc` to check `f == NULL || f->fail())`.

closed 2025-04-30 by widlarizer

sdjasj, 2025-05-07:
@KrystalDelusion Hi, I just came across this issue and found it quite interesting, so I did some additional testing. I noticed that when using `read_verilog` to read an existing directory, it doesn't report an error, even though it's not a valid Verilog file. I'm not sure if this behavior is expected.

```bash
mkdir temp && yosys -p "read_verilog temp"
```
```

 /----------------------------------------------------------------------------\
 |  yosys -- Yosys Open SYnthesis Suite                                       |
 |  Copyright (C) 2012 - 2025  Claire Xenia Wolf <claire@yosyshq.com>         |
 |  Distributed under an ISC-like license, type "license" to see terms        |
 \----------------------------------------------------------------------------/
 Yosys 0.52+137 (git sha1 aa30589c1, clang++ 14.0.0-1ubuntu1.1 -Og -fPIC -O1 -fno-omit-frame-pointer -fno-optimize-sibling-calls -fsanitize=address,undefined,leak)

-- Running command `read_verilog temp' --

1. Executing Verilog-2005 frontend: temp
Parsing Verilog input from `temp' to AST representation.
Successfully finished Verilog frontend.

End of script. Logfile hash: 405974766d, CPU: user 0.03s system 0.04s, MEM: 84.32 MB peak
Yosys 0.52+137 (git sha1 aa30589c1, clang++ 14.0.0-1ubuntu1.1 -Og -fPIC -O1 -fno-omit-frame-pointer -fno-optimize-sibling-calls -fsanitize=address,undefined,leak)
Time spent: 100% 2x read_verilog (0 sec)
```

PR #5073 (2025-04-30): gzip: refactor file open failure errors
Fixes #5013

By refactoring any error handling away from the passes, we now report `ERROR: Can't open input file '$FILENAME' for reading: No such file or directory` every time we fail to open a file with `uncompress`. This includes cases when the file isn't actually compressed. This changes some error outputs (`Can't open liberty file` -> `Can't open input file`) but I doubt anybody minds since you see the filename right after anyway