YosysHQ/yosys #5026: Possibly incorrect non-blocking assignment in ecp5 cells_sim

secworks, 2025-04-15:
The cells_sim library for ECP5 contains an always comb (always(*)) select assignment that use non-blocking assignments. This makes verilator choke, and is normally an incorrect use of non-blocking assignments (AFAIK). If this is not intentional I can provide a fix.

https://github.com/YosysHQ/yosys/blob/19845be85c1fd2385dcfc07cdc39c2c6aae26a7f/techlibs/ecp5/cells_sim.v#L389

KrystalDelusion, 2025-04-22:
I don't think this is intentional, and most of the other `always @(*)` blocks use `=`.  At a glance it looks like some of the other techlibs have the same issue, gatemate and greenpak4, as well as some of the tests (in .ys files).  Would you be willing to fix those as well?

secworks, 2025-04-23:
I've created a PR for the issue in sim_cells: #5055.
Will look at the others and create PRs for those separately.

secworks, 2025-04-23:
I've also created a PR for greenpak4, see #5057.

KrystalDelusion, 2025-04-28:
Thanks for doing all of those, I'll go ahead and close this issue now that they're all merged :)

closed 2025-04-28 by KrystalDelusion