YosysHQ/yosys #5028: failure in yosys-abc with return code 134

Cronus-38, 2025-04-16:
### Version

Yosys 0.49 (git sha1 427b5a251, g++ 11.4.0-1ubuntu1~22.04 -fPIC -O3)

### On which OS did this happen?

Linux

### Reproduction Steps

I'm running Yosys with a proprietary library via the Openroad-flow-scripts (but the issue can be reproduced standalone, as well). 

To reproduce the problem simply run the commands below.  You can use any RTL file to demonstrate the problem.  Please contact me directly regarding the proprietary library I'm using (I don't want to broadcast it here).

```
read_verilog -defer gcd.v
hierarchy -top gcd
synth -top gcd
dfflibmap -liberty <proprietary_library>.lib
abc -script abc_speed.script -liberty <proprietary_library>.lib 
```

abc_speed.script
```
&get -n
&st
&dch
&nf
&st
&syn2
&if -g -K 6
&synch2
&nf
&st
&syn2
&if -g -K 6
&synch2
&nf
&st
&syn2
&if -g -K 6
&synch2
&nf
&st
&syn2
&if -g -K 6
&synch2
&nf
&st
&syn2
&if -g -K 6
&synch2
&nf
&put
buffer -c
topo
stime -c
upsize -c
dnsize -c
```

### Expected Behavior

I expect synthesis to complete successfully.

### Actual Behavior

ABC: yosys-abc: src/map/scl/sclSize.h:153: abc::SC_Man *abc::Abc_SclManAlloc(abc::SC_Lib *, abc::Abc_Ntk_t *): Assertion `pLib->unit_cap_snd == 15' failed.
sh: line 1: 259316 Aborted                 (core dumped) "/mnt/data/tools/bin/yosys/bin/yosys-abc" -s -f /tmp/yosys-abc-ARUONx/abc.script 2>&1
ERROR: ABC: execution of command ""/mnt/data/tools/bin/yosys/bin/yosys-abc" -s -f /tmp/yosys-abc-ARUONx/abc.script 2>&1" failed: return code 134.

povik, 2025-04-16:
This looks to be the same issue tracked at The-OpenROAD-project/OpenROAD-flow-scripts#3011

@Cronus-38 you can see the workaround linked therein

Cronus-38, 2025-04-16:
Hi Martin,

Thank you for the quick reply!  Indeed, if I change the cap unit from 0.001pf to 1ff in the .lib then it runs perfectly.

All the best,
Cronus