YosysHQ/yosys #5029: CXXRTL: ERROR: Unsupported internal cell `$bwmux'.

david-sawatzke, 2025-04-16:
### Version

Yosys 0.52+50 (git sha1 c894685f2, clang++ 18.1.8 -fPIC -O3)

### On which OS did this happen?

Linux

### Reproduction Steps

Use the [the current nightly build](https://github.com/YosysHQ/oss-cad-suite-build/releases/tag/2025-04-16) with the [Hazard3)[https://github.com/Wren6991/Hazard3/commit/787da131a1e982543d9b308c1c25a09160e71a65) trying to compile the testbench with yosys-slang with 2 breakpoints (with the [attached diff](https://github.com/user-attachments/files/19773970/hazardslang.diff.txt)) fails.
```
$ cd test/sim/tb_cxxrtl
$ make
```

### Expected Behavior

Compiles

### Actual Behavior

Fails with:
```
ERROR: Unsupported internal cell `$bwmux'.
```

By inserting a `bxmuxmap` pass before `write_cxxrtl` the design compiles.

This is likely fixed by adding `bxmuxmap` somewhere in the [cxxrtl `prepare_design` ](https://github.com/YosysHQ/yosys/blob/c894685f26850a7313dcbf800c306dbab8681517/backends/cxxrtl/cxxrtl_backend.cc#L3430) function.

povik, 2025-04-16:
I have an unupstreamed patch for first class bwmux support on file. @david-sawatzke feel free to use it/open PR for it. I'm not sure whether it applies cleanly.

```
diff --git a/backends/cxxrtl/cxxrtl_backend.cc b/backends/cxxrtl/cxxrtl_backend.cc
index 8dc14863d..08bd68457 100644
--- a/backends/cxxrtl/cxxrtl_backend.cc
+++ b/backends/cxxrtl/cxxrtl_backend.cc
@@ -200,7 +200,7 @@ bool is_extending_cell(RTLIL::IdString type)
 bool is_inlinable_cell(RTLIL::IdString type)
 {
 	return is_unary_cell(type) || is_binary_cell(type) || type.in(
-		ID($mux), ID($concat), ID($slice), ID($pmux), ID($bmux), ID($demux));
+		ID($mux), ID($concat), ID($slice), ID($pmux), ID($bmux), ID($demux), ID($bwmux));
 }
 
 bool is_ff_cell(RTLIL::IdString type)
@@ -1198,6 +1198,14 @@ struct CxxrtlWorker {
 			f << ">(";
 			dump_sigspec_rhs(cell->getPort(ID::S), for_debug);
 			f << ").val()";
+		// Bitwise muxes
+		} else if (cell->type == ID($bwmux)) {
+			dump_sigspec_rhs(cell->getPort(ID::A), for_debug);
+			f << ".bwmux(";
+			dump_sigspec_rhs(cell->getPort(ID::B), for_debug);
+			f << ",";
+			dump_sigspec_rhs(cell->getPort(ID::S), for_debug);
+			f << ").val()";
 		// Demuxes
 		} else if (cell->type == ID($demux)) {
 			dump_sigspec_rhs(cell->getPort(ID::A), for_debug);
diff --git a/backends/cxxrtl/runtime/cxxrtl/cxxrtl.h b/backends/cxxrtl/runtime/cxxrtl/cxxrtl.h
index d1d6bd8dc..edfb07e9a 100644
--- a/backends/cxxrtl/runtime/cxxrtl/cxxrtl.h
+++ b/backends/cxxrtl/runtime/cxxrtl/cxxrtl.h
@@ -498,6 +498,11 @@ struct value : public expr_base<value<Bits>> {
 		return result;
 	}
 
+	CXXRTL_ALWAYS_INLINE
+	value<Bits> bwmux(const value<Bits> &b, const value<Bits> &s) const {
+		return (bit_and(s.bit_not())).bit_or(b.bit_and(s));
+	}
+
 	template<size_t ResultBits, size_t SelBits>
 	value<ResultBits> demux(const value<SelBits> &sel) const {
 		static_assert(Bits << SelBits == ResultBits, "invalid sizes used in demux()");
```

whitequark, 2025-04-16:
I prefer @povik's patch.

david-sawatzke, 2025-04-16:
I just noticed, it's certainly nicer. I'll (try to) open a new pr with his changes

PR #5032 (2025-04-16): cxxrtl: Add internal cell "bwmux"
_What are the reasons/motivation for this change?_

Fixes #5029 

_Explain how this is achieved._

By adding the cell to the cxxrtl backend.

_If applicable, please suggest to reviewers how they can test the change._

Follow the steps in #5029 to create the error.

For a trivial example, create:
```
module bwmux_test(
input [8-1:0] A, B,
input [8-1:0] S,
output [8-1:0] Y
);
	\$bwmux #(.WIDTH(8)) mux (.A(A), .B(B), .S(S), .Y(Y));
endmodule
```
and build the .cpp with 
```
yosys -p "read_verilog -icells test_bwmux.v; select =*; proc; clean; write_cxxrtl cxxrtl-t [... cut]
closed 2025-04-16 by whitequark