YosysHQ/yosys #5038: Failure when building Yosys

eder-matheus, 2025-04-17:
Version: Yosys 0.52+63 (git sha1 088240c03, clang++ 18.1.3 -fPIC -O3)

When updating the Yosys version in OpenROAD-flow-scripts, we had a build error in our public CI. It seems related to abc/cadical:

```
[...]
[ 94%] ABC: `` Compiling: /src/sat/cadical/cadical_version.cpp
[ 94%] ABC: `` Compiling: /src/sat/cadical/cadical_vivify.cpp
src/sat/cadical/cadical_version.cpp:108:30: error: expected ';' after return statement
const char *date () { return DATE; }
                             ^
src/sat/cadical/cadical_version.cpp:90:23: note: expanded from macro 'DATE'
#define DATE __DATE__ " " __TIME__
                      ^
1 error generated.
[...]
```

I guess it is related to the macro `DATE`. It's possible that this name is defined somewhere else by the CI system, leading to the build error:
```
// Build Time and operating system.
//
#ifndef DATE
#define DATE __DATE__ " " __TIME__
#endif
```

widlarizer, 2025-04-30:
@KrystalDelusion were you able to reproduce this? I can't reproduce with clang 16 or 18 on your exact commit, the macro behavior is the same on those, on ubuntu the value is the correct date and time of build, on nix it's 1980-01-01 as it should. Check your CI config and any clang wrappers your distro might be including. I used this test file:
```
#ifndef DATE
#define DATE __DATE__ " " __TIME__
#endif

const char* x = DATE;
```
`clang -E /tmp/a.c` then shows something like `const char* x = "Apr 30 2025" " " "16:41:19";` which is valid string literal concatenation and shouldn't syntax error

KrystalDelusion, 2025-04-30:
I did not, I knew we had run into issues with cadical (with WASI) and thought it might've been related.  I also remember thinking that a malformed __DATE__ macro was the likely culprit, though in hindsight that would probably be a problem with the ORFS build system rather than ours, so I'm not sure why I re-tagged this 😅 
@eder-matheus are you able to provide any other information about this? Maybe a link to the CI failure in question, or at least the OS it's happening on.

eder-matheus, 2025-05-01:
@KrystalDelusion Sure, I'll list some information about our CI configuration:
- We're using docker and ubuntu22.04: `docker pull openroad/flow-ubuntu22.04-dev:fb9b5a`
- Compiler: GNU 11.4.0

I'll try to run it in a public CI so I can share the link with you.

widlarizer, 2025-05-08:
I managed to reproduce it with your docker image and clang on the 0.53 tarball

widlarizer, 2025-05-08:
```
$ echo $CFLAGS
-D__TIME__=0 -D__DATE__=0 -D__TIMESTAMP__=0 -Wno-builtin-macro-redefined
```
Your Docker image has this set, so you have to find out which layer the CFLAGS are coming from. Code that uses the DATE macro expects it to be a string literal, [see gcc docs](https://gcc.gnu.org/onlinedocs/cpp/Standard-Predefined-Macros.html)

closed 2025-05-08 by widlarizer