YosysHQ/yosys #5042: The `value<N>` in CXXRTL does not automatically truncate the value to fit within the bit-width range.

sdjasj, 2025-04-19:
### Version

Yosys 0.52+32 (git sha1 9d3d0a433, clang++ 14.0.0-1ubuntu1.1 -fsanitize=address -fsanitize=undefined -fPIC -O3 -O1 -fno-omit-frame-pointer -fno-optimize-sibling-calls -fsanitize=address,undefined)

### On which OS did this happen?

Linux

### Reproduction Steps

The following is the problematic `main.cpp` code, which I’ve condensed into just a few lines for the developer’s convenience:

```cpp
#include <iostream>
#include <cxxrtl/cxxrtl.h>

using namespace std;

int main() {
	auto a = cxxrtl::value<1>(511u);
	cout << a << endl;
	return 0;
}
```

The issue can reproduce with the following command:

```bash
clang++ -w -g -O3 -std=c++14 -I $(yosys-config --datdir)/include/backends/cxxrtl/runtime main.cpp -o a.out
./a.out
```

I discovered this issue while comparing the output of CXXRTL with Verilator and Icarus Verilog. When a CXXRTL `value<N>` is assigned a value greater than its declared bit-width (in the example above, `a` is a 1-bit wide variable assigned the value `511`), CXXRTL does not automatically truncate the value to fit the bit-width. Both Verilator and Icarus Verilog truncate the value to `1`, but CXXRTL outputs `1'1ff`.

### Expected Behavior

CXXRTL's output should perhaps be consistent with Verilator and Icarus Verilog, i.e., `1'1`.

### Actual Behavior

CXXRTL outputs `1'1ff`.

whitequark, 2025-04-19:
This is on purpose. The generated code truncates values on accesses rather than on assignments, which I hoped would result in slightly better performance.

whitequark, 2025-04-19:
I am open to changing that if you can demonstrate with a benchmark that it doesn't have an outsized negative impact.

sdjasj, 2025-04-19:
Thank you for the prompt reply. It’s indeed a trade-off. At first glance, I think it depends on how frequently the value is read during simulation. I’ll take some time to investigate this further!