YosysHQ/yosys #5043: In some cases, the visibility of outputs in `CXXRTL` requires two calls to `step()`.

sdjasj, 2025-04-20:
### Version

Yosys 0.52+32 (git sha1 9d3d0a433, clang++ 14.0.0-1ubuntu1.1 -fsanitize=address -fsanitize=undefined -fPIC -O3 -O1 -fno-omit-frame-pointer -fno-optimize-sibling-calls -fsanitize=address,undefined

### On which OS did this happen?

Linux

### Reproduction Steps

Below is the Verilog file related to the issue:

```verilog
`timescale 1ns/1ps

module a (
    input  wire       in0,
    input  wire       clock_0,
    input  wire       clock_1,
    output wire [5:0] out1
);

    reg [5:0] reg_0;
    reg [5:0] reg_1;

    initial begin
        reg_0 = 0;
        reg_1 = 0;
    end

    assign out1 = (reg_1 - reg_0);

    always @(posedge clock_0 or posedge clock_1) begin
        if (clock_1) begin
            reg_0 <= 1;
            reg_1 <= 2;
        end else begin
            reg_0 <= 1;
            reg_1 <= 3;
        end
    end

endmodule
```

You can generate the corresponding C++ file using `yosys -p "read_verilog test.v; write_cxxrtl test.cpp"`, and then drive it using the following code:

```cpp
#include <iostream>
#include <vector>
#include <string>
#include "test.cpp"
    
using namespace std;
using namespace cxxrtl_design;

int main() {
	std::unique_ptr<p_a> mod = std::make_unique<p_a>();

	mod->p_in0  = cxxrtl::value<1>(0u);
	mod->p_clock__0  = cxxrtl::value<1>(0u);
	mod->p_clock__1  = cxxrtl::value<1>(0u);
	mod->step();
	cout << mod->p_out1.val() << endl;
	mod->p_clock__0  = cxxrtl::value<1>(1u);
	mod->p_clock__1  = cxxrtl::value<1>(1u);
	mod->step();
	cout << mod->p_out1.val() << endl; // output 6'00
	mod->step();
	cout << mod->p_out1.val() << endl; // output 6'01
	return 0;
}
```

I found that the output values right after two consecutive `step()` calls are not the same. Upon inspecting the generated `test.cpp`, I discovered that it's caused by the register values not being updated in time during a single `eval` call (`p_out1 = sub_uu<6>(p_reg__1.curr, p_reg__0.curr);`):

```cpp
bool p_a::eval(performer *performer) {
	bool converged = true;
	bool posedge_p_clock__0 = this->posedge_p_clock__0();
	// \src: test_new.v:16.1-24.4
	// cell $procdff$9
	if (posedge_p_clock__0) {
		p_reg__1.next = value<6>{0x3u};
	}
	if (p_clock__1 == value<1> {1u}) {
		p_reg__1.next = value<6>{0x2u};
	}
	// \src: test_new.v:16.1-24.4
	// cell $procdff$6
	if (posedge_p_clock__0) {
		p_reg__0.next = value<6>{0x1u};
	}
	if (p_clock__1 == value<1> {1u}) {
		p_reg__0.next = value<6>{0x1u};
	}
	// \src: test_new.v:15.20-15.33
	// cell $sub$test_new.v:15$1
	p_out1 = sub_uu<6>(p_reg__1.curr, p_reg__0.curr);
	return converged;
}
```

I searched through historical issues and found that this might be related to [https://github.com/YosysHQ/yosys/issues/4149](https://github.com/YosysHQ/yosys/issues/4149). However, since that code was not synthesizable, the developers did not investigate it further.

Unlike that issue, my code passes `yosys -p 'hierarchy ; prep' test.v` without any error. I'm now a bit confused—**is it expected behavior** that two consecutive calls to `step()` result in different output states? Because from what I saw in the source code, the purpose of `step()` is to evaluate until convergence.

### Expected Behavior

The outputs of two consecutive `step()` calls should be the same.

### Actual Behavior

The outputs of two consecutive `step()` calls are different.

whitequark, 2025-04-20:
This is a well known issue and it requires re-engineering the core of the simulator. I'll do this eventually, most likely as a part of [prjunnamed](https://github.com/prjunnamed/prjunnamed) (still generating C++).

sdjasj, 2025-04-20:
Thank you for your prompt reply! I will use a few more step() to resolve the current issue.

closed 2025-04-20 by sdjasj