YosysHQ/yosys #5046: std::out_of_range crash in Verilog backend after alumacc pass

danderson, 2025-04-22:
### Version

Yosys 0.52+63 (git sha1 6a2f2f181, clang++  -fPIC -O3)

### On which OS did this happen?

Linux

### Reproduction Steps

- Grab reproducer source [synth.v.txt](https://github.com/user-attachments/files/19842921/synth.v.txt) and yosys script [synth.ys.txt](https://github.com/user-attachments/files/19842926/synth.ys.txt). Fix file extensions by removing `.txt` suffix (github apparently forbids .v and .ys attachments...).
- `yosys -T synth.ys`

The source is not minified I'm afraid, it's a big pile of Verilog that was output by the Bluespec compiler. The yosys script is trivial:

```
read_verilog -sv -defer synth.v
hierarchy -top mkTop
proc
write_verilog -sv synth.before-alumacc.v
write_rtlil synth.before-alumacc.rtlil
alumacc
write_rtlil synth.after-alumacc.rtlil
write_verilog -sv synth.crash.v
```

### Expected Behavior

The script run a few very preliminary synthesis passes, then uses `write_verilog` to write the transformed source back out. I expect it to successfully write the verilog file.

### Actual Behavior

Yosys crashes with:

```
7. Executing Verilog backend.

7.1. Executing BMUXMAP pass.

7.2. Executing DEMUXMAP pass.
terminate called after throwing an instance of 'std::out_of_range'
  what():  Cell::getParam()
```

I originally encountered this with the synth_lattice command, but I bisected it down to something `alumacc` does to the design: `write_verilog` completes successfully before alumacc has run, and crashes as above after it runs.

This does _not_ reproduce on Yosys 0.51. At a glance, `Added $macc_v2 cell` in the 0.52 release notes sounds related (well, has overlapping sequences of characters with the pass that leads to the crash at least...).

The repro synthesis script above also dumps RTLIL before/after alumacc, and that works without issues. For convenience, here's 4 RTLIL outputs:
 - Yosys 0.51, before `alumacc` pass: [0_51.before-alumacc.rtlil.txt](https://github.com/user-attachments/files/19842979/0_51.before-alumacc.rtlil.txt)
- Yosys 0.51, after `alumacc` pass: [0_51.after-alumacc.rtlil.txt](https://github.com/user-attachments/files/19842977/0_51.after-alumacc.rtlil.txt)
- Yosys HEAD (6a2f2f181831a91d23caf32d4429fc0eea389cf8), before `alumacc` pass: [HEAD.before-alumacc.rtlil.txt](https://github.com/user-attachments/files/19842978/HEAD.before-alumacc.rtlil.txt)
- Yosys HEAD (6a2f2f181831a91d23caf32d4429fc0eea389cf8), after `alumacc` pass: [HEAD.after-alumacc.rtlil.txt](https://github.com/user-attachments/files/19842980/HEAD.after-alumacc.rtlil.txt) **(this is the design that crashes `write_verilog`)**

danderson, 2025-04-22:
Oops, I just remembered that `bugpoint` is a thing. Running it now to minimize the crashy input, I'll post updated reproducers and rtlil once it's done.

danderson, 2025-04-22:
Okay that's better, much more minimal repro: grab [minimized.rtlil.txt](https://github.com/user-attachments/files/19843362/minimized.rtlil.txt), then:

```
yosys -p 'read_rtlil minimized.rtlil.txt; hierarchy -top mkTop; alumacc; write_verilog -sv crash.v'
```

And here's the RTLIL after the `alumacc` pass, but before the `write_verilog` that crashes: [after-alumacc.rtlil.txt](https://github.com/user-attachments/files/19843381/after-alumacc.rtlil.txt)

KrystalDelusion, 2025-04-22:
To be clear, this is technically an error in the `demuxmap` pass, not the Verilog backend.

KrystalDelusion, 2025-04-22:
Did a quickfix for the `clean_zerowidth` problem, but I'm not convinced that's actually the issue you were running into with `synth_lattice`.  If possible, I would suggest `bugpoint`ing your `synth_lattice` flow instead of the `alumacc; write_verilog` one.

danderson, 2025-04-22:
The original synthesis script where I hit this crash was:

```
read_verilog -sv -defer synth.v
hierarchy -top mkTop
ynth_lattice -family ecp5 -abc9 -no-rw-check -run :map_ram
opt -full
clean -purge
write_verilog -sv synth_mem.v
```

I was using it to learn/inspect how yosys was handling memory mapping.

That was run on the synth.v linked in the top post. I'm running bugpoint again now, I'll report back if I get a different reproducer!

KrystalDelusion, 2025-04-22:
Ah I see, with the `-run :map_ram` I am more convinced that it is the underlying issue :)

danderson, 2025-04-22:
Bugpoint on the full synth script produces the same minimal rtlil.

I also rebuilt yosys with your patch, and there are no more crashes on any variation ({minimal rtlil, full verilog}x{alumacc only, synth_lattice}). Thank you :)

commit 7f8d0e31 (2025-05-13): Fix #5046

`clean_zerowidth` had skipped $macc, but not $macc_v2
closed 2025-04-22 by povik