YosysHQ/yosys #5050: Error in the opt and the post synthesis testing and simulation issue

dev-ece, 2025-04-22:
I am working on a project and there on simulating the verilog file I am getting the desired result but on running the synth command on yosys I found that all the inputs which were under the if else condition were not mapped, on running the program step by step I noticed that on running the opt command in yosys that error in mapping was happening and on running command like:-
proc; fsm; memory; techmap; read_liberty -lib "Library_Name"; dfflibmap -liberty "Library_Name"; abc -liberty "Library_Name"; flatten
it was running fine, but then also on testing and simulating the synthesized output I was getting all the internals as X or don't care.

Even on running simple dflipflop command as given in:

 https://github.com/dev-ece/examples/blob/main/df.v

I am getting the same output on running synth  or all the command in synth with opt as:

[df.pdf](https://github.com/user-attachments/files/19854255/df.pdf)

And for the same file I am getting the desired gate level netlist on running all command in synth but without opt command as:

[df_without_opt.pdf](https://github.com/user-attachments/files/19854267/df_without_opt.pdf)

But again on running the synthesis with the testbench : 

https://github.com/dev-ece/examples/blob/main/df_test.v,
 
I am getting the desired result with original file but Don't care with synthesised file which is given as :
 
https://github.com/dev-ece/examples/blob/main/df_synthesised.v 

Please provide the solution for the opt command and the post synthesis testing issue

KrystalDelusion, 2025-04-22:
Left (df.dot) is your df.v, right (df_fixed.dot) is the code below, both after `read_verilog` and `proc`.
[image]

```verilog
module df(in,clk,clr,out);
input [12:0]in;
input clk,clr;
output reg [12:0]out;

always @(posedge clk, posedge clr) begin

    if(clr)
        out <= 12'b0;
    else
        out <= in;
end

endmodule
```

The problem here is with your verilog code, not the opt command.
Calling `verilator --lint-only df.v` also gives the following (among others):
```
%Warning-MULTIDRIVEN: bug_5050.v:4:18: Signal has multiple driving blocks with different clocking: 'out'
                      bug_5050.v:8:10: ... Location of first driving block
    8 |  if(clr) out<=12'b0;
      |          ^~~
                      bug_5050.v:14:11: ... Location of other driving block
   14 |   if(clk) out<=in;
      |           ^~~
```

KrystalDelusion, 2025-04-22:
The other warning from verilator:
```
%Warning-WIDTHEXPAND: df.v:8:13: Operator ASSIGNDLY expects 13 bits on the Assign RHS, but Assign RHS's CONST '12'h0' generates 12 bits.
                                     : ... note: In instance 'df'
    8 |  if(clr) out<=12'b0;
      |             ^~
                      ... For warning description see https://verilator.org/warn/WIDTHEXPAND?v=5.033
                      ... Use "/* verilator lint_off WIDTHEXPAND */" and lint_on around source to disable this message.
```
`[12:0]` is 13 bits wide, not 12.

closed 2025-04-22 by KrystalDelusion

KrystalDelusion, 2025-04-22:
From the log of `yosys -p 'read_verilog df.v; synth -top df'`:
```
2.6.7. Executing OPT_CLEAN pass (remove unused cells and wires).
Finding unused cells or wires in module \df..
Warning: Driver-driver conflict for \out [12] between cell $procdff$3.Q and constant 1'0 in df: Resolved using constant.
Warning: Driver-driver conflict for \out [11] between cell $procdff$3.Q and constant 1'0 in df: Resolved using constant.
Warning: Driver-driver conflict for \out [10] between cell $procdff$3.Q and constant 1'0 in df: Resolved using constant.
Warning: Driver-driver conflict for \out [9] between cell $procdff$3.Q and constant 1'0 in df: Resolved using constant.
Warning: Driver-driver conflict for \out [8] between cell $procdff$3.Q and constant 1'0 in df: Resolved using constant.
Warning: Driver-driver conflict for \out [7] between cell $procdff$3.Q and constant 1'0 in df: Resolved using constant.
Warning: Driver-driver conflict for \out [6] between cell $procdff$3.Q and constant 1'0 in df: Resolved using constant.
Warning: Driver-driver conflict for \out [5] between cell $procdff$3.Q and constant 1'0 in df: Resolved using constant.
Warning: Driver-driver conflict for \out [4] between cell $procdff$3.Q and constant 1'0 in df: Resolved using constant.
Warning: Driver-driver conflict for \out [3] between cell $procdff$3.Q and constant 1'0 in df: Resolved using constant.
Warning: Driver-driver conflict for \out [2] between cell $procdff$3.Q and constant 1'0 in df: Resolved using constant.
Warning: Driver-driver conflict for \out [1] between cell $procdff$3.Q and constant 1'0 in df: Resolved using constant.
Warning: Driver-driver conflict for \out [0] between cell $procdff$3.Q and constant 1'0 in df: Resolved using constant.
Removed 1 unused cells and 0 unused wires.
<suppressed ~1 debug messages>
```

It tells you why it's setting the output to all zero and ignoring the other inputs right there.