YosysHQ/yosys #5069: read_verilog: empty parameter value leaks memory

widlarizer, 2025-04-28:
### Version

main

### On which OS did this happen?

Linux

### Reproduction Steps

```
module leaky (
);
 	bar #(.paramname()) barinstance ();
endmodule

module ok (
);
 	bar #() barinstance (.portname());
endmodule
```
Build yosys with SANITIZER=address and `read_verilog bug.sv`

### Expected Behavior

Yosys runs and exits peacefully

### Actual Behavior

See quicklogic leaks in #5020

KrystalDelusion, 2025-04-28:
Is an empty parameter supposed to mean "use the default value"?

widlarizer, 2025-04-29:
That seems to be the intention, but I haven't checked in the standard

KrystalDelusion, 2025-05-04:
Changing this slightly to
```verilog
module bar ( input portname );
parameter paramname = 7;
endmodule

module leaky (
);
 	bar #(.paramname()) barinstance ();
endmodule

module ok (
);
 	bar #() barinstance (.portname());
endmodule

module reference (
	input a
);
	bar #(.paramname(3)) barinstance (.portname(a));
endmodule

```
and calling `hierarchy; dump`, `read_verilog` gives
```
attribute \hdlname "leaky"
attribute \src "bug.sv:5.1-8.10"
module \leaky

  attribute \module_not_derived 1
  attribute \src "bug.sv:7.23-7.37"
  cell \bar \barinstance
  end
end

attribute \hdlname "ok"
attribute \src "bug.sv:10.1-13.10"
module \ok

  attribute \module_not_derived 1
  attribute \src "bug.sv:12.11-12.36"
  cell \bar \barinstance
    connect \portname { }
  end
end

attribute \hdlname "reference"
attribute \src "bug.sv:15.1-19.10"
module \reference

  attribute \src "bug.sv:16.8-16.9"
  wire input 1 \a

  attribute \module_not_derived 1
  attribute \src "bug.sv:18.23-18.49"
  cell \bar \barinstance
    parameter signed \paramname 3
    connect \portname \a
  end
end
```

While `verific` gives
```
attribute \library "work"
attribute \hdlname "reference"
attribute \src "bug.sv:15.8-15.11"
module \reference

  wire $verific$barinstance$bug.sv:18$1

  attribute \src "bug.sv:16.8-16.9"
  wire input 1 \a

  attribute \src "bug.sv:18.23-18.49"
  cell \bar \barinstance
    parameter \paramname 3
    connect \portname \a
  end
end

attribute \library "work"
attribute \hdlname "ok"
attribute \src "bug.sv:10.8-10.10"
module \ok

  attribute \src "bug.sv:12.11-12.36"
  cell \bar \barinstance
    parameter \paramname 7
  end
end

attribute \library "work"
attribute \hdlname "leaky"
attribute \src "bug.sv:5.8-5.13"
module \leaky

  attribute \src "bug.sv:7.23-7.37"
  cell \bar \barinstance
    parameter \paramname 7
  end
end
```

A couple observations; the result is the same with `read -vlog2k` and `read -sv2005`, so I don't think this is SystemVerilog specific (at least, not as far as `read_verilog` or `verific` are concerned).  The second is the inconsistency between the two, where `verific` drops the empty connection to `{ }` but always provides the parameter whether it gets set or not, while `read_verilog` skips the default parameter but retains the empty connection.

PR #5085 (2025-05-05): verilog_parser.y: Delete unused TOK_ID
_What are the reasons/motivation for this change?_

Fix #5069, problem appears to be that the `TOK_ID` is created (and allocated space in memory) but never cleaned up, leading to ASAN detected memory leak.

_Explain how this is achieved._

Delete the `TOK_ID` (`$2`), similarly to other cases such as:
```
non_opt_delay:
	'#' TOK_ID { delete $2; } |
```

_If applicable, please suggest to reviewers how they can test the change._

Test file included as `tests/verilog/param_default.ys`.  Running tests in directory *without* fix fails ASAN on this file and `tes [... cut]
closed 2025-05-05 by widlarizer