YosysHQ/yosys #5074: CXXRTL: Missing sign extension before shift operation for signed values

sdjasj, 2025-04-29:
### Version

Yosys 0.52+117 (git sha1 bfe05965f, clang++ 14.0.0-1ubuntu1.1 -fsanitize=address -fsanitize=undefined -fPIC -O3 -O1 -fno-omit-frame-pointer -fno-optimize-sibling-calls -fsanitize=address,undefined)

### On which OS did this happen?

Linux

### Reproduction Steps

This is the original Verilog file `test.v`:
```verilog
`timescale 1ns/1ps
module cqsnqdxwkr (in2, out0);
reg signed [9:0] reg_7;
input wire in2;
output wire [12:0] out0;

    initial begin
        reg_7 = 848;
    end
  
    assign out0 = reg_7 >> in2;
endmodule
```

The following command is used to convert it into a CXXRTL file:
```
yosys -p "read_verilog test.v; write_cxxrtl test.cpp"
```

Then, the following `main.cpp` is written to perform simulation:
```cpp
#include <iostream>
#include "test.cpp"
    
using namespace std;
using namespace cxxrtl_design;

int main() {
    std::unique_ptr<p_cqsnqdxwkr> mod = std::make_unique<p_cqsnqdxwkr>();
    mod->p_in2 = cxxrtl::value<1>(1u);

    mod->step();

    cout << (mod->p_out0.get<uint32_t>()) << endl;

    return 0;
}
```

Compiled with:
```
clang++ -w -g -O3 -std=c++14 -I $(yosys-config --datdir)/include/backends/cxxrtl/runtime main.cpp -o a.out
```

Run:
```
./a.out
```

The output is:
```
424
```

I believe there is an issue with this result. Here's the analysis:

1. `reg_7` is a 10-bit signed number initialized to 848, which in binary is `10'b11_0101_0000`.

2.  
> The right operand is always treated as an unsigned number and has no effect on the signedness of the result. The result signedness is determined by the left-hand operand and the remainder of the expression.

According to the description on page 53 of the IEEE Std 1364™-2005 standard, the signedness of the expression `reg_7 >> in2` depends on the left operand `reg_7`. Since `reg_7` is signed, the result of the expression is also signed.

3. According to the IEEE Std 1364™-2005 standard (pages 62–66) and the discussion in [[this StackOverflow post](https://stackoverflow.com/questions/23322737/verilog-shift-extending-result)](https://stackoverflow.com/questions/23322737/verilog-shift-extending-result), in the assignment `assign out0 = reg_7 >> in2;`, the width of each operand is determined by the maximum width among the operands (13 for the left-hand side, and 10 for `reg_7` on the right-hand side). Therefore, before the operation, `reg_7` should be **sign-extended** to 13 bits because it is a signed value.

4. CXXRTL generated the following corresponding code:
```cpp
p_out0 = shr_su<13>(value<10>{0x350u}, p_in2);
```
Looking at the implementation of `shr_su` in `cxxrtl.h`:
```cpp
template<size_t BitsY, size_t BitsA, size_t BitsB>
CXXRTL_ALWAYS_INLINE
value<BitsY> shr_su(const value<BitsA> &a, const value<BitsB> &b) {
    return a.shr(b).template scast<BitsY>();
}
```
It can be seen that `a` is **not sign-extended** before the shift operation, which leads to incorrect results.

5. To verify my hypothesis, I tested it with both **Verilator** and **Icarus Verilog (iverilog)**. Both simulators output `4008`, which is the correct result that requires `reg_7` to be sign-extended before the shift.

### Expected Behavior

 Before evaluating `assign a = b op c`, if `b` is a signed value and its bit-width is smaller than the width of the expression, `b` should be sign-extended to match the expression's width.

### Actual Behavior

Currently, this expected behavior is missing

whitequark, 2025-04-29:
Thanks for the report! Do you think you could prepare a fix?

sdjasj, 2025-04-29:
> Thanks for the report! Do you think you could prepare a fix?

Of course! I’m honored.

closed 2025-05-03 by whitequark