YosysHQ/yosys #5076: Parsing long attributes is very slow

rroohhh, 2025-04-29:
### Version

Yosys 0.50+14 (git sha1 077819572222094b30b3cc08e62132456095ca0d,  g++ 11.4.0 -fPIC -O3) 

### On which OS did this happen?

Linux

### Reproduction Steps

Running 
```
yosys -p 'read -sv test.v'
```
against the attached `test.v` takes about 8 seconds on my system: 
```
Time spent: 99% 2x read_verilog (8 sec), 0% 1x read (0 sec)
```
comparing this to the equivalent file in `rtlil` gives 
```
Time spent: 100% 2x read_rtlil (0 sec)
```

(Had to zip the input files for github to allow them)
[test.zip](https://github.com/user-attachments/files/19959661/test.zip)

### Expected Behavior

Parsing the rtlil and the verilog version should have roughly the same performance.

### Actual Behavior

The verilog input file takes drastically longer to parse.

KrystalDelusion, 2025-04-30:
For my system `read_verilog -sv` takes 0.80s vs `verific` at 0.01s. (I didn't test the rtlil but I expect it is similar to the verific).  Which *is* a difference of ~80x, but it *is* a **254kB file**, with 8 attributes that are each 28700 characters long.  The verific also gives a few warnings about the line length:
```
VERIFIC-WARNING [SYS-1203] test.sv:9: column 28720 exceeds maximum value 1023
```
though it does appear to import the whole thing.  I'm inclined to relabel this as a feature request rather than a bug, given that I don't such extremely long strings are likely to be very common and as you showed `read_slang` offers a perfectly acceptable work around if the time taken is actually relevant.

This also isn't exclusive to attributes and instead affects all extremely long strings.

rroohhh, 2025-05-04:
Sure, classifying this as a feature request sounds reasonable. My actual input file is even bigger (~15MB) and after 5 minutes I stopped the yosys process trying to read it, because it looked like it did not make progress at all, so I thought it would be a good idea to atleast document that parsing long strings it currently very slow (compared to `read_slang` or `read_rtlil` that is).

garytwong, 2025-06-01:
I think I have a fix for this; yosys with pull request #5160 applied processes `test.v` in a few milliseconds on my system.

commit 67f8de54 (2025-06-12): Merge pull request #5160 from garytwong/fast-lex

verilog: improve string literal matching speed (fixes #5076)
commit ca7d94af (2025-06-12): verilog: improve string literal matching speed (fixes #5076)

Use a greedy regular expression to match input inside a string
literal, so that flex can accumulate a longer match instead of
invoking a rule for each individual character.
closed 2025-06-12 by KrystalDelusion