YosysHQ/yosys #5077: Flag to avoid writing Yosys version numbers into AIG files.

dinoruic, 2025-04-29:
### Feature Description

Hi all,

I'm using the Yosys's `write_aiger` function to generate AIGs. But Yosys writes its version number into the comment section which makes things non-reproducible every time there is a Yosys update.

Is it possible to add a flag to `write_aiger` that would simply leave the comment section empty?

Thanks!

mikesinouye, 2025-04-29:
Hi @widlarizer and @jix , @dinoruic works with us. Is it possible that the `-T` flag that omits the footer containing the version in the Yosys log file can also be respected in all of the passes, or that this flag can be added to the passes that do so? It seems that there are a few `Generated by <version>` footers per pass.

widlarizer, 2025-05-08:
hope the newly introduced `--no-version` works for you from now on

mikesinouye, 2025-05-08:
This is great! Thank you very much @widlarizer , I will integrate this with our setup.