YosysHQ/yosys #5082: write_verilog: Vivado transparent memory port regression

piotro888, 2025-05-02:
### Version

0.40

### On which OS did this happen?

Linux

### Reproduction Steps

1. Emit Verilog code for memory with one write port and read port that is transparent to write port.

Code is emitted with https://github.com/YosysHQ/yosys/blob/aa30589c123844bc576fc3c9938157b5db35a7c1/backends/verilog/verilog_backend.cc#L732 creating registered address input on read port.

Example:
```verilog
  reg [20:0] tag_mem_0 [127:0];
  always @(posedge clk) begin
    if (tag_mem_wp__en)
      tag_mem_0[tag_mem_wp__addr] <= tag_mem_wp__data;
  end
  reg [6:0] _2_;
  always @(posedge clk) begin
    _2_ <= tag_mem_rp__addr;
  end
  assign tag_mem_rp__data = tag_mem_0[_2_];
```

2. Synthesize it with Vivado

Tested on Vivado 2024.2
This issue comes from Amaranth language: https://github.com/amaranth-lang/amaranth/issues/1589

### Expected Behavior

Vivado synthesizes it correctly to block ram with transparency.

I assume it worked in previous Vivado versions, because this issue was not reported earlier, with successful Xilinx designs made.

### Actual Behavior

Vivado synthesis emits warning `The block RAM may get collision error if read and write address collide` and leads to non-functional gateware.

Workaround is to add `(* rw_addr_collision = "yes" *)` attribute for each memory declaration. Without it, the default behaviour is `design writes to the same address it is reading from, the RAM output is unpredictable`. Warnings are not emitted for non-transparent ports.
This is problematic for HDL generated code, that depends on verilog backend for external synthesis.

whitequark, 2025-05-03:
Do you happen to know which Verilog pattern to emit to have the correct behavior?

piotro888, 2025-05-03:
Apart of using the attribute in current pattern, I'm not aware of any.
Manual "soft" bypass logic (as in https://github.com/kuznia-rdzeni/coreblocks/issues/749) gets sytnetised to LUT RAM.

whitequark, 2025-05-04:
We can't really emit an attribute like this, so to fix this issue, a pattern will need to be found. You should take a look at Vivado documentation which includes these.

piotro888, 2025-05-04:
I found two patterns that work:

A:
```verilog
  reg [20:0] tag_mem_0 [127:0];

  reg [20:0] _4_;
  always @(posedge clk) begin
    _4_ <= tag_mem_wp__data;
    if (tag_mem_wp__en)
      tag_mem_0[tag_mem_wp__addr] <= tag_mem_wp__data;
  end

  reg [6:0] _2_;
  reg  _3_;
  always @(posedge clk) begin
    _3_ <= tag_mem_rp__addr == tag_mem_wp__addr && tag_mem_wp__en;
    _2_ <= tag_mem_rp__addr;
  end

  assign tag_mem_rp__data = _3_ ? _4_ : tag_mem_0[_2_];
```

B:
```verilog
  reg [20:0] tag_mem_0 [127:0];

  reg [20:0] _4_;
  always @(posedge clk) begin
    _4_ <= tag_mem_wp__data;
    if (tag_mem_wp__en)
      tag_mem_0[tag_mem_wp__addr] <= tag_mem_wp__data;
  end

  reg [6:0] _2_;
  reg  _3_;
  always @(posedge clk) begin
    _3_ <= tag_mem_rp__addr == tag_mem_wp__addr && tag_mem_wp__en;
    _2_ <= tag_mem_0[tag_mem_rp_addr];
  end

  assign tag_mem_rp__data = _3_ ? _4_ : _2_;
```

Both A & B synthesize to Block RAM with equivalent circuit and resource utilization as using the attribute (and `_2_` is embedded into memory).
I'm not sure what to expect from optimization/inference in other toolchains, so i guess that B may be better, because bypass logic could not be duplicated (and better represents the resulting logic?).

Doing it directly (like previously from Amaranth) is not recognized by Vivado and get synthesized to Distributed RAM (LUT RAM). It is not recognized by Quartus too. Bypass logic needs to be in external block.
```verilog
  always @(posedge clk) begin
    _2_ <= mem[tag_mem_rp__addr];
    if(tag_mem_rp__addr == tag_mem_wp__addr && tag_mem_wp__en)
      _2_ <= tag_mem_wp__data;
  end
```

Vivado 2023.1 is the first version with `RW_ADDR_COLLISION` in docs (and likely changed behaviour), but of course they didn't add   HDL inference samples for port collision case :/

whitequark, 2025-05-05:
Pattern B looks better to me. @wanda-phi, what do you think?

@piotro888 To clarify, have you tested Pattern B with Quartus?