YosysHQ/yosys #5083: Incorrect constant folding  in `always` block causes missed execution path

sdjasj, 2025-05-03:
### Version

Yosys 0.52+117 (git sha1 bfe05965f, clang++ 14.0.0-1ubuntu1.1 -fsanitize=address -fsanitize=undefined -fPIC -O3 -O1 -fno-omit-frame-pointer -fno-optimize-sibling-calls -fsanitize=address,undefined)

### On which OS did this happen?

Linux

### Reproduction Steps

Below is the Verilog file `test.v` that causes the bug, and I have already minimized it.

```verilog
`timescale 1ns/1ps

module top (
    input  wire signed [21:21] in0,
    input        clock_0,
    input        clock_2
);

    wire clock_0;
    wire clock_2;
    reg signed [28:26] reg_10;

    always @(posedge clock_0 or posedge clock_2) begin
        $display("always block is triggered......");

        if (clock_2) begin
            $display("now in clock_2 == true branch");
        end else begin
            $display("now in clock_2 == false branch");

            if (in0) begin
                reg_10[28:26] <= 2;
            end else begin
                reg_10[28:27] <= 1;
            end
        end
    end

endmodule
```

Run the following commands to optimize it:

```
read_verilog test.v
opt
proc
write_verilog opt.v
```

Yosys generates the corresponding optimized file `opt.v`:

```verilog
/* Generated by Yosys 0.52+117 (git sha1 bfe05965f, clang++ 14.0.0-1ubuntu1.1 -fsanitize=address -fsanitize=undefined -fPIC -O3 -O1 -fno-omit-frame-pointer -fno-optimize-sibling-calls -fsanitize=address,undefined) */

(* cells_not_processed =  1  *)
(* src = "test.v:3.1-29.10" *)
module top(in0, clock_0, clock_2);
  (* src = "test.v:13.5-27.8" *)
  wire [2:0] _00_;
  wire _01_;
  wire [2:0] _02_;
  (* src = "test.v:14.9-14.53" *)
  wire _03_;
  (* src = "test.v:17.13-17.55" *)
  wire _04_;
  (* src = "test.v:19.13-19.56" *)
  wire _05_;
  wire _06_;
  wire _07_;
  (* src = "test.v:5.18-5.25" *)
  input clock_0;
  wire clock_0;
  (* src = "test.v:6.18-6.25" *)
  input clock_2;
  wire clock_2;
  (* src = "test.v:4.32-4.35" *)
  input in0;
  wire in0;
  (* src = "test.v:11.24-11.30" *)
  reg [28:26] reg_10;
  assign _02_ = _01_ ? reg_10 : { 2'h1, _06_ };
  (* src = "test.v:13.5-27.8" *)
  always @(posedge clock_0)
    reg_10 <= _02_;
  assign _06_ = _07_ ? (* full_case = 32'd1 *) (* src = "test.v:21.17-21.20|test.v:21.13-25.16" *) 1'h0 : reg_10[26];
  always @(posedge clock_0 or posedge clock_2) begin
    if (1'h1) begin
      $write("always block is triggered......\n");
    end
    if (1'h0) begin
      $write("now in clock_2 == true branch\n");
    end
    if (1'h1) begin
      $write("now in clock_2 == false branch\n");
    end
  end
  assign _03_ = 1'h1;
  assign _00_[2:1] = 2'h1;
  assign _07_ = in0;
  assign _00_[0] = _06_;
  assign _04_ = 1'h0;
  assign _05_ = 1'h1;
  assign _01_ = clock_2;
endmodule
```

Looking at the following lines in `opt.v`:

```verilog
  always @(posedge clock_0 or posedge clock_2) begin
    if (1'h1) begin
      $write("always block is triggered......\n");
    end
    if (1'h0) begin
      $write("now in clock_2 == true branch\n");
    end
    if (1'h1) begin
      $write("now in clock_2 == false branch\n");
    end
  end
```

We can see that in the original unoptimized code, the following conditional branches should be triggered, but in the optimized code, they are hardcoded and will never be triggered:

```verilog
        if (clock_2) begin
            $display("now in clock_2 == true branch");
```

In fact, this branch should be triggered when `clock_2` has a rising edge.

By the way, during local debugging I found that if the following lines are removed from the original (unoptimized) code, the optimized code no longer exhibits this error. Hope this helps with debugging the issue:

```verilog
            if (in0) begin
                reg_10[28:26] <= 2;
            end else begin
                reg_10[28:27] <= 1;
            end
```

### Expected Behavior

When there is a rising edge on `clock_2`, the `always @(posedge clock_0 or posedge clock_2)` block should be triggered. Inside this block, the condition `if (clock_2)` should evaluate to `true`, and the message `"now in clock_2 == true branch"` should be printed via `$display`.

### Actual Behavior

In the optimized output (`opt.v`), Yosys has replaced `if (clock_2)` with `if (1'h0)`, making the `"now in clock_2 == true branch"` branch unreachable. As a result, even when `clock_2` has a rising edge, this message is never printed. This breaks the functional behavior of the original design.

georgerennie, 2025-05-03:
This looks like an issue in `proc_arst`. The following is a minimal example where EN becomes set to constant `0`.
```
read_rtlil <<EOT
autoidx 5
module \top
  wire \EN
  wire input 2 \clock_a
  wire input 3 \clock_b
  wire \reg_nxt
  wire \reg

  process $proc$test.v:11$1
    assign \reg_nxt \reg
    assign \EN 1'0
    switch \clock_b
      case 1'1
        assign \EN 1'1
      case
        assign \reg_nxt 1'0
    end
    sync posedge \clock_a
      update \reg \reg_nxt
    sync posedge \clock_b
      update \reg \reg_nxt
  end
end
EOT

proc_arst
dump
```

KrystalDelusion, 2025-05-05:
What happens is `proc_arst` checks if any of the edge `sync` signals match a signal in a `switch`, using the example from @georgerennie, this means `clock_b`, which it does.  Then it checks if there are multiple edge `sync`s, which there are.  It updates the `sync` `update` to use the value assigned in the block, here setting `update \reg \reg` on `sync high \clock_b`, because when `clock_b` is high, `reg_next` can be simplified to `reg`.  The problem being it only checks for actions on the `sync` and not the `case`, totally missing that it should be setting `EN` to `1'1` on `clock_b`.  It *also* makes the base case non conditional:
```
    assign \reg_nxt \reg
    assign \EN 1'0
    assign \reg_nxt 1'0
```

Raising an error when it happens isn't too hard:
```diff
diff --git a/passes/proc/proc_arst.cc b/passes/proc/proc_arst.cc
index f01682957..d67d9047c 100644
--- a/passes/proc/proc_arst.cc
+++ b/passes/proc/proc_arst.cc
@@ -118,7 +118,7 @@ void apply_const(RTLIL::Module *mod, const RTLIL::SigSpec rspec, RTLIL::SigSpec
 	}
 }
 
-void eliminate_const(RTLIL::Module *mod, RTLIL::CaseRule *cs, RTLIL::SigSpec const_sig, bool polarity)
+void eliminate_const(RTLIL::Module *mod, RTLIL::CaseRule *cs, RTLIL::SigSpec const_sig, bool polarity, pool<SigSpec> *action_targets = nullptr)
 {
 	for (auto sw : cs->switches) {
 		bool this_polarity = polarity;
@@ -131,6 +131,10 @@ void eliminate_const(RTLIL::Module *mod, RTLIL::CaseRule *cs, RTLIL::SigSpec con
 						goto matched_case;
 				if (found_rem_path) {
 			matched_case:
+					if (action_targets != nullptr)
+						for (auto action : cs2->actions)
+							if (!action_targets->count(action.first))
+								log_error("Unhandled signal %s in switch %s!\n", log_signal(action.first), log_signal(sw->signal));
 					sw->cases.erase(sw->cases.begin() + (i--));
 					delete cs2;
 					continue;
@@ -141,7 +145,7 @@ void eliminate_const(RTLIL::Module *mod, RTLIL::CaseRule *cs, RTLIL::SigSpec con
 			sw->signal = RTLIL::SigSpec();
 		} else {
 			for (auto cs2 : sw->cases)
-				eliminate_const(mod, cs2, const_sig, polarity);
+				eliminate_const(mod, cs2, const_sig, polarity, action_targets);
 		}
 	}
 
@@ -208,7 +212,9 @@ void proc_arst(RTLIL::Module *mod, RTLIL::Process *proc, SigMap &assign_map)
 					sync->type = sync->type == RTLIL::SyncType::STp ? RTLIL::SyncType::ST1 : RTLIL::SyncType::ST0;
 					arst_syncs.push_back(sync);
 					edge_syncs.erase(it);
+					pool<SigSpec> action_targets;
 					for (auto &action : sync->actions) {
+						action_targets.insert(action.second);
 						action.second = apply_reset(mod, proc, sync, assign_map, root_sig, polarity, action.second, action.first);
 					}
 					for (auto &memwr : sync->mem_write_actions) {
@@ -221,7 +227,10 @@ void proc_arst(RTLIL::Module *mod, RTLIL::Process *proc, SigMap &assign_map)
 						apply_reset(mod, proc, sync, assign_map, root_sig, polarity, memwr.data, memwr.data);
 					}
 					sync->mem_write_actions.clear();
-					eliminate_const(mod, &proc->root_case, root_sig, polarity);
+					for (auto action : proc->root_case.actions)
+						if (!action_targets.count(action.first))
+							log_error("Unhandled signal %s in switch %s!\n", log_signal(action.first), log_signal(root_sig));
+					eliminate_const(mod, &proc->root_case, root_sig, polarity, &action_targets);
 				} else {
 					log("Found VHDL-style edge-trigger %s in `%s.%s'.\n", log_signal(sync->signal), mod->name.c_str(), proc->name.c_str());
 					eliminate_const(mod, &proc->root_case, root_sig, !polarity);
```

Though I'm not sure it won't cause other issues.  I haven't managed to figure out how to add the extra assignments to the `sync`, but I'm also not sure how likely this is to be *desirable* behaviour, rather than malformed input that should be rejected.  It also might be better to detect these cases *beforehand* and skip handling them as asynchronous resets, at which point they will almost certainly get rejected with `ERROR: Multiple edge sensitive events found for this signal!` (i.e. if you run this example with all the `proc` commands except `proc_arst`).