YosysHQ/yosys #5084: Add a -noscopeinfo flag to synth_ice40.

vladov3000, 2025-05-04:
### Feature Description

I am running the following yosys program:
```
yosys -m ghdl -Q <<END
ghdl ${dependency_paths[*]} $input_path -e $input;
synth_ice40 -noscopeinfo -json output/$input.json;
END
```
It fails with the following:
```
ERROR: Unable to place cell 'ledtoggle', no BELs remaining to implement cell type '$scopeinfo'
6 warnings, 1 error
```
This is because `synth_ice40` writes json like so:
```
yosys> help synth_ice40
...
    json:
        write_json <file-name>
...
```
However, in the [0.49 release](https://github.com/YosysHQ/yosys/blob/765485a375cc766e9f3e4e9c7bd245c730ad7359/CHANGELOG#L47) `write_json` started exporting `$scopeinfo` cells by default and added the flag `-noscopeinfo` to disable it. To pass that flag into `synth_ice40`, I have to copy the whole `synth_ice40` program just to tweak one flag. 

So I propose adding `-noscopeinfo` to the `synth_ice40` command. This is not a general solution, but one that would help me and probably others. I can implement it myself, but only if a maintainer confirms they are willing to merge the change.

vladov3000, 2025-05-04:
Nevermind, I just fixed it by rerunning write_json with the appropriate flag:
```
yosys -m ghdl -Q <<END
ghdl ${dependency_paths[*]} $input_path -e $input;
synth_ice40 -json output/$input.json;
write_json -noscopeinfo output/$input.json;
END
```
But you should be aware of this regression anyways, because place and route will now fail for modules that instantiate other modules where it wouldn't have before I think:
```
    -- This will make a $scopeinfo that is un-placeable.
    LedToggle : entity work.LedToggle
        port map
            ( clock   => clock
            , switch1 => debounced_switch
            , led1    => led1
            );
```

closed 2025-05-04 by vladov3000

Ravenslofty, 2025-05-04:
That regression was resolved in nextpnr [over a year ago](https://github.com/YosysHQ/nextpnr/commit/8968c84ce650bfb02a69faa74ed99c62a7ac0954).