YosysHQ/yosys #5087: Yosys hangs indefinitely when executing `read_aiger -xaiger` after `abc9` in `synth_intel_alm` flow

sdjasj, 2025-05-05:
### Version

Yosys 0.52+137 (git sha1 aa30589c1, clang++ 14.0.0-1ubuntu1.1 -Og -fPIC -O1 -fno-omit-frame-pointer -fno-optimize-sibling-calls -fsanitize=address,undefined,leak)

### On which OS did this happen?

Linux

### Reproduction Steps

Below is the problematic Verilog file `test.v`:

```
module a(b, c, h);
    input wire [22:17] b;
    input wire c;
    output wire h;
    assign h =  32'ha2f % { 1'b1 ? b : c};
endmodule
```

The issue can be reproduced by running the following command:

```
yosys -p "read_verilog test.v; synth_intel_alm"
```

During the synthesis process, the flow hangs after the following log is printed (it remains stuck for over two hours without completing):

```
2.50.12.5. Executing ABC9.
Running ABC command: "<yosys-exe-dir>/yosys-abc" -s -f <abc-temp-dir>/abc.script 2>&1
ABC: ABC command line: "source <abc-temp-dir>/abc.script".
...
ABC: Total time =     0.03 sec
ABC: + &write -n <abc-temp-dir>/output.aig 
ABC: + &mfs 
ABC: The network is not changed by "&mfs".
ABC: + &ps -l 
ABC: <abc-temp-dir>/input : i/o =     14/      2  and =     367  lev =   91 (45.50)  mem = 0.01 MB  box = 0  bb = 0
ABC: Mapping (K=6)  :  lut =      1  edge =       6  lev =    1 (0.50)  mem = 0.00 MB
ABC: LUT = 1 : 2=0 0.0 %  3=0 0.0 %  4=0 0.0 %  5=0 0.0 %  6=1 100.0 %  Ave = 6.00
ABC: + &write -n <abc-temp-dir>/output.aig 
ABC: + time 
ABC: elapse: 0.09 seconds, total: 0.09 seconds

2.50.12.6. Executing AIGER frontend.

hang....... //no yosys output
```

According to the documentation at [https://yosyshq.readthedocs.io/projects/yosys/en/latest/cmd/synth\_intel\_alm.html#cmd-synth\_intel\_alm](https://yosyshq.readthedocs.io/projects/yosys/en/latest/cmd/synth_intel_alm.html#cmd-synth_intel_alm), I located the problem in the `map_luts` stage of `synth_intel_alm`, specifically the `abc9 -maxlut 6 -W 600` command. Further referencing [https://yosyshq.readthedocs.io/projects/yosys/en/latest/cmd/abc9.html#cmd-abc9](https://yosyshq.readthedocs.io/projects/yosys/en/latest/cmd/abc9.html#cmd-abc9), I found that the issue arises in the following command:

```
read_aiger -xaiger -wideports -module_name <module-name>$abc9 -map <abc-temp-dir>/input.sym <abc-temp-dir>/output.aig
```

The synthesis process hangs when executing this command.

However, when I attempted to minimize this using:

```
bugpoint -runner "timeout 1" -command "read_aiger -xaiger -wideports -module_name <module-name>$abc9 -map <abc-temp-dir>/input.sym <abc-temp-dir>/output.aig"
```

The minimization process terminated immediately and produced the following empty file `min.il`:

```
# Generated by Yosys 0.52+137 (git sha1 aa30589c1, clang++ 14.0.0-1ubuntu1.1 -Og -fPIC -O1 -fno-omit-frame-pointer -fno-optimize-sibling-calls -fsanitize=address,undefined,leak)

autoidx 69031
```

When I tried the following steps on this empty file, it also hung:

```
read_rtlil min.il
read_aiger -xaiger -wideports -module_name <module-name>$abc9 -map <abc-temp-dir>/input.sym <abc-temp-dir>/output.aig
```

The output was:

```
yosys> read_aiger -xaiger -wideports -module_name a$abc9 -map /tmp/yosys-abc-jlXgDz/input.sym /tmp/yosys-abc-jlXgDz/output.aig

3. Executing AIGER frontend.

hang....... //no yosys output
```

### Expected Behavior

Yosys should either successfully complete the synthesis or exit with an error message.

### Actual Behavior

Yosys hangs indefinitely during the `read_aiger` step, producing no output or error messages, and never completes. This occurs even when running the command manually on an empty RTLIL input.

KrystalDelusion, 2025-05-06:
This one is a bit out of my wheelhouse; @jix @Ravenslofty would either of you be able to help?  It reads fine as an aiger instead of xaiger file, though I'm not sure if that means anything.
```
-- Running command `read_aiger -xaiger -wideports -module_name a$abc9 -map /tmp/yosys-abc-K4NSGt/input.sym /tmp/yosys-abc-K4NSGt/output.aig' --

1. Executing AIGER frontend.
M=381 I=14 L=0 O=2 A=367
ciNum = 14
coNum = 2
piNum = 14
poNum = 2
boxNum = 0
m: dataSize=40 lutNum=1 lutSize=6
^C
```
vs
```
-- Running command `read_aiger -wideports -module_name a$abc9 -map /tmp/yosys-abc-K4NSGt/input.sym /tmp/yosys-abc-K4NSGt/output.aig' --

1. Executing AIGER frontend.
M=381 I=14 L=0 O=2 A=367 B=0 C=0 J=0 F=0
Renaming input $i01 -> $mod$test.v:5$1.B
Renaming input $i02 -> $mod$test.v:5$1.B[1]
Renaming input $i03 -> $mod$test.v:5$1.B[2]
Renaming input $i04 -> $mod$test.v:5$1.B[3]
Renaming input $i05 -> $mod$test.v:5$1.B[4]
Renaming input $i06 -> $mod$test.v:5$1.B[5]
Renaming input $i07 -> b[17]
Renaming input $i08 -> b[18]
Renaming input $i09 -> b[19]
Renaming input $i10 -> b[20]
Renaming input $i11 -> b[21]
Renaming input $i12 -> b[22]
Renaming input $i13 -> c
Renaming input $i14 -> h
Renaming output $o0 -> $mod$test.v:5$1.Y
Renaming output $o1 -> h
Finding unused cells or wires in module \a$abc9..
  removing unused non-port wire $aiger1$1.
  removing unused non-port wire $aiger1$2.
  removing unused non-port wire $mod$test.v:5$1.B[1].
  removing unused non-port wire $aiger1$3.
  removing unused non-port wire $mod$test.v:5$1.B[2].
  removing unused non-port wire $aiger1$4.
  removing unused non-port wire $mod$test.v:5$1.B[3].
  removing unused non-port wire $aiger1$5.
  removing unused non-port wire $mod$test.v:5$1.B[4].
  removing unused non-port wire $aiger1$6.
  removing unused non-port wire $mod$test.v:5$1.B[5].
  removing unused non-port wire $aiger1$7.
  removing unused non-port wire $aiger1$8.
  removing unused non-port wire $aiger1$9.
  removing unused non-port wire $aiger1$10.
  removing unused non-port wire $aiger1$11.
  removing unused non-port wire $aiger1$12.
  removing unused non-port wire $aiger1$13.
  removing unused non-port wire $aiger1$14.
  removing unused non-port wire $aiger1$381b.
  removing unused non-port wire $o1.
  removing unused non-port wire $mod$test.v:5$1.B[0].
Removed 0 unused cells and 22 unused wires.
```

Looks like it is getting stuck in this line, https://github.com/YosysHQ/yosys/blob/5924f2de7b706e589b981db9bbbbc5d47fa0094c/frontends/aiger/aigerparse.cc#L442, which is recursive but shouldn't be infinite.  Is there any reason why this might be problematic here?  Having let it run for a few minutes it looks like it might be slowly growing in memory, but it seems to be mostly CPU.

Ravenslofty, 2026-10-19:
Um, for future persons that may be myself: from an exploration, it appears `ConstEvalAig::compute_deps` (where this test case gets stuck) is worst-case exponential on the number of AIG nodes inside a LUT, but the common case is traversing only a handful of AIG nodes, so it's never manifested before.

This test case produces a single LUT6 which covers **701** gates.

Ravenslofty, 2026-11-10:
Fixed by #6411.

closed 2026-11-10 by Ravenslofty