YosysHQ/yosys #5088: Segfault with GCC 15

Xiretza, 2025-05-05:
### Version

Yosys 0.52+139 (git sha1 f60bbe64a, g++ 15.1.1 -march=x86-64 -mtune=generic -O2 -fno-plt -fexceptions -fstack-clash-protection -fcf-protection -fno-omit-frame-pointer -mno-omit-leaf-frame-pointer -ffile-prefix-map=/build/yosys-git/src=/usr/src/debug/yosys-git -flto=auto -fPIC -O3 -fPIC -O3)

### On which OS did this happen?

Linux

### Reproduction Steps

When building the [`yosys-git` AUR package](https://aur.archlinux.org/packages/yosys-git), the testsuite has recently started failing. It seems to be at least incidentally connected to the update to GCC 15.

I've got a crash narrowed down to this:

```
$ echo 'module top(); endmodule' > test.v
$ ./yosys -p 'read_verilog test.v; hierarchy -top top; proc; design -save foo'
```

### Expected Behavior

Yosys does not crash and completes the testsuite successfully.

### Actual Behavior

```
 /----------------------------------------------------------------------------\
 |  yosys -- Yosys Open SYnthesis Suite                                       |
 |  Copyright (C) 2012 - 2025  Claire Xenia Wolf <claire@yosyshq.com>         |
 |  Distributed under an ISC-like license, type "license" to see terms        |
 \----------------------------------------------------------------------------/
 Yosys 0.52+139 (git sha1 f60bbe64a, g++ 15.1.1 -march=x86-64 -mtune=generic -O2 -fno-plt -fexceptions -fstack-clash-protection -fcf-protection -fno-omit-frame-pointer -mno-omit-leaf-frame-pointer -ffile-prefix-map=/build/yosys-git/src=/usr/src/debug/yosys-git -flto=auto -fPIC -O3 -fPIC -O3)
#X# New IdString '\A' with index 1.
#X# New IdString '\abc9_box' with index 2.
#X# New IdString '\abc9_box_id' with index 3.
#X# New IdString '\abc9_box_seq' with index 4.
#X# New IdString '\abc9_bypass' with index 5.
#X# New IdString '\abc9_carry' with index 6.
#X# New IdString '\abc9_flop' with index 7.
#X# New IdString '\abc9_keep' with index 8.
#X# New IdString '\abc9_lut' with index 9.
#X# New IdString '\abc9_mergeability' with index 10.
#X# New IdString '\abc9_scc_id' with index 11.
#X# New IdString '\abcgroup' with index 12.
#X# New IdString '\ABITS' with index 13.
#X# New IdString '\AD' with index 14.
#X# New IdString '\ADDR' with index 15.
#X# New IdString '\allconst' with index 16.
#X# New IdString '\allseq' with index 17.
#X# New IdString '\ALOAD' with index 18.
#X# New IdString '\ALOAD_POLARITY' with index 19.
#X# New IdString '\always_comb' with index 20.
#X# New IdString '\always_ff' with index 21.
#X# New IdString '\always_latch' with index 22.
#X# New IdString '\anyconst' with index 23.
#X# New IdString '\anyseq' with index 24.
#X# New IdString '\ARGS' with index 25.
#X# New IdString '\ARGS_WIDTH' with index 26.
#X# New IdString '\ARST' with index 27.
#X# New IdString '\ARST_POLARITY' with index 28.
#X# New IdString '\ARST_VALUE' with index 29.
#X# New IdString '\A_SIGNED' with index 30.
#X# New IdString '\A_WIDTH' with index 31.
#X# New IdString '\B' with index 32.
#X# New IdString '\BI' with index 33.
#X# New IdString '\BITS_USED' with index 34.
#X# New IdString '\blackbox' with index 35.
#X# New IdString '\B_SIGNED' with index 36.
#X# New IdString '\bugpoint_keep' with index 37.
#X# New IdString '\B_WIDTH' with index 38.
[... 534 lines cut ...]
3.5. Executing PROC_ARST pass (detect async resets in processes).

3.6. Executing PROC_ROM pass (convert switches to ROMs).
Converted 0 switches.

3.7. Executing PROC_MUX pass (convert decision trees to multiplexers).

3.8. Executing PROC_DLATCH pass (convert process syncs to latches).

3.9. Executing PROC_DFF pass (convert process syncs to FFs).

3.10. Executing PROC_MEMWR pass (convert process memory writes to cells).

3.11. Executing PROC_CLEAN pass (remove empty switches from decision trees).
Cleaned up 0 empty switches.

3.12. Executing OPT_EXPR pass (perform const folding).
#X# New Module: top

End of script. Logfile hash: ab2d0b48f2, CPU: user 0.01s system 0.00s, MEM: 25.39 MB peak
Yosys 0.52+139 (git sha1 f60bbe64a, g++ 15.1.1 -march=x86-64 -mtune=generic -O2 -fno-plt -fexceptions -fstack-clash-protection -fcf-protection -fno-omit-frame-pointer -mno-omit-leaf-frame-pointer -ffile-prefix-map=/build/yosys-git/src=/usr/src/debug/yosys-git -flto=auto -fPIC -O3 -fPIC -O3)
Time spent: 38% 1x opt_expr (0 sec), 29% 2x read_verilog (0 sec), ...
corrupted size vs. prev_size in fastbins
fish: Job 1, './yosys --trace -p 'read_verilo…' terminated by signal SIGABRT (Abort)
```

Xiretza, 2025-05-05:
Just noticed that the reproducer hits an abort rather than segfaulting; I have seen several slightly different crashes while reducing this, which makes me think that this is a memory corruption that can lead to many different symptoms depending on exact conditions.

KrystalDelusion, 2025-05-06:
Sounds similar to #4964 (gcc-15, variable errors, probably memory corruption, building for distro, failing on tests).  Unfortunately I'm not seeing anything with ASAN enabled on latest main (`Yosys 0.52+143 (git sha1 5924f2de7, ccache clang++ 18.1.8 -fPIC -O3 -O1 -fno-omit-frame-pointer -fno-optimize-sibling-calls -fsanitize=address)`).  Are you able to get a stack trace out of the crash?

Xiretza, 2025-05-06:
Sure, I've also got it reduced to `CXXFLAGS='-flto=auto'` now:

```
(gdb) bt full
#0  __pthread_kill_implementation (threadid=<optimized out>, signo=signo@entry=6, no_tid=no_tid@entry=0) at pthread_kill.c:44
        tid = <optimized out>
        ret = 0
        pd = <optimized out>
        old_mask = {__val = {140737488348064}}
        ret = <optimized out>
#1  0x00007ffff70f2813 in __pthread_kill_internal (threadid=<optimized out>, signo=6) at pthread_kill.c:89
No locals.
#2  0x00007ffff7098dc0 in __GI_raise (sig=sig@entry=6) at ../sysdeps/posix/raise.c:26
        ret = <optimized out>
#3  0x00007ffff708057a in __GI_abort () at abort.c:73
        act = {__sigaction_handler = {sa_handler = 0x5555572f8c20, sa_sigaction = 0x5555572f8c20}, sa_mask = {__val = {93825023314976, 140737488348192,
              93824997130682, 140737488348208, 93825023314960, 140737488348240, 93824996797260, 140737488348256, 93825023314960, 140737488348304,
              140737338416722, 140737488348272, 93825023314960, 140737488348304, 93825023314960, 140737488348304}}, sa_flags = -450819840,
          sa_restorer = 0x7ffff720d3a2}
#4  0x00007ffff70815c9 in __libc_message_impl (fmt=fmt@entry=0x7ffff720d39f "%s\n") at ../sysdeps/posix/libc_fatal.c:134
        ap = {{gp_offset = 16, fp_offset = 21845, overflow_arg_area = 0x7fffffffe5c0, reg_save_area = 0x7fffffffe550}}
        fd = 2
        iov = {{iov_base = 0x7ffff7210410, iov_len = 40}, {iov_base = 0x7ffff720d3a1, iov_len = 1}, {iov_base = 0x5555572f8bf0, iov_len = 1}, {
            iov_base = 0x7fffffffe520, iov_len = 93825023493440}, {iov_base = 0x7fffffffe540, iov_len = 93824997344061}, {iov_base = 0x7fffffffe540,
            iov_len = 23}, {iov_base = 0x55555732f020, iov_len = 93825023493568}}
        iovcnt = <optimized out>
        total = <optimized out>
        cp = <optimized out>
#5  0x00007ffff70fca35 in malloc_printerr (str=str@entry=0x7ffff7210410 "corrupted size vs. prev_size in fastbins") at malloc.c:5829
No locals.
#6  0x00007ffff70fd8bc in malloc_consolidate (av=av@entry=0x7ffff7241ac0 <main_arena>) at malloc.c:4924
        fb = 0x7ffff7241ae0 <main_arena+32>
        maxfb = 0x7ffff7241b18 <main_arena+88>
        p = 0x5555572fd5a0
        nextp = <optimized out>
        unsorted_bin = 0x7ffff7241b20 <main_arena+96>
        first_unsorted = <optimized out>
        nextchunk = 0x5555572fd670
        size = <optimized out>
        nextsize = <optimized out>
        prevsize = <optimized out>
        nextinuse = <optimized out>
#7  0x00007ffff70fea90 in _int_free_maybe_consolidate (av=0x7ffff7241ac0 <main_arena>, size=<optimized out>) at malloc.c:4836
        __PRETTY_FUNCTION__ = "_int_free_maybe_consolidate"
#8  0x00007ffff70fee2a in _int_free_maybe_consolidate (av=<optimized out>, size=<optimized out>) at malloc.c:4744
        __PRETTY_FUNCTION__ = "_int_free_maybe_consolidate"
        heap = <optimized out>
#9  0x00007ffff70fefdf in _int_free_chunk (av=<optimized out>, p=<optimized out>, size=<optimized out>, have_lock=<optimized out>, have_lock@entry=0)
    at malloc.c:4667
        fb = <optimized out>
#10 0x00007ffff7101e52 in _int_free (av=<optimized out>, p=<optimized out>, have_lock=0) at malloc.c:4699
        size = <optimized out>
#11 __GI___libc_free (mem=<optimized out>) at malloc.c:3476
        ar_ptr = <optimized out>
        p = <optimized out>
        err = 22
#12 0x00005555557fb77d in Yosys::DesignPass::~DesignPass (this=<optimized out>, this=<optimized out>) at passes/cmds/design.cc:33
        it = {first = "foo", second = 0x555557324540}
        __for_range = std::map with 1 element = {["foo"] = 0x555557324540}
        __for_begin = {first = "foo", second = 0x555557324540}
        __for_end = {first = "", second = 0x0}
        it = <optimized out>
        __for_range = <optimized out>
        __for_begin = <optimized out>
        __for_end = <optimized out>
        it = <optimized out>
        __for_range = <optimized out>
        __for_begin = <optimized out>
        __for_end = <optimized out>
#13 0x00007ffff709b5e1 in __run_exit_handlers (status=0, listp=0x7ffff7241680 <__exit_funcs>, run_list_atexit=run_list_atexit@entry=true,
    run_dtors=run_dtors@entry=true) at exit.c:118
        atfct = <optimized out>
        onfct = <optimized out>
        cxafct = <optimized out>
        arg = <optimized out>
        f = <optimized out>
        new_exitfn_called = 878
        cur = 0x5555571dcd40
        restart = <optimized out>
#14 0x00007ffff709b6be in __GI_exit (status=<optimized out>) at exit.c:148
No locals.
#15 0x00007ffff70826bc in __libc_start_call_main (main=main@entry=0x555555bb7ed4 <main(int, char**)>, argc=argc@entry=3, argv=argv@entry=0x7fffffffe8f8)
    at ../sysdeps/nptl/libc_start_call_main.h:74
        result = <optimized out>
        unwind_buf = {cancel_jmp_buf = {{jmp_buf = {0, -4707427482906557298, 140737488349432, 3, 140737354125312, 93825021652888, -4707427483026094962,
                -4707446563368106866}, mask_was_saved = 0}}, priv = {pad = {0x0, 0x0, 0x555555bb18bd <_GLOBAL__sub_I_python_wrappers.cc(void)+14>,
              0x7fffffffe870}, data = {prev = 0x0, cleanup = 0x0, canceltype = 1438324925}}}
        not_first_call = <optimized out>
#16 0x00007ffff7082769 in __libc_start_main_impl (main=0x555555bb7ed4 <main(int, char**)>, argc=3, argv=0x7fffffffe8f8, init=<optimized out>,
    fini=<optimized out>, rtld_fini=<optimized out>, stack_end=0x7fffffffe8e8) at ../csu/libc-start.c:360
No locals.
#17 0x00005555556e9de5 in _start ()
No symbol table info available.
```

KrystalDelusion, 2025-05-07:
https://github.com/YosysHQ/yosys/pull/4502/commits/bf758b909779491c9195c86b76594a5bf9803ef6 may or may not be related: `Makefile: turn off LTO on gcc due to regression`, but looking at #4502 this seems to have been a regression of build performance rather than crashes.  Testing locally with GCC 13.1.0 and `-flto=auto` doesn't reproduce this.

KrystalDelusion, 2025-05-07:
Can you test what happens if you add `design -delete foo` at the end?

Xiretza, 2025-05-08:
1. `design -delete foo` makes the bug go away
2. `--trace` or not makes no difference
3. ASAN actually finds something!

```
==867469==ERROR: AddressSanitizer: heap-use-after-free on address 0x78734fbeb530 at pc 0x5fb833c798dd bp 0x7ffc2914d300 sp 0x7ffc2914d2f0
READ of size 4 at 0x78734fbeb530 thread T0
    #0 0x5fb833c798dc in std::less<unsigned int>::operator()(unsigned int const&, unsigned int const&) const /usr/include/c++/15.1.1/bits/stl_function.h:405
    #1 0x5fb834d73ff7 in std::_Rb_tree<unsigned int, std::pair<unsigned int const, Yosys::RTLIL::Module*>, std::_Select1st<std::pair<unsigned int const, Yosys::RTLIL::Module*> >, std::less<unsigned int>, std::allocator<std::pair<unsigned int const, Yosys::RTLIL::Module*> > >::equal_range(unsigned int const&) /usr/include/c++/15.1.1/bits/stl_tree.h:2642
    #2 0x5fb834d2f9bd in std::_Rb_tree<unsigned int, std::pair<unsigned int const, Yosys::RTLIL::Module*>, std::_Select1st<std::pair<unsigned int const, Yosys::RTLIL::Module*> >, std::less<unsigned int>, std::allocator<std::pair<unsigned int const, Yosys::RTLIL::Module*> > >::erase(unsigned int const&) /usr/include/c++/15.1.1/bits/stl_tree.h:3139
    #3 0x5fb834cf598e in std::map<unsigned int, Yosys::RTLIL::Module*, std::less<unsigned int>, std::allocator<std::pair<unsigned int const, Yosys::RTLIL::Module*> > >::erase(unsigned int const&) /usr/include/c++/15.1.1/bits/stl_map.h:1159
    #4 0x5fb83427a36f in Yosys::RTLIL::Module::~Module() kernel/rtlil.cc:1275
    #5 0x5fb8343c9f70 in Yosys::AST::AstModule::~AstModule() frontends/ast/ast.cc:1480
    #6 0x5fb8357d6ba3 in Yosys::AST::AstModule::~AstModule() frontends/ast/ast.cc:1480
    #7 0x5fb83427577f in Yosys::RTLIL::Design::~Design() kernel/rtlil.cc:920
    #8 0x5fb83441916e in Yosys::DesignPass::~DesignPass() passes/cmds/design.cc:33
    #9 0x7c3350c505e0  (/usr/lib/libc.so.6+0x405e0) (BuildId: 468e3585c794491a48ea75fceb9e4d6b1464fc35)
    #10 0x7c3350c506bd in exit (/usr/lib/libc.so.6+0x406bd) (BuildId: 468e3585c794491a48ea75fceb9e4d6b1464fc35)
    #11 0x7c3350c376bb  (/usr/lib/libc.so.6+0x276bb) (BuildId: 468e3585c794491a48ea75fceb9e4d6b1464fc35)
    #12 0x7c3350c37768 in __libc_start_main (/usr/lib/libc.so.6+0x27768) (BuildId: 468e3585c794491a48ea75fceb9e4d6b1464fc35)
    #13 0x5fb8336a6354 in _start (/home/xiretza/dev/yosys/yosys+0x777c354) (BuildId: bf52c608e4bc8bf67328507bcb0b26de1ca0bc34)

0x78734fbeb530 is located 32 bytes inside of 48-byte region [0x78734fbeb510,0x78734fbeb540)
freed by thread T0 here:
    #0 0x7c3352122b8d in operator delete(void*, unsigned long) /usr/src/debug/gcc/gcc/libsanitizer/asan/asan_new_delete.cpp:155
    #1 0x5fb834de95da in std::__new_allocator<std::_Rb_tree_node<std::pair<unsigned int const, Yosys::RTLIL::Module*> > >::deallocate(std::_Rb_tree_node<std::pair<unsigned int const, Yosys::RTLIL::Module*> >*, unsigned long) /usr/include/c++/15.1.1/bits/new_allocator.h:172
    #2 0x5fb834da2d6c in std::allocator_traits<std::allocator<std::_Rb_tree_node<std::pair<unsigned int const, Yosys::RTLIL::Module*> > > >::deallocate(std::allocator<std::_Rb_tree_node<std::pair<unsigned int const, Yosys::RTLIL::Module*> > >&, std::_Rb_tree_node<std::pair<unsigned int const, Yosys::RTLIL::Module*> >*, unsigned long) /usr/include/c++/15.1.1/bits/alloc_traits.h:649
    #3 0x5fb834da2d6c in std::_Rb_tree<unsigned int, std::pair<unsigned int const, Yosys::RTLIL::Module*>, std::_Select1st<std::pair<unsigned int const, Yosys::RTLIL::Module*> >, std::less<unsigned int>, std::allocator<std::pair<unsigned int const, Yosys::RTLIL::Module*> > >::_M_put_node(std::_Rb_tree_node<std::pair<unsigned int const, Yosys::RTLIL::Module*> >*) /usr/include/c++/15.1.1/bits/stl_tree.h:1191
    #4 0x5fb834d74b91 in std::_Rb_tree<unsigned int, std::pair<unsigned int const, Yosys::RTLIL::Module*>, std::_Select1st<std::pair<unsigned int const, Yosys::RTLIL::Module*> >, std::less<unsigned int>, std::allocator<std::pair<unsigned int const, Yosys::RTLIL::Module*> > >::_M_drop_node(std::_Rb_tree_node<std::pair<unsigned int const, Yosys::RTLIL::Module*> >*) /usr/include/c++/15.1.1/bits/stl_tree.h:1274
    #5 0x5fb834d2fc39 in std::_Rb_tree<unsigned int, std::pair<unsigned int const, Yosys::RTLIL::Module*>, std::_Select1st<std::pair<unsigned int const, Yosys::RTLIL::Module*> >, std::less<unsigned int>, std::allocator<std::pair<unsigned int const, Yosys::RTLIL::Module*> > >::_M_erase(std::_Rb_tree_node<std::pair<unsigned int const, Yosys::RTLIL::Module*> >*) /usr/include/c++/15.1.1/bits/stl_tree.h:2590
    #6 0x5fb83428bac6 in std::_Rb_tree<unsigned int, std::pair<unsigned int const, Yosys::RTLIL::Module*>, std::_Select1st<std::pair<unsigned int const, Yosys::RTLIL::Module*> >, std::less<unsigned int>, std::allocator<std::pair<unsigned int const, Yosys::RTLIL::Module*> > >::~_Rb_tree() /usr/include/c++/15.1.1/bits/stl_tree.h:1609
    #7 0x5fb83429ebcf in std::map<unsigned int, Yosys::RTLIL::Module*, std::less<unsigned int>, std::allocator<std::pair<unsigned int const, Yosys::RTLIL::Module*> > >::~map() /usr/include/c++/15.1.1/bits/stl_map.h:337
    #8 0x7c3350c505e0  (/usr/lib/libc.so.6+0x405e0) (BuildId: 468e3585c794491a48ea75fceb9e4d6b1464fc35)
    #9 0x7c3350c506bd in exit (/usr/lib/libc.so.6+0x406bd) (BuildId: 468e3585c794491a48ea75fceb9e4d6b1464fc35)
    #10 0x7c3350c376bb  (/usr/lib/libc.so.6+0x276bb) (BuildId: 468e3585c794491a48ea75fceb9e4d6b1464fc35)
[... cut ...]
  Heap left redzone:       fa
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
==867469==ABORTING
```

KrystalDelusion, 2025-05-08:
> Yosys::RTLIL::Module::~Module() kernel/rtlil.cc:1275

You're building `WITH_PYTHON`?  That'll probably be why my ASAN missed it.  Just built with that enabled and ASAN does pick it up for me too, so that's good.  It's probably not the only issue hidden in there so it would probably good to give that a pass of the test suite with sanitizers enabled.

It is an optional feature and, as far as I'm aware, is mostly used for `pyosys` builds as opposed to normal `yosys`.  Is there a reason the AUR package needs it?

Xiretza, 2025-05-08:
> Is there a reason the AUR package needs it?

No reason other than "so that it's there". No idea how many users of the package actually use pyosys.

KrystalDelusion, 2025-05-09:
Okay so my local reproduction was actually a false-positive from Python that I was able to get rid of by building cpython from source with `--without-pymalloc` and `--with-address-sanitizer` flags.  (it is also now python3.15 but I don't think that is related).  

Looking at ASAN report though, it appears that `static std::map<unsigned int, RTLIL::Module*> all_modules;` is being deleted 
```
freed by thread T0 here:
    ...
    #7 0x5fb83429ebcf in std::map<unsigned int, Yosys::RTLIL::Module*, std::less<unsigned int>, std::allocator<std::pair<unsigned int const, Yosys::RTLIL::Module*> > >::~map() /usr/include/c++/15.1.1/bits/stl_map.h:337
    #8 0x7c3350c505e0  (/usr/lib/libc.so.6+0x405e0) (BuildId: 468e3585c794491a48ea75fceb9e4d6b1464fc35)
    ...
```
*before* the module, so when the module tries to get a reference to `all_modules` and call the `erase` method we get 
```
ERROR: AddressSanitizer: heap-use-after-free on address 0x78734fbeb530 at pc 0x5fb833c798dd bp 0x7ffc2914d300 sp 0x7ffc2914d2f0
READ of size 4 at 0x78734fbeb530 thread T0
    ...
    #3 0x5fb834cf598e in std::map<unsigned int, Yosys::RTLIL::Module*, std::less<unsigned int>, std::allocator<std::pair<unsigned int const, Yosys::RTLIL::Module*> > >::erase(unsigned int const&) /usr/include/c++/15.1.1/bits/stl_map.h:1159
    #4 0x5fb83427a36f in Yosys::RTLIL::Module::~Module() kernel/rtlil.cc:1275
    #5 0x5fb8343c9f70 in Yosys::AST::AstModule::~AstModule() frontends/ast/ast.cc:1480
    #6 0x5fb8357d6ba3 in Yosys::AST::AstModule::~AstModule() frontends/ast/ast.cc:1480
    #7 0x5fb83427577f in Yosys::RTLIL::Design::~Design() kernel/rtlil.cc:920
    #8 0x5fb83441916e in Yosys::DesignPass::~DesignPass() passes/cmds/design.cc:33
    #9 0x7c3350c505e0  (/usr/lib/libc.so.6+0x405e0) (BuildId: 468e3585c794491a48ea75fceb9e4d6b1464fc35)
    ...
```
Calling `design -delete foo` ensures foo and its modules are deleted before `all_modules` is.

So rather than relying on `~Pass()`, it may be better to use `Pass::on_shutdown()` instead, which is explicitly called during the shutdown process.  `DesignPass` is the only (builtin) pass that uses either method, so it might actually be the only problem.  Could you try with the following patch:

```patch
diff --git a/passes/cmds/design.cc b/passes/cmds/design.cc
index efc72dfb7..a2ae45ef3 100644
--- a/passes/cmds/design.cc
+++ b/passes/cmds/design.cc
@@ -28,7 +28,7 @@ std::vector<RTLIL::Design*> pushed_designs;
 
 struct DesignPass : public Pass {
 	DesignPass() : Pass("design", "save, restore and reset current design") { }
-	~DesignPass() override {
+	void on_shutdown() override {
 		for (auto &it : saved_designs)
 			delete it.second;
 		saved_designs.clear();
```

Xiretza, 2025-05-09:
Yeah, that patch makes the package build and test cleanly again!

KrystalDelusion, 2025-05-11:
@Xiretza can you test with #5114 and see if that also works?

PR #5112 (2025-05-15): design.cc: Use on_shutdown method
_What are the reasons/motivation for this change?_

Fix #5088.

When building `WITH_PYTHON`, where a global list of modules is maintained, deleting a module also erases the entry in said global list.  This can lead to memory corruption if the global list is destructed before the module.

_Explain how this is achieved._

Using `on_shutdown()` instead means the module destructor is explicitly called before the global list can be destructed, preventing the issue.

Also add a comment to `Pass::~Pass()` to suggest the same for future passes that might try to use th [... cut]
closed 2025-05-15 by KrystalDelusion