YosysHQ/yosys #5091: ABC: The command has to terminate. Boxes are not in a topological order.

spth, 2025-05-06:
### Version

Yosys 0.52 (git sha1 fee39a3284c90249e1d9684cf6944ffbbcbb8f90)

### On which OS did this happen?

Linux

### Reproduction Steps

On my Debian GNU/Linux aarch64 machine, I tried this:

`yosys -l icesynth2.yosyslog -p "read_verilog -sv cpu2_nosv.v; synth_ice40 -top cpu -json icesynth2.json"`

[cpu2_nosv.v.gz](https://github.com/user-attachments/files/20054694/cpu2_nosv.v.gz)

### Expected Behavior

yosys synthesizes

### Actual Behavior

I get an error:

```
…
ABC: ** cmd error: aborting 'source <abc-temp-dir>/abc.script'
ABC: The command has to terminate. Boxes are not in a topological order.
ABC: The following information may help debugging (numbers are 0-based):
ABC: Input 0 of BoxA 5 (1stCI = 143; 1stCO = 21) has TFI with CI 992,
ABC: which corresponds to output 0 of BoxB 445 (1stCI = 992; 1stCO = 2097).
ABC: In a correct topological order, BoxB should precede BoxA.
ABC: Error: Abc_CommandAbc9If(): Mapping of GIA has failed.

2.41.16.6. Executing AIGER frontend.
ERROR: Can't open input file `/tmp/yosys-abc-y5d1HF/output.aig' for reading: No such file or directory
```

spth, 2025-05-06:
I was not able to reproduce the issue using yosys 0.52 on a Debian GNU/Linux amd64 machine.

spth, 2025-05-08:
Can't reproduce this on Debian GNU/Linux on the aarch64 machine using yosys 0.52 and yosys 0.53.
So this bug either got fixed after 0.51, or my Verilog code just doesn't trigger it in newer yosys.

spth, 2025-05-15:
Here is a source file that can be used to reproduce the issue using yosys 0.52 on aarch64 (but neither yosys 0.52 on amd64, nor yosys 0.53 on aarch64).

[cpu2_nosv.v.gz](https://github.com/user-attachments/files/20234122/cpu2_nosv.v.gz)

KrystalDelusion, 2025-05-19:
Given that you can't reproduce on newer versions of Yosys, I'm going to go ahead and close this.

KrystalDelusion, 2025-05-29:
@spth Are you able to share the `.aig` file that's causing ABC to fail?  It should be `input.xaig` in the same directory as the `output.xaig` that's failing to read, alternatively you should be able to add `scratchpad -set abc9.debug true` to the start of the yosys script and you should get something like `ABC: + &read _tmp_yosys-abc-vcq6f4/input.xaig` in the log for ABC.

The corresponding RTLIL would also be helpful, 
`yosys -p 'read_verilog -sv cpu2_nosv.v; synth_ice40 -top cpu -run :map_luts; techmap -map +/ice40/latches_map.v; read_verilog -D ICE40_HX -icells -lib -specify +/ice40/abc9_model.v; select =*; write_rtlil -selected bad.il'` should do it.

spth, 2025-06-01:
from the Debian GNU/Linux aarch64 machine:

[bad.il.gz](https://github.com/user-attachments/files/20541561/bad.il.gz)

[input.xaig.gz](https://github.com/user-attachments/files/20541563/input.xaig.gz)

[cpu2_nosv.v.gz](https://github.com/user-attachments/files/20541565/cpu2_nosv.v.gz)

[log.gz](https://github.com/user-attachments/files/20541566/log.gz)

KrystalDelusion, 2025-06-02:
Unfortunately the RTLIL isn't complete enough to reproduce the failing `input.xaig`; it complains about modules not being parametric, and reloading those modules leads to it passing the `abc9 -W 250` call.

I was able to use the `.box` and `.lut` files from the passing run with the `input.xaig` you provided and did get the same error, which does at least mean that the error is in the `xaig` file output by Yosys and not ABC itself.

Can you try this script instead:
```
read_verilog -sv cpu2_nosv.v
synth_ice40 -top cpu -run :map_luts
techmap -map +/ice40/latches_map.v
read_verilog -D ICE40_HX -icells -lib -specify +/ice40/abc9_model.v
abc9 -W 250 -run :exe

tee -o input.lut scratchpad -get abc9_ops.lut_library
tee -o input.box scratchpad -get abc9_ops.box_library
aigmap
write_rtlil -selected abc9.il
design -load $abc9_holes
write_rtlil -selected abc9_holes.il

write_file repro.ys << EOF
!rm -rf tmp; mkdir tmp; cp -t tmp input.box input.lut
read_rtlil abc9_holes.il
design -stash $abc9_holes
read_rtlil abc9.il
write_xaiger tmp/input.xaig
abc9_exe -W 250 -cwd tmp -lut tmp/input.lut -box tmp/input.box
read_aiger -xaiger -wideports -module_name bad$abc9 tmp/output.aig
EOF

exec -expect-return 1 -- yosys repro.ys
```

That'll give `input.lut`, `input.box`, `abc9.il`, and `abc9_holes.il`, which is hopefully enough to reproduce.  If it is, the `repro.ys` script should fail, letting the outer script pass.  Testing locally, the `repro.ys` runs fine, but the xaig file generated does differ slightly compared to if I use `write_xaiger` in the outer script, so there is a non-zero chance even just writing out the il and reading it back is enough of change to make the inner script pass (though in that case I think it would also nail down the problem to `write_xaiger` rather than any of the preparatory work, which would still be useful information).

spth, 2025-06-09:
[abc9.il.gz](https://github.com/user-attachments/files/20653241/abc9.il.gz)
[abc9_holes.il.gz](https://github.com/user-attachments/files/20653238/abc9_holes.il.gz)
[cpu2_nosv.v.gz](https://github.com/user-attachments/files/20653243/cpu2_nosv.v.gz)
[input.box.gz](https://github.com/user-attachments/files/20653240/input.box.gz)
[input.lut.gz](https://github.com/user-attachments/files/20653242/input.lut.gz)
[log.gz](https://github.com/user-attachments/files/20653239/log.gz)

spth, 2025-06-17:
Here is a variant that triggers the bug for me using yosys 0.54 on aarch64.

[cpu2_nosv.v.gz](https://github.com/user-attachments/files/20768090/cpu2_nosv.v.gz)

KrystalDelusion, 2025-06-17:
> by shuffling the opcodes of the soft CPU in the cpu2_nosv.v, I can always find some code that triggers the bug

Just for aarch64 or on any architecture?  That would be super helpful if I could get that to play with locally on amd64

spth, 2025-06-18:
I was not able to reproduce the issue on amd64 using this input.

KrystalDelusion, 2025-07-01:
I am actually able to reproduce this with Yosys v0.52, built locally on Debian x86-64, but *only* when building with clang.  Building with gcc gives a different result from `aigmap` and leads to the design being accepted by ABC, *every* other step produces identical output.  Now to figure out if that's a compiler bug.

KrystalDelusion, 2025-07-01:
> I'm trying to minimize the code that causes the crash, I might be able to share that.

That would be great if you can, but this particular bug tends to be very transient and difficult to minimize.

KrystalDelusion, 2025-07-01:
On debian I still get the error (rather than a sanitizer abort) with both ASAN and UBSAN.  I was also able to confirm that the error is not sensitive to the location of the Yosys executable (and in particular the absolute path of the share directory, since that can affect the names of auto generated signals)

KrystalDelusion, 2025-07-11:
The source of the difference seems to be order of operations in converting gates to AIG.  Take the mux as an example:
```c++
int mux_gate(int A, int B, int S)
{
	//       w        x           y             z
	return or_gate(and_gate(A, not_gate(S)), and_gate(B, S));
}
```

Under gcc (`g++ 12.2.0-14+deb12u1`) these resolve z -> y -> x -> w:
```
z: and 2 0 = 3
y: not 0 = 4
x: and 1 4 = 5
not 3 = 6
not 5 = 7
and 7 6 = 8
w: or 5 3 = 8
mux 1 2 0 = 8
```
Meanwhile clang (`clang++ 14.0.6`) resolves y -> x -> z -> w:
```
y: not 0 = 3
x: and 1 3 = 4
z: and 2 0 = 5
not 4 = 6
not 5 = 7
and 6 7 = 8
w: or 4 5 = 8
mux 1 2 0 = 8
```

This change in order then propagates out and ends up with the cells in a different order.  Which doesn't explain why one order is rejected by ABC while the other is accepted, but it does at least tell us why there is a difference in ordering.

KrystalDelusion, 2025-07-12:
Kept seeing `Boxes are not in a topological order. Switching to level computation without boxes.` pop up from ABC even when it wasn't crashing.  Noticed #3946 changed topological sorting circa 0.35.  Tried Yosys 0.32, no more warning about non topological order.

So, after all that work trying to nail down the difference in ABC input...  I think the problem here is that topological sorting broke some time between 0.32 and 0.44 (#4618), with #3946 being the most likely culprit. It's just only a deal breaker *sometimes*.

KrystalDelusion, 2025-07-12:
Correction, checking out YosysHQ/abc@bb64142 (the version of ABC included with 0.32) and compiling with Yosys 0.52 *also* removes the warning about non topological ordering.  As far as I can tell, if the boxes in question are the ones from `$abc9_holes` (which is what gets referred to as boxes in Yosys) then they've never been topologically sorted.  Oh, and the older version of ABC works fine with the version of the design that was getting rejected.  So 🤷

ShinyKate, 2025-08-20:
ping reminder on this 👋

KrystalDelusion, 2025-09-04:
From https://github.com/berkeley-abc/abc/issues/432, using `&dch -f -r` in place of `&dch -f` is able to get ABC to complete successfully.

Take the following example (based on the design from @spth):
[abc9_repro.tar.gz](https://github.com/user-attachments/files/22162162/abc9_repro.tar.gz)

Running `yosys fail.ys` completes normally, but editing the script to remove the `-r` leads to the error (independent of platform):
```
ABC: ** cmd error: aborting 'source <abc-temp-dir>/abc.script'
ABC: The command has to terminate. Boxes are not in a topological order.
ABC: The following information may help debugging (numbers are 0-based):
ABC: Input 0 of BoxA 372 (1stCI = 859; 1stCO = 1784) has TFI with CI 992,
ABC: which corresponds to output 0 of BoxB 445 (1stCI = 992; 1stCO = 2097).
ABC: In a correct topological order, BoxB should precede BoxA.
ABC: Error: Abc_CommandAbc9If(): Mapping of GIA has failed.
```

The ABC script used in `fail.ys` is based on the default script
```
RTLIL::constpad["abc9.script.default"] = "+&scorr; &sweep; &dc2; &dch -f; &ps; &if {C} {W} {D} {R} -v; &mfs";
```

I was initially having issues with reproducing, so the `{C} {W} {D} {R}` variables are expanded to ` -W 250  ` in the attached script, but it's not necessary to do.

@spth IIRC you had some setup which was able to produce failing variants by moving the opcodes around.  If you still have that lying around, could you run it with the following yosys script to check that the `-r` flag is actually resolving the issue rather than simply moving it?
```
read_verilog -sv cpu2_nosv.v
synth_ice40 -top cpu -run :map_luts
techmap -map +/ice40/latches_map.v
read_verilog -D ICE40_HX -icells -lib -specify +/ice40/abc9_model.v
abc9 -W 250 -run :exe

aigmap
abc9_ops -write_lut input.lut
abc9_ops -write_box input.box

!rm -rf tmp; mkdir tmp
write_xaiger tmp/input.xaig
abc9_exe -W 250 -cwd tmp -lut input.lut -box input.box -script "+&scorr; &sweep; &dc2; &dch -f -r; &ps; &if {C} {W} {D} {R} -v; &mfs"
read_aiger -xaiger -wideports -module_name bad$abc9 tmp/output.aig
```

I can locally confirm (on x86 debian) that the version of `cpu2_nosv.v` you provided for Yosys 0.52 changes from failing to passing if the `-r` flag is used.

closed 2025-10-06 by Ravenslofty

KrystalDelusion, 2025-10-06:
I had a design locally run into this issue, which the suggested fix worked for, so I went ahead and made a PR to apply it (at least on the `default` and `default.area` abc9 scripts).  It was also a much smaller design initially, so I was able to run some minimizers on it and get it stable enough to confirm it is platform independent.