YosysHQ/yosys #5097: std::out_of_range during `write_aiger`

nblei, 2025-05-06:
### Version

Yosys 0.44 (git sha1 80ba43d26, g++ 14.2.0-4ubuntu2~24.04 -fPIC -O3)

### On which OS did this happen?

Linux

### Reproduction Steps

```
$ yosys -s syn.txt
```

syn.txt
```
read_verilog -sv ../sources_1/new/*.sv
hierarchy -check -top Datapath
proc; opt; fsm; opt; memory; opt
techmap;
async2sync;
opt; flatten;
alumacc; dffunmap; aigmap; opt;

write_aiger -zinit -symbols -vmap datapath.aim datapath.aig
```

verilog is from: https://github.com/martinKindall/risc-v-single-cycle commit 5c6bf3c94553fdb00e7622c7ee5e6975fa5a38b4

```bash
git clone --depth 1 https://github.com/martinKindall/risc-v-single-cycle commit temp_dir
cd temp_dir
git fetch --depth 1 origin 5c6bf3c94553fdb00e7622c7ee5e6975fa5a38b4
git checkout 5c6bf3c94553fdb00e7622c7ee5e6975fa5a38b4
mkdir syn
cd syn
echo "read_verilog -sv ../sources_1/new/*.sv \
                                                    hierarchy -check -top Datapath \
                                                    proc; opt; fsm; opt; memory; opt \
                                                    techmap; \
                                                    async2sync; \
                                                    opt; flatten; \
                                                    alumacc; dffunmap; aigmap; opt;\ 

                                                    write_aiger -zinit -symbols -vmap datapath.aim datapath.aig \
                                                    " > syn.txt
yosys -s syn.txt
```

### Expected Behavior

Export and-inverter graph and mapping files, or report error.

### Actual Behavior

uncaught std::out_of_range exception.

nblei, 2025-05-06:
This bug also occurred when using the Ubuntu 24.04 packaged version (yosys 0.33)

widlarizer, 2025-05-07:
Reduced
```v
module bad(
	input in,
	output reg [1:0] out
);
	assign out = {in, 1'b0};
endmodule
```
`read_verilog bug.il; proc; aigmap;  write_aiger -vmap /dev/null /dev/null`

nblei, 2025-05-07:
`read_verilog bug.il; proc; aigmap;  write_aiger -map /dev/null /dev/null`

also crashes

KrystalDelusion, 2025-05-29:
@zhaoxuying can you test with the PR #5109 which is linked on this issue as fixing the problem?  If your case still fails on that PR then post a comment on the PR, not here.  If you're just sharing this code to show that the issue still exists on Yosys 0.50 that is not helpful since there is a fix pending, and 0.53 is the latest not 0.50.

zhaoxuying, 2025-05-30:
The latest version yosys 0.53 still exists the issue. I have post a comment on that PR #5109.

closed 2025-06-02 by widlarizer