YosysHQ/yosys #5099: Incorrect shift optimization when shift amount ≥ `(1 << 32) - 1`

sdjasj, 2025-05-07:
### Version

Yosys 0.53+3 (git sha1 6378ba10e, clang++ 14.0.0-1ubuntu1.1 -Og -fPIC -O1 -fno-omit-frame-pointer -fno-optimize-sibling-calls -fsanitize=address,undefined,leak)

### On which OS did this happen?

Linux

### Reproduction Steps

The following code `test.v` triggers an error:

```
module top (in1, out40);

    input wire  [3:0] in1;
    output wire [6:5] out40;

    assign out40 = (in1[1:1] >> 32'hffffffffffff);
endmodule
```

Running the following command to optimize it:

```
yosys -p "read_verilog test.v; opt; write_verilog bug.v"
```

produces the optimized code `bug.v`:

```
/* Generated by Yosys 0.53+3 (git sha1 6378ba10e, clang++ 14.0.0-1ubuntu1.1 -Og -fPIC -O1 -fno-omit-frame-pointer -fno-optimize-sibling-calls -fsanitize=address,undefined,leak) */

(* cells_not_processed =  1  *)
(* src = "test.v:2.1-8.10" *)
module top(in1, out40);
  (* src = "test.v:3.23-3.26" *)
  input [3:0] in1;
  wire [3:0] in1;
  (* src = "test.v:4.23-4.28" *)
  output [6:5] out40;
  wire [6:5] out40;
  assign out40 = { in1[1], 1'h0 };
endmodule
```

In theory, `in1[1:1] >> 32'hffffffffffff` should be optimized to `2'h0`, but it is instead optimized to `{ in1[1], 1'h0 }`. This error only occurs when the shift amount is greater than or equal to `(1 << 32) - 1`; when the shift amount is less than `(1 << 32) - 1`, the optimization result is correct as `2'h0`.

### Expected Behavior

The expression `in1[1:1] >> 32'hffffffffffff` should be optimized to a constant `2'h0`, since the shift amount far exceeds the width of the operand, making the result definitively zero.

### Actual Behavior

Yosys incorrectly optimizes the expression to `{ in1[1], 1'h0 }`

georgerennie, 2025-05-07:
Ugh I should have caught this in #5066.

We have
```cpp
			// Limit indexing to the size of a, which is behaviourally identical (result is all 0)
			// and avoids integer overflow of i + shift_bits when e.g. ID::B == INT_MAX.
			// We do this after sign-extending a so this accounts for the output size
			shift_bits = min(shift_bits, GetSize(sig_a));
```
but obviously if shift_bits overflows an `int` this is wrong

sdjasj, 2025-05-07:
> I wonder what proportion of the bugs fuzzers have found in the last few years have been exclusively due to `as_int`...

Haha, overflows are always such a headache :)

closed 2025-05-22 by georgerennie