YosysHQ/yosys #5115: Incorrect constant folding of power operator with signed exponent results in loss of 'x propagation

sdjasj, 2025-05-12:
### Version

Yosys 0.53+3 (git sha1 f0c726ff1, clang++ 14.0.0-1ubuntu1.1 -fsanitize=address -fPIC -O3 -O1 -fno-omit-frame-pointer -fno-optimize-sibling-calls -fsanitize=address,undefined)

### On which OS did this happen?

Linux

### Reproduction Steps

The following is the problematic code `test.v`:

```verilog
module top (in2, out7);

    input wire signed [1:0] in2;
    wire wire_3;
    wire signed wire_4;
    output wire out7;

    assign wire_3 = 1'b1;
    assign wire_4 = in2[0:0];
    assign out7 = (wire_3 ** (in2[1:1] ** wire_4));

endmodule
```

I used the following command to optimize the code:

```
yosys -p "read_verilog test.v; opt; proc; write_verilog opt.v"
```

The optimized output `opt.v` is:

```verilog
/* Generated by Yosys 0.53+3 (git sha1 f0c726ff1, clang++ 14.0.0-1ubuntu1.1 -fsanitize=address -fPIC -O3 -O1 -fno-omit-frame-pointer -fno-optimize-sibling-calls -fsanitize=address,undefined) */

(* cells_not_processed =  1  *)
(* src = "test.v:2.1-13.10" *)
module top(in2, out7);
  (* src = "test.v:4.29-4.32" *)
  input [1:0] in2;
  wire [1:0] in2;
  (* src = "test.v:7.17-7.21" *)
  output out7;
  wire out7;
  (* src = "test.v:5.10-5.16" *)
  wire wire_3;
  (* src = "test.v:6.17-6.23" *)
  (* unused_bits = "0" *)
  wire wire_4;
  assign out7 = 1'h1;
  assign wire_3 = 1'h1;
  assign wire_4 = in2[0];
endmodule
```

As shown:

```verilog
assign out7 = (wire_3 ** (in2[1:1] ** wire_4));
```

has been optimized to:

```verilog
assign out7 = 1'h1;
```

However, since `wire_4` is a signed value, its value could be `-1`. When `in2[1:1] == 0`, according to IEEE 1364-2005 Section 5.1.5:

> If neither operand of the power operator is real, then the result type shall be determined as outlined in 5.4.1 and 5.5.1. The result value is 'bx if the first operand is zero and the second operand is negative. The result value is 1 if the second operand is zero.

In this case, `in2[1:1] ** wire_4` would evaluate to `'x'`, so the value of `out7` could potentially be `'x'`. But the optimized code has incorrectly fixed `out7` to a constant value `1`.

### Expected Behavior

`out7` should reflect possible `'x` results when the exponent is negative, as per IEEE 1364-2005 rules.

### Actual Behavior

Yosys optimizes `out7` to a constant `1`, ignoring potential `'x` outcomes from signed power operations.

KrystalDelusion, 2025-05-12:
Can be further reduced 
```
read_verilog << EOF
module top (input wire signed [1:0] a, output wire b);
    assign b = (1'b1 ** (1'b1 ** a));
endmodule
EOF
opt_expr
select -assert-any o:* %ci* i:* %i
```

Changing to `assign b = (1'b1 ** $signed((1'b1 ** a)));` passes assertion, suggesting that the intermediate product is not being detected as signed (and dumping after the read does show this `wire $pow$<<EOF:2$1_Y`).

Changing to `assign b = (1'b1 ** ($signed(1'b1) ** a));` *also* passes assertion, so I guess the question is *should* the output of `(unsigned) ** (signed)` be signed or unsigned?

KrystalDelusion, 2025-05-12:
@Anhijkt can you take a look at this?  Looks like it is related to #4971

povik, 2025-05-12:
> `out7` should reflect possible 'x results when the exponent is negative, as per IEEE 1364-2005 rules.

Note optimization is free to alter the circuit such that it outputs, in place of x-bits, an arbitrary defined value. This is only a bug if it happens with `opt_expr -keepdc`.

KrystalDelusion, 2025-05-12:
Thanks @povik :) I thought there might be something like that.  It does not happen with `opt_expr -keepdc`, so I'll go ahead and close this.

closed 2025-05-12 by KrystalDelusion