YosysHQ/yosys #5126: Yosys equivalence checker produces different outcomes depending on the name of internal variable

tsitdikov, 2025-05-16:
### Version

0.53+3

### On which OS did this happen?

Linux

### Reproduction Steps

Run equivalence check in two scenarios.

- In the 1st scenario the two compared designs have certain internal variable with the same name `sum`, the equivalence check passes.
- In the 2nd scenario we renamed the variable `sum` in one of the designs. Now the equivalence check fails.

The 1st design:

```
module adder_small(clk, enable, rst, a, b, enable2, sum2);
  input wire a;
  input wire b;
  input wire clk;
  input wire enable;
  input wire rst;
  output logic enable2;
  output logic sum2;
  logic spec__02_;
  logic spec__03_;
  wire c0;
  logic sum;
  assign spec__03_ = enable | rst;
  always_ff @(posedge clk)
    enable2 <= enable;
  assign c0 = a + b;
  always_ff @(posedge clk)
    if (spec__03_) sum <= c0;
  assign spec__02_ = enable2 | rst;
  always_ff @(posedge clk)
    if (spec__02_) sum2 <= sum;
endmodule
```

The 2nd design:

```
module adder_small(clk, enable, rst, a, b, enable2, sum2);
  input wire a;
  input wire b;
  input wire clk;
  input wire enable;
  input wire rst;
  output logic enable2;
  output logic sum2;
  reg _00_;
  reg _01_;
  logic impl__02_;
  logic impl__03_;
  logic sum;
  assign impl__03_ = enable | rst;
  always_ff @(posedge clk)
    enable2 <= enable;
  always_ff @(posedge clk)
    if (impl__03_) _01_ <= a;
  always_ff @(posedge clk)
    if (impl__03_) _00_ <= b;
  assign sum = _01_ + _00_;
  assign impl__02_ = enable2 | rst;
  always_ff @(posedge clk)
    if (impl__02_) sum2 <= sum;
endmodule
```

The 2nd design with renamed variable `sum`:

```
module adder_small(clk, enable, rst, a, b, enable2, sum2);
  input wire a;
  input wire b;
  input wire clk;
  input wire enable;
  input wire rst;
  output logic enable2;
  output logic sum2;
  reg _00_;
  reg _01_;
  logic impl__02_;
  logic impl__03_;
  logic impl_sum;
  assign impl__03_ = enable | rst;
  always_ff @(posedge clk)
    enable2 <= enable;
  always_ff @(posedge clk)
    if (impl__03_) _01_ <= a;
  always_ff @(posedge clk)
    if (impl__03_) _00_ <= b;
  assign impl_sum = _01_ + _00_;
  assign impl__02_ = enable2 | rst;
  always_ff @(posedge clk)
    if (impl__02_) sum2 <= impl_sum;
endmodule
```

Equivalence checking script:

```
yosys verific -sv adder_small.1.sv
yosys verific -import -gates -flatten  -no-split-complex-ports adder_small
yosys hierarchy -top adder_small
yosys rename -top gold;
yosys prep -flatten -top gold;
yosys setundef -undriven -anyconst;
yosys rename -witness;
yosys design -stash gold;
yosys verific -sv adder_small.2.sv
yosys verific -import -gates -flatten  -no-split-complex-ports adder_small
yosys hierarchy -top adder_small
yosys rename -top gate;
yosys prep -flatten -top gate;
yosys setundef -undriven -anyconst;
yosys rename -witness;
yosys design -stash gate;
yosys design -copy-from gold -as gold gold;
yosys design -copy-from gate -as gate gate;

yosys equiv_make gold gate equiv;
yosys prep -flatten -top equiv;
yosys opt_clean -purge;
yosys opt -full;
yosys equiv_simple -seq 5;
yosys equiv_struct -maxiter 5;
yosys equiv_induct -seq 5;
yosys equiv_simple -seq 5;
yosys equiv_status;
```

1st outcome:

```
Found 3 $equiv cells in equiv:
  Of those cells 3 are proven and 0 are unproven.
  Equivalence successfully proven!
```

2nd outcome:

```
Found 2 $equiv cells in equiv:
  Of those cells 1 are proven and 1 are unproven.
  Unproven $equiv $auto$equiv_make.cc:257:find_same_wires$61: \sum2_gold \sum2_gate
Found a total of 1 unproven $equiv cells.
```

### Expected Behavior

We expect that equivalence check passes in both cases.

### Actual Behavior

Equivalence check passes in the 1st scenario and fails in the 2nd scenario.

georgerennie, 2025-05-16:
This is an expected behaviour - if the equivalence checker finds a pair of signals in the gold and gate designs with the same names it will assume that it should try to prove them equivalent. Having these internal equivalences makes it stronger and more capable to prove larger designs equivalent (assuming these pairs are actually equivalent). Renaming means it no longer finds this pair for `sum`, meaning that the equivalence solver is unable to solve the problem anymore - this however does not mean that it thinks the designs aren't equivalent, just that it can no longer prove them equivalent as it doesn't have that hint to use. Increasing the depth `-seq` may make it pass.

Note that the restriction that it expects signals with the same names to have the same behaviour is a very strong one because the equivalence checking infrastructure is mainly used for testing yosys rather than being intended to be used by end users. [EQY](https://github.com/YosysHQ/eqy) provides a more complete equivalence checker without this restriction.

closed 2025-05-16 by georgerennie

georgerennie, 2025-05-16:
Oh I looked a little closer at this, and diagnosed a little more about why the proof fails in this case. I believe actually no value of `-seq` will be sufficient here unfortunately. `equiv_induct` uses a variant of k-induction, where it tries to prove "if the designs are equivalent for `seq` cycles, they must be equivalent in the next". Unfortunately, a basic implementation of this as used in `equiv_induct` is weak in some cases.

If the design can be arbitrarily stalled (in this case by holding `enable` low), then the solver can always find a counterexample where it loads different values into `sum` and `impl_sum` for the first `seq` cycles whilst keeping `enable` low, then enables at the end so that their difference in values propagates through to `sum2`. When `equiv_induct` can instead detect that `sum` and `impl_sum` should be equivalent, this counterexample is impossible because they must start out with the same value.