YosysHQ/yosys #5129: proc_init fails on registers declared with constant expressions in range bounds

1353369570, 2025-05-19:
### Version

Yosys 0.53

### On which OS did this happen?

Linux

### Reproduction Steps

## Summary

The `proc_init` pass fails when encountering a register declaration whose range bounds are given using constant expressions, even when the initialization value is a constant.

## Example

This Verilog code is valid and supported by many tools:

```verilog
module example();
  reg signed [(5'h12):(1'h0)] reg10 = (1'h0);
endmodule

### Expected Behavior

yosys -p "read_verilog rtl.v; ; hierarchy； proc_init"

proc_init should recognize this as [18:0] and extract the init attribute.

### Actual Behavior

ERROR: Failed to get a constant init value for \reg10: $1\reg10[18:0]

![Image](https://github.com/user-attachments/assets/397ae68d-7d4a-4f85-9c5f-46b6cebd1324)

KrystalDelusion, 2025-05-19:
The error message says `reg10[18:0]`, so it is recognising the constant expressions just fine.  I was going to suggest the problem may be because the left hand side is 19 bits while the right hand side is only 1 bit, but I can't actually reproduce this with 0.51, 0.53, or 0.53+39 (latest main as of this comment).  Can you share the full output of `yosys --version`?

1353369570, 2025-05-23:
/Desktop/5-12/failed_cases/fuzz_138$ yosys

 /----------------------------------------------------------------------------\
 |  yosys -- Yosys Open SYnthesis Suite                                       |
 |  Copyright (C) 2012 - 2025  Claire Xenia Wolf <claire@yosyshq.com>         |
 |  Distributed under an ISC-like license, type "license" to see terms        |
 \----------------------------------------------------------------------------/
 Yosys 0.53 (git sha1 53c22ab7c0ced80861c7536c5dae682c30fb5834, g++ 9.4.0-1ubuntu1~20.04.3 -fPIC -O3)

yosys> read_verilog rtl.v; hierarchy; proc_init;

1. Executing Verilog-2005 frontend: rtl.v
Parsing Verilog input from `rtl.v' to AST representation.
Generating RTLIL representation for module `\top'.
Successfully finished Verilog frontend.

2. Executing HIERARCHY pass (managing design hierarchy).

3. Executing PROC_INIT pass (extract init attributes).
Found init rule in `\top.$proc$rtl.v:9$8'.
ERROR: Failed to get a constant init value for \reg10: $1\reg10[18:0]

yosys> 

there is the rtl.v:

module top  (y, clk, wire4, wire3, wire2, wire0);
  output wire [(32'h277):(32'h0)] y;
  input wire [(1'h0):(1'h0)] clk;
  input wire signed [(5'h12):(1'h0)] wire4;
  input wire signed [(5'h15):(1'h0)] wire3;
  input wire [(5'h10):(1'h0)] wire2;
  input wire signed [(3'h4):(1'h0)] wire0;
  wire signed [(5'h10):(1'h0)] wire26;
  reg signed [(5'h12):(1'h0)] reg10 = (1'h0);
  wire [(5'h15):(1'h0)] wire9;
  wire [(3'h7):(1'h0)] wire8;
  assign y = {wire26, reg10, wire9, wire8};
  assign wire8 = {(^~(wire0 | $unsigned((wire2 ? wire3 : wire4))))};
  assign wire9 = ((1'h0) < wire2[(4'hb):(3'h6)]);
  always
    @(posedge clk) reg10 <= wire0;
  assign wire26 = wire0[(2'h3):(2'h2)];
endmodule

KrystalDelusion, 2025-05-24:
Okay, so your minimized reproducer doesn't actually reproduce, but this is a bug.  It's also not the bug you said it was, because changing the line to `reg signed [18:0] reg10 = (1'h0);` still raises the error.

Here's a better reproducer
```verilog
module top (input clk, input signed[4:0] wire0);
  reg signed [18:0] reg10 = 0;
  always @(posedge clk) reg10 <= wire0;
endmodule
```

KrystalDelusion, 2025-05-24:
```
read_verilog << EOF
module top (input clk, input signed[4:0] wire0);
  reg signed [18:0] reg10 = 0;
  always @(posedge clk) reg10 <= wire0;
endmodule
EOF
hierarchy
proc
```
Works fine.  You're misusing `proc_init`.

closed 2025-05-24 by KrystalDelusion