YosysHQ/yosys #5151: Incorrect handling of post decrement operation in `always_comb`

toby-bro, 2025-05-30:
### Version

Yosys 0.51+46 (git sha1 8bde6ac17, g++ 12.2.0-14 -fPIC -O3)

### On which OS did this happen?

Linux

### Reproduction Steps

```sv
module expr_postsub_comb (
    input logic [7:0] in_val_m2,
    input logic [7:0] sub_val_m2,
    output logic [7:0] out_diff_m2,
    output logic [7:0] var_out_m2
);
    logic [7:0] var_m2;
      always_comb begin
        var_m2 = in_val_m2;
        out_diff_m2 = (var_m2--) - sub_val_m2;
        var_out_m2 = var_m2;
      end
endmodule
```

sythesize with `yosys -p "read_verilog -sv expr_postsub_comb.sv ; proc; opt ; write_verilog -sv -noattr synth.sv"`

simulate with these values as input
```
  in_val_m2 = ff
  sub_val_m2 = 11
```

### Expected Behavior

output of `out_diff_m2` should be `238`

### Actual Behavior

output of `out_diff_m2` is in fact `236`

georgerennie, 2025-05-30:
This comes about because the verilog frontend treats post operators weirdly - it first applies the operation then tries to apply the inverse to undo the affect of it after, e.g. `b = a++` becomes `a = a + 1; b = a - 1`. This is going wrong in this case because in the following snippet it tries to create `-1`, but actually it just creates a constant that sign extends as `-1` does, but is still interpreted as unsigned once sign extended.

https://github.com/YosysHQ/yosys/blob/d96d1e36302e2bdc5028d601bb06dbcf0c9e4f43/frontends/verilog/verilog_parser.y#L338-L341

One fix for this is to properly create a `-1` by creating an `AST_NEG` node fed by an unsigned `1`, but looking at the output after `opt` it seems that `opt` doesn't fold the `-1` and `+1` together. I think it would be better if I can work out how to fix this by not needing the whole undo thing at all.

toby-bro, 2025-05-30:
I looked at how [verilator](https://github.com/verilator/verilator/blob/master/src/V3LinkInc.cpp) handle's this but from the little I grasped, it seems as if they do something following the same lines...
I agree I am a little surprised that no one ran into this before. 
Thanks for the fix, btw I realised that yosys-slang has a similar problem so I opened an issue there too...