YosysHQ/yosys #5157: Incorrect handling of pre-decrement in `task`

toby-bro, 2025-05-30:
### Version

Yosys 0.53+70 (git sha1 3ef4c91c3, g++ 12.2.0-14+deb12u1 -fPIC -O3)

### On which OS did this happen?

Linux

### Reproduction Steps

```sv
module stmt_if_task (
    output logic [7:0] out_val_m6,
    input logic [7:0] in_val_m6,
    input bit condition_m6
);
    logic [7:0] var_m6;
    task automatic update_conditional_m6(input bit cond, inout logic [7:0] val);
        if (cond) begin
            val++;
        end else begin
            --val;
        end
    endtask
    always_comb begin
        var_m6 = in_val_m6;
        update_conditional_m6(condition_m6, var_m6);
        out_val_m6 = var_m6;
    end
endmodule
```

Simulate with these hexadecimal values as input
```txt
  in_val_m6 = 79 // or 121 in base10
  condition_m6 = 0
```

Synthesize with
```bash
yosys -p "read_verilog -sv stmt_if_task.sv ; proc; opt ; write_verilog -sv -noattr synth.sv"
```

### Expected Behavior

output of `out_val_m6` should be `120`

### Actual Behavior

output of `out_val_m6` is in fact `121`

georgerennie, 2025-05-30:
I think the actual underlying issue here is that the values assigned to `inout` parameters are not being copied out of tasks correctly (as mentioned at the top of page 337 in 1800-2023). See the following reproducer:
```verilog
read_verilog -sv <<EOT
module top (
	output logic [7:0] out
);
	task automatic set_to_5(inout logic [7:0] val);
		val = 5;
	endtask

	always_comb begin
		out = 0;
		set_to_5(out);
	end

	always_comb assert(out == 5);
endmodule
EOT

prep
chformal -lower
sat -prove-asserts -verify
```

toby-bro, 2025-05-31:
Ok sorry, I didn't know it was possible to write such yosys scripts. I'll do so for the next bugs (if there are any).

georgerennie, 2025-05-31:
> Ok sorry, I didn't know it was possible to write such yosys scripts. I'll do so for the next bugs (if there are any).

No worries, a lot of the techniques for this are slightly arcane knowledge. If you look in the `tests` directory you will see many testcases in `.ys` files that are written in this style which may serve as inspiration.

closed 2025-06-04 by georgerennie