YosysHQ/yosys #5169: write_xaiger2: out of range exception

Ravenslofty, 2025-06-06:
### Version

Yosys 0.53+70 (git sha1 e046e3cdb, g++ 15.1.1 -fPIC -O3)

### On which OS did this happen?

Linux

### Reproduction Steps

```
read_rtlil <<EOF

# Generated by Yosys 0.53+98 (git sha1 780b12271, g++ 15.1.1 -fPIC -O3)
autoidx 30
attribute \top 1
attribute \src "top.v:1.1-21.10"
module \top
  attribute \src "top.v:7.15-7.18"
  wire output 1 \led
  attribute \src "top.v:9.8-9.9"
  wire \w
  attribute \src "top.v:20.16-20.18"
  cell $not $not$top.v:20$1
    parameter \A_SIGNED 0
    parameter \A_WIDTH 1
    parameter \Y_WIDTH 1
    connect \A \w
    connect \Y \led
  end
  attribute \module_not_derived 1
  attribute \src "top.v:10.10-18.4"
  cell \CC_MX4 \mux
    connect \D0 1'x
    connect \D1 1'x
    connect \D2 1'x
    connect \D3 { }
    connect \S0 1'x
    connect \S1 1'x
    connect \Y \w
  end
end

EOF

read_verilog -lib -specify <<EOF

(* abc9_box, lib_whitebox *)
module CC_MX4 (
        input  D0, D1, D2, D3,
        input  S0, S1,
        output Y
);
        specify
                (D0 => Y) = 453;
                (D1 => Y) = 449;
                (D2 => Y) = 488;
                (D3 => Y) = 484;
                (S0 => Y) = 422;
                (S1 => Y) = 385;
        endspecify

        assign Y = S1 ? (S0 ? D3 : D2) :
                        (S0 ? D1 : D0);

endmodule

EOF

abc_new
```

### Expected Behavior

It doesn't crash.

### Actual Behavior

```
3.4.2. Executing XAIGER2 backend.
terminate called after throwing an instance of 'std::out_of_range'
  what():  vector::_M_range_check: __n (which is 0) >= this->size() (which is 0)
Aborted (core dumped)
```

georgerennie, 2025-06-07:
The bit xaiger2 takes issue with is `connect \D3 { }`, it (or maybe bufnorm) seems to assume this is impossible

KrystalDelusion, 2025-06-07:
Does `connect \D3 { }` get cleaned up by one of the `opt` passes?  That feels like it should be equivalent to `1'x`.

> The state of this stuff seems experimental enough to me that i seem to hit lots of assertion failures and similar so idk how to go about satisfying fixes for this that aren't just "make bufnorm/aiger2 handle all the (not so) corner cases".

That was my initial thought too, and last time it came up (#5005) I closed the issue as not planned (mostly because I wanted to dissuade the user from fuzzing experimental passes).

Ravenslofty, 2025-06-07:
So, I agree that fuzzing experimental passes is not necessarily desirable, but simultaneously, having issues about the current state of the passes is useful for assessing the viability of using them. 

(And on a personal note, I would like to avoid a repeat of the four-year gap between 1098 (ABC9 being introduced) and 4019 (ABC9 being made default))

ShinyKate, 2025-11-17:
@Ravenslofty can you recheck to see if this is still an issue?