YosysHQ/yosys #5175: Regression in `opt_expr` in Yosys 0.54

whitequark, 2025-06-10:
### Version

Yosys 0.54 (git sha1 db72ec3bd, ccache clang 18.1.3 -O3 -flto -flto)

### On which OS did this happen?

Linux

### Reproduction Steps

Reproducer: [video-rgb-input.il.txt](https://github.com/user-attachments/files/20663224/video-rgb-input.il.txt)

```
yosys video-rgb-input.il -p proc
```

### Expected Behavior

Works (which it did in 0.53).

### Actual Behavior

```
...

2.12. Executing OPT_EXPR pass (perform const folding).
Optimizing module top.cd_sync.clk_buf.
Optimizing module top.cd_sync.
Optimizing module top.U$5.U$7.
Optimizing module top.U$5.U$6.
Optimizing module top.U$5.U$5.
Optimizing module top.U$5.U$4.
Optimizing module top.U$5.dck_buffer.io.
Optimizing module top.U$5.dck_buffer.
Optimizing module top.U$5.b_buffer.io.
Optimizing module top.U$5.b_buffer.
Optimizing module top.U$5.g_buffer.io.
Optimizing module top.U$5.g_buffer.
Optimizing module top.U$5.r_buffer.io.
Optimizing module top.U$5.r_buffer.
Optimizing module top.U$5.
<suppressed ~7 debug messages>
ERROR: Assert `!empty()' failed in ../yosys-src/kernel/rtlil.cc:5532.
```

whitequark, 2025-06-10:
It crashes here:

https://github.com/YosysHQ/yosys/blob/c561ba84e37991ff9f1d4404308d4d5d6f06ec25/passes/opt/opt_expr.cc#L1307-L1315

Presumably because of this cell:

```
  cell $shr $85
    parameter \A_SIGNED 0
    parameter \B_SIGNED 0
    parameter \A_WIDTH 15
    parameter \B_WIDTH 0
    parameter \Y_WIDTH 15
    connect \A \pixel [14:0]
    connect \B {  }
    connect \Y $20
  end
```

Presumably introduced in: 0dcd94b6ad22612e884cb55337c1ec0441715805 (https://github.com/YosysHQ/yosys/pull/5101)

KrystalDelusion, 2025-06-23:
> Any chance we could get a quick point release to fix this?

FWIW Yosys does (minor point) releases every month, and as far as I know has never made a release for a patch

PR #5182 (2025-07-08): kernel: treat zero width constant as zero
_What are the reasons/motivation for this change?_

Fixes #5175 

_Explain how this is achieved._

Zero width constants are special cased and treated as the constant value zero.

_If applicable, please suggest to reviewers how they can test the change._

The testcase from #5175