YosysHQ/yosys #5207: Functional backend error when bitwidths too wide

gussmith23, 2025-07-02:
### Version

Yosys 0.53+39 (git sha1 d8b27d41c, c++ 16.0.0 -fPIC -O3)

### On which OS did this happen?

macOS

### Reproduction Steps

Use the file: 
```sv
(* use_dsp = "yes" *) module top(
	input  [17:0] a,
	input  [17:0] b,
	input  [17:0] c,
	input  [17:0] d,
	output [17:0] out,
	input clk);

	logic  [35:0] stage0;
	logic  [35:0] stage1;

	always @(posedge clk) begin
	stage0 <= ((d + a) * b) + c;
	stage1 <= stage0;
	end

	assign out = stage1;
endmodule
```

Run the command:
```
 yosys -q p 'read_verilog -sv file.sv; hierarchy -top top; prep; clk2fflogic; write_functional_rosette -provides'
```

### Expected Behavior

No error, converts to Rosette

### Actual Behavior

ERROR: The design contains an undriven signal stage1. This is not supported by the functional backend. Call setundef with appropriate options to avoid this error. 

Note that shrinking the  registers to the correct size fixes the issue. However, I wouldn't expect this to be something that would cause a failure. Is there a way around this?

gussmith23, 2025-07-02:
Oh my gosh, I didn't see that the solution was right in the error message. It was being clipped in my terminal 🤦 I'll try out setundef.

nakengelhardt, 2025-07-03:
This made me realize that some of the important documentation about the expectations of the functional backend ended up only in a PDF we had to send to Sandia; the first page of this doc should really be added to the [functional IR docs page](https://yosyshq.readthedocs.io/projects/yosys/en/latest/yosys_internals/extending_yosys/functional_ir.html), ideally with the suggested yosys commands for achieving that design state consistently...

[Interface Document Functional Backend.pdf](https://github.com/user-attachments/files/21033438/Interface.Document.Functional.Backend.pdf)

The concrete problem here is that `wreduce` (which is part of `prep`) recognizes that half of the intermediary signals isn't actually used, and optimizes the bitwidth of all the operators, which leaves the upper half of `stage0` and `stage1` (and some privately-named intermediate signals) undriven. (And partially-undriven is not supported under the "no undriven signals" requirement.)

`clean` doesn't cut down the width of unused and undriven signals when they have public names (but it will helpfully add a `unused_bits` attribute telling you about it); `clean -purge` will do so for `stage0` and `stage1` but not for an intermediate wire, and I'm honestly not sure why:
```
  attribute \src "functional.v:13.13-13.24"
  attribute \unused_bits "18 19 20 21 22 23 24 25 26 27 28 29 30 31 32 33 34 35"
  wire width 36 $auto$wreduce.cc:514:run$9
[...]
  attribute \src "functional.v:13.12-13.29"
  cell $add $add$functional.v:13$4
    parameter \A_SIGNED 0
    parameter \A_WIDTH 18
    parameter \B_SIGNED 0
    parameter \B_WIDTH 18
    parameter \Y_WIDTH 18
    connect \A $auto$wreduce.cc:514:run$9 [17:0]
    connect \B \c
    connect \Y $0\stage0[35:0]
  end
  attribute \src "functional.v:13.13-13.24"
  cell $mul $mul$functional.v:13$3
    parameter \A_SIGNED 0
    parameter \A_WIDTH 19
    parameter \B_SIGNED 0
    parameter \B_WIDTH 18
    parameter \Y_WIDTH 18
    connect \A $add$functional.v:13$2_Y
    connect \B \b
    connect \Y $auto$wreduce.cc:514:run$9 [17:0]
  end
```

Calling `setundef -undriven -undef` does solve the immediate issue, but I am a bit concerned that I can't figure out how to get rid of that last useless piece of wire!

An alternate option that may make sense in formal contexts is to run `prep -ifx`, which avoids running `wreduce` (that also changes X behavior in `if` blocks, depends on context whether that is something you want here).

georgerennie, 2025-07-03:
Its also worth noting that `setundef -undriven -undef` doesnt actually preserve the semantics for those bits of `stage1`, so if you were to try to do things like generate model checking counterexamples from this output you could see values for those bits in the trace that aren't allowed by the original verilog.

widlarizer, 2025-08-29:
Yep, `prep` runs `wreduce` but due to bad pass sequencing it doesn't `opt_clean` the old full width wires

Other repro: `read_verilog tests/functional/picorv32.v; prep -top picorv32; clk2fflogic; write_functional_cxx`

widlarizer, 2025-08-29:
I don't think sequencing is to blame exactly, but this shows we're doing _something_ wrong:
```
2.8. Executing WREDUCE pass (reducing word size of cells).
Removed top 17 bits (of 36) from port Y of cell top.$add$garbage/slice.v:13$2 ($add).
Removed top 17 bits (of 36) from port A of cell top.$mul$garbage/slice.v:13$3 ($mul).
Removed top 18 bits (of 36) from FF cell top.$procdff$6 ($dff).
Removed top 18 bits (of 36) from FF cell top.$procdff$5 ($dff).
Removed top 18 bits (of 36) from port Y of cell top.$add$garbage/slice.v:13$4 ($add).
Removed top 18 bits (of 36) from port A of cell top.$add$garbage/slice.v:13$4 ($add).
Removed top 18 bits (of 36) from port Y of cell top.$mul$garbage/slice.v:13$3 ($mul).
Removed top 18 bits (of 36) from wire top.$0\stage0[35:0].
old name $auto$wreduce.cc:514:run$7
Removed top 17 bits (of 36) from wire top.$add$garbage/slice.v:13$2_Y.
old name $auto$wreduce.cc:514:run$8
Removed top 18 bits (of 36) from wire top.$mul$garbage/slice.v:13$3_Y.
old name $auto$wreduce.cc:514:run$9
Removed top 18 bits (of 36) from wire top.stage0.
old name $auto$wreduce.cc:514:run$10

2.9. Executing OPT_CLEAN pass (remove unused cells and wires).
Finding unused cells or wires in module \top..
  removing unused non-port wire $mul$garbage/slice.v:13$3_Y.
  removing unused non-port wire $auto$wreduce.cc:514:run$7.
  removing unused non-port wire $auto$wreduce.cc:514:run$8.
  removing unused non-port wire $auto$wreduce.cc:514:run$10.
Removed 0 unused cells and 4 unused wires.

2.10. Executing WREDUCE pass (reducing word size of cells).
Removed top 18 bits (of 36) from wire top.$auto$wreduce.cc:514:run$9.
old name $auto$wreduce.cc:514:run$11

2.11. Executing OPT_CLEAN pass (remove unused cells and wires).
Finding unused cells or wires in module \top..
  removing unused non-port wire $auto$wreduce.cc:514:run$9.
Removed 0 unused cells and 1 unused wires.

2.12. Executing WREDUCE pass (reducing word size of cells).
Removed top 18 bits (of 36) from wire top.$auto$wreduce.cc:514:run$11.
old name $auto$wreduce.cc:514:run$12

2.13. Executing OPT_CLEAN pass (remove unused cells and wires).
Finding unused cells or wires in module \top..
  removing unused non-port wire $auto$wreduce.cc:514:run$12.
Removed 0 unused cells and 1 unused wires.
```

This is what I see when in `prep` I just add a couple `opt_clean; wreduce` iterations by hand

widlarizer, 2025-08-29:
`wreduce` refuses to reduce width of wires connected to output ports. Compare the pass run on
```
module \top
  wire width 18 input 1 \a
  wire width 18 output 2 \out

  wire width 18 $stage0
  wire width 36 $stage1

  connect $stage0 \a

  cell $not $asdfgh
    parameter \A_WIDTH 18
    parameter \A_SIGNED 0
    parameter \Y_WIDTH 18
    connect \A $stage0
    connect \Y $stage1 [17:0]
  end

  cell $not $qwertyui
    parameter \A_WIDTH 18
    parameter \A_SIGNED 0
    parameter \Y_WIDTH 18
    connect \A $stage1 [17:0]
    connect \Y \out
  end
end
```
vs
```
module \top
  wire width 18 input 1 \a
  wire width 18 output 2 \out

  wire width 18 $stage0
  wire width 36 $stage1

  connect $stage0 \a

  cell $not $asdfgh
    parameter \A_WIDTH 18
    parameter \A_SIGNED 0
    parameter \Y_WIDTH 18
    connect \A $stage0
    connect \Y $stage1 [17:0]
  end

  connect \out $stage1 [17:0]
end
```