YosysHQ/yosys #5210: `log_id_cache` memory leaks are problematic

rocallahan, 2025-07-03:
Every time `log_id()` is called the string is leaked into the `log_id_cache` until `log_pop()` or `log_reset_stack()` is called, typically at the end of a pass but not always even then. For passes that log a lot, this is bad. For some passes the situation is worse because they use `log_id()` in code not related to logging. For example:
* `CheckPass` uses `log_id()` to [build strings](https://github.com/YosysHQ/yosys/blob/7b0c1fe491fd0c90b8b513a89cf033406c620185/passes/cmds/check.cc#L149) even if it ends up not reporting any warnings.
* `SatGen::importSigSpecWorker()` (called by `SharePass`, at least) uses `log_id()` to build temporary strings used for lookups. Lots of lookups.

This causes memory usage to increase steadily during each pass, and for some passes very rapidly. When this string-copying was introduced, the performance impact was expected to be limited because `log_id()` was [expected to be used only for logging](https://github.com/YosysHQ/yosys/issues/1178#issuecomment-511362071), but see above.

whitequark, 2025-07-03:
From memory, I suspect it's because it's used together with `printf()` style functions, and perhaps doing the equivalent `std::string("foo").c_str()` doesn't keep the string alive long enough?

rocallahan, 2025-07-03:
No problem!

rocallahan, 2025-07-03:
One way to tackle this would be to make `log_id()` a macro like so:
```
std::string log_id_internal(const RTLIL::IdString &id);
#define log_id(id) log_id_internal(id).c_str()
```
And then audit every single call to `log_id()` to make sure it's OK.

Or if we're going to audit `log_id()` anyway, it could continue to return a `char*` and we could require that the `id` parameter be kept alive as long as the `char*` result. I would expect that to normally be the case since it's typically the `name` field of some `RTLIL` object which we shouldn't be destroying or renaming during a logging call!

Slinging `char*`s around and having to worry about their lifetimes is a bit fragile though. So another way would be to reimplement all the `log()` related functions to use C++ variadic templates, so you can print `std::string` directly, and make `log_id()` return an `std::string`. That is unfortunately a lot of work, especially if you want to keep compile-time format string checking comparable to what you currently have via the compiler's built-in checking. I'm not sure it's really practical in C++17.

whitequark, 2025-07-03:
We should have ASAN builds on CI, right? If we do, then with sufficient test coverage it's probably fine to leave discovering the lifetime issues to ASAN. Or maybe not, I'm not sure how good the test coverage is.

widlarizer, 2025-07-08:
Thanks for the report!

As additional clarification, this isn't a memory leak as I know the concept as at all times all mentioned allocated objects are pointed to by something and deallocated at some point by the program. An orthodox memory leak would be detected and reported by a sanitizer at shutdown. Wikipedia calls this a "space leak" which is a word I just learned now

widlarizer, 2025-11-20:
As things stand now, how hard can we gut `log_id` and replace it with just `unescape_id` to reduce the actual dependence on this mechanism?

rocallahan, 2025-11-27:
I want to implement [this](https://yosyshq.discourse.group/t/cleaning-up-log-id-and-rtlil-unescape-id/82/3) but I haven't gotten around to it. Actually I have a volunteer who's started working on it but I don't know when they'll be able to finish it. This doesn't affect my multithreaded stuff because for now I just ban logging in multithreaded phases.